lbbrhzn/ocpp #1505: Logging logs almost everything with INFO - should it be DEBUG?

BJReplay, 2025-01-30:
## Version of the custom_component

v0.7.0

## Configuration

```yaml
logger:
  default: warn
  logs:
     custom_components.solcast_solar: debug
     custom_components.epa_victoria_air_quality: debug
```

## Describe the bug

The integration currently logs almost everything with INFO level logging.

I believe that most of the information below should probably be logged as DEBUG level logging, so that it isn't logged unless the user has specified debug level logging for the integration in configuration.yaml, or has enabled debug logging in the UI.

## Debug log

```text
homeassistant  | 2025-01-30 14:41:45.253 INFO (MainThread) [ocpp] Switch_EV: send [3,"3c2d6a702dd04900852a06231d6816ce",{"idTagInfo":{"status":"Accepted"}}]
homeassistant  | 2025-01-30 14:41:45.254 INFO (MainThread) [ocpp] Switch_EV: receive message [2,"ea54de85000b4d90bba26a90ab3b7829","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2025-01-30T14:41:45.244665+11:00"}]
homeassistant  | 2025-01-30 14:41:45.260 INFO (MainThread) [ocpp] Switch_EV: send [3,"ea54de85000b4d90bba26a90ab3b7829",{}]
homeassistant  | 2025-01-30 14:41:54.482 INFO (MainThread) [ocpp] Switch_EV: receive message [2,"1928ba92861b43cc93dc21f4b2746cec","StartTransaction",{"connectorId":1,"idTag":"9bcaece2","meterStart":19838,"reservationId":0,"timestamp":"2025-01-30T14:41:54.4697659+11:00"}]
homeassistant  | 2025-01-30 14:41:54.490 INFO (MainThread) [ocpp] Switch_EV: send [3,"1928ba92861b43cc93dc21f4b2746cec",{"transactionId":1738208514,"idTagInfo":{"status":"Accepted"}}]
homeassistant  | 2025-01-30 14:41:54.490 INFO (MainThread) [ocpp] Switch_EV: receive message [2,"9b0ff6e53f8c45aea6f9716c9ca5d10a","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Charging","timestamp":"2025-01-30T14:41:54.4831188+11:00"}]
homeassistant  | 2025-01-30 14:41:54.496 INFO (MainThread) [ocpp] Switch_EV: send [3,"9b0ff6e53f8c45aea6f9716c9ca5d10a",{}]
homeassistant  | 2025-01-30 14:41:56.486 INFO (MainThread) [ocpp] Switch_EV: receive message [2,"cd8637db4e50423c9cd95268bf76b481","MeterValues",{"connectorId":1,"transactionId":1738208514,"meterValue":[{"timestamp":"2025-01-30T14:41:56.4848372+11:00","sampledValue":[{"value":"19838","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"},{"value":"230","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V"},{"value":"0","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A"},{"value":"0","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W"},{"value":"51.89","context":"Sample.Periodic","measurand":"SoC","location":"Outlet","unit":"Percent"}]}]}]
homeassistant  | 2025-01-30 14:41:56.497 INFO (MainThread) [ocpp] Switch_EV: send [3,"cd8637db4e50423c9cd95268bf76b481",{}]
homeassistant  | 2025-01-30 14:42:04.296 INFO (MainThread) [ocpp] Switch_EV: receive message [2,"98781305330043658d537f1e85e39da0","Heartbeat",{}]
homeassistant  | 2025-01-30 14:42:04.306 INFO (MainThread) [ocpp] Switch_EV: send [3,"98781305330043658d537f1e85e39da0",{"currentTime":"2025-01-30T03:42:04Z"}]
homeassistant  | 2025-01-30 14:42:12.463 INFO (MainThread) [ocpp] Switch_EV: receive message [2,"cc755eebc0874c99a993b5ccc8ac3e0e","MeterValues",{"connectorId":1,"transactionId":1738208514,"meterValue":[{"timestamp":"2025-01-30T14:42:12.4623859+11:00","sampledValue":[{"value":"19838","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"},{"value":"230","context":"Sample.Periodic","measurand":"Voltage","location":"Outlet","unit":"V"},{"value":"0","context":"Sample.Periodic","measurand":"Current.Import","location":"Outlet","unit":"A"},{"value":"0","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W"},{"value":"51.89","context":"Sample.Periodic","measurand":"SoC","location":"Outlet","unit":"Percent"}]}]}]
homeassistant  | 2025-01-30 14:42:12.467 INFO (MainThread) [ocpp] Switch_EV: send [3,"cc755eebc0874c99a993b5ccc8ac3e0e",{}]
```

drc38, 2025-01-31:
The messages come from the underlying occp library rather than the integration. You can add `ocpp: debug` to remove them from the log.

BJReplay, 2025-01-31:
> You can add `ocpp: warn`

Thanks! works.

I'm testing my automations against [MicroOcppSimulator](https://github.com/matth-x/MicroOcppSimulator) while getting ready for my charger to be installed next week.

I'll provide a PR for an update to the docs with anything I discover: it is a EN PlusTech caro.

closed 2025-01-31 by BJReplay