lbbrhzn/ocpp #1510: Wallbox Pulsar Max cannot complete handshake

Delorien84, 2025-02-11:
**Describe the bug**
Hello, I have Wallbox Pulsar Max with firmware 6.6.12 (latest) and I am not able to connect to the OCPP server. Charger is able to connect, sends `BootNotification` but then it is disconnected. It may be a problem, that the application sends `GetConfiguration` request twice. First before it receives `BootNotification` and second just after accepting it.

**Logs:**

```
2025-02-11 14:33:11.538 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
2025-02-11 14:33:11.538 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/wallbox
2025-02-11 14:33:11.538 INFO (MainThread) [custom_components.ocpp] Charger wallbox reconnected to 0.0.0.0:9000.
2025-02-11 14:33:13.539 INFO (MainThread) [ocpp] wallbox: send [2,"4800cc15-80d0-4e48-9816-0c383108ff3c","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2025-02-11 14:33:14.539 INFO (MainThread) [ocpp] wallbox: receive message [2,"15171824","BootNotification",{"chargePointSerialNumber": "530486","chargePointVendor": "Wall Box Chargers","meterType": "Internal NON compliant","meterSerialNumber": "","chargePointModel": "PLP2-M-2-4","iccid": "","chargeBoxSerialNumber": "530486","firmwareVersion": "6.6.12","imsi": ""}]
2025-02-11 14:33:14.563 INFO (MainThread) [ocpp] wallbox: send [3,"15171824",{"currentTime":"2025-02-11T13:33:14Z","interval":3600,"status":"Accepted"}]
2025-02-11 14:33:21.499 INFO (MainThread) [custom_components.evcc_intg.pyevcc_ha] going to read all frequent_data from evcc@http://localhost:7070
2025-02-11 14:33:23.540 INFO (MainThread) [ocpp] wallbox: send [2,"c3cc408b-afa4-4bd8-9c0f-1d5c7b77591d","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2025-02-11 14:33:23.548 INFO (MainThread) [custom_components.ocpp] Charger wallbox disconnected from 0.0.0.0:9000.
```

**Expected behavior**
Wallbox is connected to ocpp successfully 

**Screenshots**

[image]

delainus, 2025-02-13:
https://github.com/lbbrhzn/ocpp/issues/1503#issuecomment-2646438936 #

rbb-dev, 2025-02-16:
@drc38 @lbbrhzn
This is actually a bug.

In OCPP 2.0, the specification explicitly states that BootNotification must be the first message processed after establishing a connection. However, OCPP 1.6 does not enforce this requirement as strictly.

Looking back at my logs, I can see that the central system sends GetConfiguration for SupportedFeatureProfiles immediately after accepting the connection, even before the BootNotification was received and processed.

drc38, 2025-02-16:
Needs someone with the issue to try a fix eg while self.received_boot_notification is False wait with a timeout as a backstop if the boot notification is never received.

rbb-dev, 2025-02-17:
> Needs someone with the issue to try a fix eg while self.received_boot_notification is False wait with a timeout as a backstop if the boot notification is never received.
> 
> Also see [#1503 (comment)](https://github.com/lbbrhzn/ocpp/issues/1503#issuecomment-2628852339)

I’ve found and tested the issue. Right now, .post_connect() is being executed from start(), reconnect(), and _register_boot_notification(), but it should only run from _register_boot_notification(), since that’s already triggered by Action.boot_notification.

After removing .post_connect() from start() and reconnect(), the command sequence is now correct. The boot notification gets processed first, and that then kicks off post_connect() at the right time. This ensures that "GetConfiguration" for "SupportedFeatureProfiles" follows immediately after Action.boot_notification, as expected.

drc38, 2025-02-17:
@rbb-dev can a boot notification be guaranteed to be sent by the charger and received by HA? If not received then the charger would not be registered. By having post_connect in start and reconnect it is guaranteed to be run irrespective of charger start up behaviour at the time of the websocket connection. Hence the suggestion to remove from the boot notification logic and instead in post connect wait for the boot notification or timeout and continue.

rbb-dev, 2025-02-17:
https://github.com/mobilityhouse/ocpp/blob/master/docs/v16/ocpp-1.6%20edition%202.pdf

Sections 4.2 and 4.2.1
CP must be accepted by CS by way of BootNotification.
BootNotification must be the first and only message from CP, only if accepted by CS with Accepted, can it proceed, otherwise, it has to disconnect and retry again.

[image]

rbb-dev, 2025-02-17:
I just tested the modified code int he following 4 scenarios:
1 - CP hard reboot - BootNotification -> GetConfiguration :: Correct
2 - CP soft reboot - BootNotification -> GetConfiguration :: Correct
3 - Kick off the wifi - BootNotification -> GetConfiguration :: Correct
4 - Power Off/On - BootNotification -> GetConfiguration :: Correct

drc38, 2025-02-18:
I haven't had a chance to test but agree with your approach as SHALL == MUST it should be guaranteed for compliant chargers.

closed 2025-05-11 by github-actions[bot]