lbbrhzn/ocpp #1514: Growatt Thor A/S 22 - Voltage/Current/Power/Temperature values not populated

Plawasan, 2025-02-16:
**Describe the bug**
Growatt Thor A/S 22 Charger, OCPP integration v.0.7.0
When charging, I can see in the HA logs that the per-phase Voltage/Current/Power/Temperature values are correctly shown in the communication with the charger however there are not populated in any of the entities in HA (session energy is). I'm only using the Core Profile.

```
2025-02-16 21:33:01.842 INFO (MainThread) [ocpp] XGJ00003215101B9: receive message [2,"3779","MeterValues",{"connectorId":1,"transactionId":1739737116,"meterValue":[{"timestamp":"2025-02-16T22:33:00","sampledValue":[{"value":"999","context":"Sample.Periodic","format":"Raw","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"6.08","context":"Sample.Periodic","format":"Raw","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"6.05","context":"Sample.Periodic","format":"Raw","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"6.05","context":"Sample.Periodic","format":"Raw","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"225.5","context":"Sample.Periodic","format":"Raw","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"240.7","context":"Sample.Periodic","format":"Raw","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"226.9","context":"Sample.Periodic","format":"Raw","measurand":"Voltage","phase":"L3","unit":"V"},{"value":"1374","context":"Sample.Periodic","format":"Raw","measurand":"Power.Active.Import","phase":"L1","unit":"W"},{"value":"1445","context":"Sample.Periodic","format":"Raw","measurand":"Power.Active.Import","phase":"L2","unit":"W"},{"value":"1367","context":"Sample.Periodic","format":"Raw","measurand":"Power.Active.Import","phase":"L3","unit":"W"},{"value":"2.0","context":"Sample.Periodic","format":"Raw","measurand":"Temperature","unit":"Celsius"}]}]}]
2025-02-16 21:33:01.844 DEBUG (MainThread) [custom_components.ocpp] process_phases: metric: Current.Import, phase_info: {'unit': 'A', 'L1': 6.08, 'L2': 6.05, 'L3': 6.05} value: 18.180000 unit :A
2025-02-16 21:33:01.845 DEBUG (MainThread) [custom_components.ocpp] process_phases: metric: Voltage, phase_info: {'unit': 'V', 'L1': 225.5, 'L2': 240.7, 'L3': 226.9} value: 231.033333 unit :V
2025-02-16 21:33:01.845 DEBUG (MainThread) [custom_components.ocpp] process_phases: metric: Power.Active.Import, phase_info: {'unit': 'W', 'L1': 1374.0, 'L2': 1445.0, 'L3': 1367.0} value: 4186.000000 unit :W
```

**Screenshots**
[image]

Now if it only also reported SoC...

Thanks

altserg, 2025-02-18:
Welcome to the club. See my submitted issues #1452 and #1466, and also pull requests to fix them. See also discussion on #1487. I think that #1487 partially breaks the internal changer FW data structures.
I wonder which FW version you have.
Growatt does not support most of the sensors, i recommend to disable all not supported garbage. My working and making sense sensors are attached. SoC is not supported. Totalenergy is utility meter associated sensor, not provided by the charger

[image]

Plawasan, 2025-02-18:
Thanks, I'm on THOR_22AS-V2.2.16-20240902, if there is a newer one I'll have to switch back to the default ocpp server and see if I get an update.

Plawasan, 2025-03-06:
Solved in 0.8.2:

closed 2025-03-06 by Plawasan

altserg, 2025-03-07:
Have you managed to get the SoC values ?Sent from my iPhone

Plawasan, 2025-03-07:
No SoC unfortunately, only those values that I saw in debug are now populated in entities.

altserg, 2025-03-07:
I’m just curious what is pre condition to receive soc values. Is it car dependent ? Charger dependent ? Is it documented somewhere ?Sent from my iPhone

lbbrhzn, 2025-03-07:
The charger needs to be able to communicate with the car, for instance using ISO 15118.

alexisml, 2025-04-06:
Sorry to hijack the thread, but couldn't find other way to contact @Plawasan. Would you mind sharing how you set up the charger to connect to this integration? I have this installed and can reach the ws if I manually test it, but my Thor charger won't connect.

altserg, 2025-04-06:
Normally all should work out of the box with ocpp integration You just need to set up in charger settings proper server address. By default it connects to growatt

Plawasan, 2025-04-06:
> Sorry to hijack the thread, but couldn't find other way to contact [@Plawasan](https://github.com/Plawasan). Would you mind sharing how you set up the charger to connect to this integration? I have this installed and can reach the ws if I manually test it, but my Thor charger won't connect.

You'll need  to set the charger server address to point to your HA instance, in my case it's ws://192.168.1.10:9000. You can do this by either connecting directly via eth to the charger and giving your PC a static IP (e.g 192.168.1.10), then going to 192.168.1.5:8080, pwd 12345678 and then configuring the settings. I still find it annoying that this admin page isn't reachable when you actually connect the charger to wifi...

Alternatively, if you still have the charger connected to the shinephone app, you can force to AP mode and then you can set the server url through the initial setup of the charger