lbbrhzn/ocpp #1517: Supported Devices: EN+ Caro Series Home Wallbox confirmed working

BJReplay, 2025-02-19:
**Supported Charger**
The [EN Plustech Caro](https://www.en-plustech.com/product/caro-series-wallbox/) series chargers - also sold as [EV Switch](https://www.evswitchstore.com.au/pages/ev-charger-range) and supplied with BYD vehicles in Australia are working.

They require initial configuration with the Evchargo app (e.g, https://play.google.com/store/apps/details?id=cn.enplus.chargein or https://apps.apple.com/au/app/evchargo/id1533765408) which can configure multiple charger types.  It is necessary to set up network connectivity first (e.g. WiFi or LAN), and then, after the charger reboots, you can specify the OCCP server.

ws and wss are both supported.

The charger works with the "Force Smart Charging feature profile" option set, or clear.

**Describe the solution you'd like**
Add this charger to the supported chargers list in the documentation

**Additional context**
On adding to HA (configuring the OCPP integration the first time, or on restarting HA), three notifications are shown, and three warnings logged:

``` text
2025-02-19 22:03:08.422 WARNING (MainThread) [custom_components.ocpp] Failed with response: Rejected
2025-02-19 22:03:09.513 WARNING (MainThread) [custom_components.ocpp] Failed with response: Rejected
2025-02-19 22:03:09.933 WARNING (MainThread) [custom_components.ocpp] Failed with response: Rejected

```

[image]

However, these warnings don't appear to affect operation, and it is possible to set the Availability using the switch created by the integration once the charger is added:

[image]

I have tested with ` AC7000-AE-35`, but all models should work:

- AC7000-AE-35
- AC7000-AB-45C
- AC7000-BE-35
- ACO11K-AE-35
- ACO11K-BE-35

jwGeorge, 2025-02-19:
@BJReplay were you able to get the active power import reading populating? 

I have the same model. Mine stays at 0kW but I can see the import current changing over time.

BJReplay, 2025-02-19:
@jwGeorge I don't think so (but I removed my integration while playing with OCPP proxies to investigate, and haven't set it up again) to confirm. 

But I am almost certain, because I have this helper: 

``` json
{{
(
states('sensor.sn<my_serial_number>_voltage') | float * states('sensor.sn<my_serial_number>_current_import') | float
)
}} 
```

Which is then used for things like 

[image]

BJReplay, 2025-02-20:
> My voltage doesn't seem to populate either. 

I could have sworn that my voltage didn't populate - originally I was using the voltage reported by a Tasmota smart switch as a proxy.

Then, a couple of days later I asked myself WTF as the charger voltage was displayed. I don't know what or whether anything changed, but it was there, so I changed the template to use it.

jwGeorge, 2025-02-20:
> I could have sworn that my voltage didn't populate - originally I was using the voltage reported by a Tasmota smart switch as a proxy.

Yeah I was going to use the voltage from my Zigbee power meter but your template had me scratching my head. Will try a few things and report back!

BJReplay, 2025-02-22:
> Thanks for the info. My voltage doesn't seem to populate either.

@jwGeorge I've had a look at my voltage, and (at a guess) it's the value detected by the charger when it starts or after it reboots:

Here's a chart for the last week of four smart plugs and the charger.  

I know I can't rely on the voltage from the charger sensor for power calculations!

[image]

PR #1519 (2025-03-02): Add EN+ Caro
Closes #1517 

## Summary by CodeRabbit

- **Documentation**
  - Introduced a new section in our supported devices guide detailing the setup and configuration for the EN+ Caro Series Home Wallbox.
  - The updates include step-by-step instructions for connecting to the charger's access point, configuring network settings, adjusting sensor configurations, and integrating the device with Home Assistant for optimal performance.
closed 2025-03-02 by lbbrhzn