lbbrhzn/ocpp #1521: Ducto Compleo Solo Wallbox and OCPP

Hubaer25, 2025-02-24:
Hello,

I wan’t to connect my Ducto Completo Solo Wallbox to Home Assistant.

So I installed OCPP Integration, made a minimum config and configured the wallbox to the Backend via ws://hy.ha.ip.xxx:9000. On the Wallbox I configured the attributes I want to send

In Home Assistand I have now in OCPP 2 devices (central and charger) and 52 disabled entitys.

In the Home Assistant log I found the following Error:

Logger: homeassistant.helpers.frame
Quelle: helpers/frame.py:324
Erstmals aufgetreten: 16:13:39 (4 Vorkommnisse)
Zuletzt protokolliert: 16:13:39

Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JMSM4ETNAWQEAY1ZH6V105WE'), with device info: {'identifiers': {('ocpp', 'charger')}, 'via_device': ('ocpp', '01JMSM4ETNAWQEAY1ZH6V105WE')} at custom_components/ocpp/sensor.py, line 77: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues
Any Ideas or other solutions to get the Data from the wallbox to HA.

HA is on the actual Release Version 

Greetings

Ingo

StefanLobbenmeier, 2025-04-04:
I am also trying to get a compleo wallbox (ebox) to work, but actually struggled during the connection step, so maybe we can help each other out. I entered the address like this `ws://raspberry:9000`, it connected successfully to wifi and supposedly also the ocpp server, but I did not get any logs from the ocpp integration, so I guess it actually failed.

> ws://hy.ha.ip.xxx:9000

I noticed you are connecting via ip, I tried via hostname

> Some chargers require the url to be specified as an IP address, i.e. ‘192.168.178.1:9000’
https://home-assistant-ocpp.readthedocs.io/en/latest/installation.html#host-address-and-port

Looks like that is something I can try tomorrow.

Regarding your issue it seems to be still present here, line number changed though:
https://github.com/lbbrhzn/ocpp/blob/56223e8ba19af1be737d55830171741879e80335/custom_components/ocpp/sensor.py#L82

But it is a parameter, so possibly injected by home assistant itself, not sure, I cannot find the caller. It seems to be more of a warning that this will stop working end of the year, not an actual error

StefanLobbenmeier, 2025-04-05:
Quick update from my side, I used the ip instead of hostname and the device appeared in hass and I was able to add it. (Un-?)fortunately I did not get your error, those are my versions so maybe you update and try again:
- Open Charge Point Protocol (OCPP) 0.8.4
- HASS 2025.4.0

closed 2025-06-13 by github-actions[bot]