lbbrhzn/ocpp #1523: Autel MaxiCharger works well with the OCPP integration

jds11111, 2025-02-25:
The documentation does not list [this charger](https://www.amazon.com/dp/B0BDLQMR51) as working, but I just got it working with your integration. 

## Version of the custom_component

Latest version 0.7.0

## Configuration

Running it unsecure on port 8887. 
Followed these instructions on how to use the app to set up the charger. It worked as described.

## Describe the bug

There is no bug with the software, just wanted to add to the documentation as another charger that works!
After setting up, there was control for availability, charger on/off and current. There were also 10 sensors available: 

[image]

I blocked the charger from the internet, and still have fully local control. 

## Debug log

Great work, folks!

jds11111, 2025-02-27:
Mine correctly reports a maximum 48amps. I haven't ever changed it from that, though. Even with original app.

rj45jack, 2025-03-03:
> 👍🏼  Same here, Autel appears to have now properly report. 
> I am seeing an issue with the max current report as 32amps, but the charger is currently set to 50 amps. Will investigate if its being reported improperly or an integration issue. 

I am having the same issue. In the app my charger is set to 48A. The dips on the board are set to 48A. But this integration reports 32A as max.

jds11111, 2025-03-03:
> I am having the same issue. In the app my charger is set to 48A. The dips on the board are set to 48A. But this integration reports 32A as max.

My charger is set for 50amps, but my car limits to 48amps, which is what this integration reports. Is it maybe your car setting the limitation? What firmware is your charger running?

rj45jack, 2025-03-03:
> My charger is set for 50amps, but my car limits to 48amps, which is what this integration reports. Is it maybe your car setting the limitation? What firmware is your charger running?

Thanks. My car isn't connected yet, but supports 48 as well. Perhaps when I get home I can plug in to check.

alejandrom2, 2025-03-04:
What appears to be happening based on some testing is that Autel is not 100% supporting OCPP 2.0.1, in fact it returns that its using 1.6, when `SupportedFeatureProfiles` which is a 2.0.1 key responds with the charger results.
The following configuration key appears to properly return the set current `MaxChargeCurrentLimit` (in my case 50) 
Follow: https://github.com/lbbrhzn/ocpp/issues/1019#issuecomment-2677597813
Which has Autel's response

rj45jack, 2025-03-04:
> What appears to be happening based on some testing is that Autel is not 100% supporting OCPP 2.0.1, in fact it returns that its using 1.6, when `SupportedFeatureProfiles` which is a 2.0.1 key responds with the charger results. The following configuration key appears to properly return the set current `MaxChargeCurrentLimit` (in my case 50) Follow: [#1019 (comment)](https://github.com/lbbrhzn/ocpp/issues/1019#issuecomment-2677597813)

Thanks. This is my issue as well.

gomsykakkar, 2025-03-15:
While I am trying to configure using configuration wizard I am getting connection failed after reboot on charger.

I am using custom port as port 9000 was being used for other docker container.

[image]
[image]
[image]

jds11111, 2025-03-15:
Try it without the trailing forward slash. Also, maybe try using the IP of your OCPP integration, instead of 0.0.0.0

gomsykakkar, 2025-03-18:
Hi @jds11111 , thanks for your feedback.
I finally managed to run ocpp.
The issue was since I am running Home Assistant in docker container the ocpp port(9000) from container to host was not opened. I managed to resetup the home assistant container with ha port as 8123 and ocpp port as 9000.

Once bith port are opened then start configuring ocpp plugin with 9000 port.

Now I am able to start charger but not able to set maximum current and I am stuck with @ocpp#1019

jds11111, 2025-03-19:
I don't understand why I don't have that problem.

gomsykakkar, 2025-03-19:
Are you using home assistant on docker?

jds11111, 2025-03-19:
Yes, I am. This is the firmware version reported to OCPP: PFA0102|V0.00.00|V1.36.00|V1.21.00|2.4.3.0, Autel MaxichargerAC, and charger features: Profiles.CORE|FW|SMART|RES|REM|AUTH. The charger is blocked from reaching the internet at my router/firewall.

I am still running OCPP version 0.7.0, since none of the 0.8.x version work for me.

gomsykakkar, 2025-03-19:
I switched to 0.7.0 after coming on this thread.
And my device info is.
MaxiChargerAC by Autel
Firmware: PFA0101|V0.00.00|V1.55.00|V1.19.00|2.4.0.0

jds11111, 2025-03-19:
So, different firmware, but it is hard to decipher the different numbers. The app says that it is verion 3.2.1. If we are running the same OCPP backend, then mustn't it be the charger firmware version that is different?

voldemarz, 2025-05-05:
Can someone confirm if you can enable/disable charging, adjust max amps on MaxiCharger via OCPP?

rj45jack, 2025-05-05:
> Can someone confirm if you can enable/disable charging, adjust max amps on MaxiCharger via OCPP?

I can stop/start the charger,, but i am unable to adjust the power output.

voldemarz, 2025-05-05:
@jds11111 I don't have MaxiCharger yet, am considering to purchase one as they seem to have longer warranty that others. Can you adjust power output on yours? One person above mention it doesn't work for him.

jds11111, 2025-05-06:
> [@jds11111](https://github.com/jds11111) I don't have MaxiCharger yet, am considering to purchase one as they seem to have longer warranty that others. Can you adjust power output on yours? One person above mention it doesn't work for him.

Yes, I can, although I leave it at 48amps.

cedric47120, 2025-09-10:
I tried installing version 0.10.2, but unfortunately it still doesn’t work. For now we still need to stick with v0.7. I’ve also been waiting for a stable update for quite some time…

michael-adler, 2025-11-02:
Since all I wanted was MQTT without control, I switched to a different method. I wrote an OCPP relay that runs locally. The Autel connects to the relay. The relay then connects to the normal Autel OCPP server. Another local process tracks messages from the relay and generates MQTT topics in Home Assistant. I can then keep using the Autel app, fully functional, and track charging and energy use in HA.

The code is at https://github.com/michael-adler/ocpp2mqtt. I've only tested it with my Autel charger, but the intent is for it to work with any OCPP CP/CPMS pair. If you give it a try and find problems, please comment there.

I'm currently running these scripts outside of HA. I suppose they could be wrapped for HACS and run as a docker area, but I'm not sure it's worth the complexity. It might also be valuable at least to wrap the current repo in docker config, but I haven't done that yet either.

annejan-svg, 2025-12-23:
Will RFID tags still work / can be used when using the HA OCCP? I installed the integration but didn’t switch the server yet as I need the RFID functionality for my guests. Thanks.

ovsims, 2026-04-25:
> > Both of my Autel Chargers the Maxicharger and MaxiCharger AC. seems to work without issues on the newest 0.10.15
> 
> Is your car single-phase or three-phase? Mine is single-phase and instead of indicating 10A it indicates 3.3A, that's 10A/3 (phases)! That's wrong!
> 
> Thanks

[image]

I'm single phase in the US. Now the Maxicharger is on firmware 2.4.0.0 and it not reporting Current.Import,Energy.Active.Import.Register,Power.Active.Import,Voltage. So I can't see any sensors basically but power but the start and current adjustments work. Now I installed mine from scratch so this wasn't an upgrade. the Maxicharger AC is on firmware 2.2.0.0 and it reports all the sensors. I'm not sure what you have that indicating 3.3A

mmgfrcs, 2026-08-15:
> My car is single-phase and only uses one phase. Since the latest OCPP updates, the displayed current consumption has dropped to one third of the expected value. For example, my car charges on one phase at 10 A. The power indication is correct at around 2200 W, but the current, which should be about 10 A, is now shown as approximately 3.3 A.
> 
> I believe that in the recent updates, the system may have assumed a three-phase configuration and divided the total current by three to display the current per phase. Previously, it correctly showed 10 A.
> 

Me too. v0.10.18 of the integration reports current and voltage that is 1/3 of the actual value. The OCPP message is as follows:
```
{"value":"219.020000","unit":"V","measurand":"Voltage","phase":"L1"},
{"value":"0.050000","unit":"V","measurand":"Voltage","phase":"L2"},
{"value":"0.050000","unit":"V","measurand":"Voltage","phase":"L3"},
{"value":"29.464000","unit":"A","measurand":"Current.Export","phase":"L1"},
{"value":"29.464000","unit":"A","measurand":"Current.Import","phase":"L1"},
{"value":"0.011000","unit":"A","measurand":"Current.Export","phase":"L2"},
{"value":"0.011000","unit":"A","measurand":"Current.Import","phase":"L2"},
{"value":"0.010000","unit":"A","measurand":"Current.Export","phase":"L3"},
{"value":"0.010000","unit":"A","measurand":"Current.Import","phase":"L3"},
```

[image]

[image]

closed 2026-11-03 by github-actions[bot]