lbbrhzn/ocpp #1529: v0.8.0 - Entities not populated, remain unavailable

Plawasan, 2025-03-02:
**Describe the bug**
After migration to v0.8.0, I have received the "Migration failure, check logs" message however there was nothing in the log to help troubleshoot the issue

**To Reproduce**
Update to v0.8.8

**Expected behavior**
Entities populated

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**

Pre-upgrade:
2025-03-02 14:27:18.988 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'XGJ0000321510194': ping=0.0 ms, pong=381.0 ms

Upgrade:
2025-03-02 14:27:35.925 INFO (MainThread) [custom_components.hacs] <Integration lbbrhzn/ocpp> Running pre installation steps
2025-03-02 14:27:35.925 INFO (MainThread) [custom_components.hacs] <Integration lbbrhzn/ocpp> Pre installation steps completed
2025-03-02 14:27:35.925 INFO (MainThread) [custom_components.hacs] <Integration lbbrhzn/ocpp> Running installation steps
2025-03-02 14:27:38.260 INFO (MainThread) [custom_components.hacs] <Integration lbbrhzn/ocpp> Download of ocpp.zip completed
2025-03-02 14:27:38.264 INFO (MainThread) [custom_components.hacs] <Integration lbbrhzn/ocpp> Installation steps completed
2025-03-02 14:27:38.264 INFO (MainThread) [custom_components.hacs] <Integration lbbrhzn/ocpp> Running post installation steps
2025-03-02 14:27:38.270 INFO (SyncWorker_0) [homeassistant.loader] Loaded ocpp from custom_components.ocpp
2025-03-02 14:27:38.271 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration ocpp which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-03-02 14:27:38.285 INFO (MainThread) [custom_components.hacs] <Integration lbbrhzn/ocpp> Post installation steps completed

Charges is still hitting the OCPP server:
2025-03-02 14:27:39.160 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'XGJ0000321510194': ping=0.0 ms, pong=171.0 ms
2025-03-02 14:27:59.331 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'XGJ0000321510194': ping=0.0 ms, pong=171.0 ms
2025-03-02 14:28:19.286 DEBUG (MainThread) [custom_components.ocpp] Closing websocket to 'XGJ0000321510194'

                                                                                                                                                                          
Charger identified:                                                                                                                                                                               
2025-03-02 14:31:05.916 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6                                                                      
2025-03-02 14:31:05.916 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/XGJ0000321510194                                                                    
                                                                                                                 

Endpoint created for charger:
2025-03-02 14:34:58.514 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
2025-03-02 14:34:58.514 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/XGJ0000321510194
2025-03-02 14:34:58.514 INFO (MainThread) [custom_components.ocpp] Charger match found for XGJ0000321510194:XGJ0000321510194
2025-03-02 14:34:58.514 DEBUG (MainThread) [custom_components.ocpp] Central settings: CentralSystemSettings(csid='central', host='0.0.0.0', port=9000, ssl=False, ssl_certfile_path='/config/fullchain.pem', ssl_keyfile_path='/config/privkey.pem', websocket_close_timeout=10, websocket_ping_interval=20, websocket_ping_timeout=20, websocket_ping_tries=2, cpids=[{'XGJ0000321510194': {'cpid': 'XGJ0000321510194', 'max_current': 32, 'monitored_variables_autoconfig': True, 'meter_interval': 60, 'idle_interval': 900, 'skip_schema_validation': True, 'force_smart_charging': True, 'monitored_variables': 'Current.Export,Current.Import,Current.Offered,Energy.Active.Export.Interval,Energy.Active.Export.Register,Energy.Active.Import.Interval,Energy.Active.Import.Register,Energy.Reactive.Export.Interval,Energy.Reactive.Export.Register,Energy.Reactive.Import.Interval,Energy.Reactive.Import.Register,Frequency,Power.Active.Export,Power.Active.Import,Power.Factor,Power.Offered,Power.Reactive.Export,Power.Reactive.Import,RPM,SoC,Temperature,Voltage'}}], subprotocols=['ocpp1.6', 'ocpp2.0.1'])
2025-03-02 14:34:58.516 INFO (MainThread) [custom_components.ocpp] Charger XGJ0000321510194:XGJ0000321510194 connected to 0.0.0.0:9000.
2025-03-02 14:34:58.516 INFO (MainThread) [custom_components.ocpp] 1 charger(s): {'XGJ0000321510194': 'XGJ0000321510194'} now connected to central system:central.
2025-03-02 14:34:59.014 INFO (MainThread) [ocpp] XGJ0000321510194: receive message [2,"141","StatusNotification",{"connectorId":1,"errorCode":"NoError","info":"","status":"Available"}]
2025-03-02 14:34:59.017 INFO (MainThread) [ocpp] XGJ0000321510194: send [3,"141",{}]
2025-03-02 14:35:18.649 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'XGJ0000321510194': ping=0.0 ms, pong=132.0 ms
                                                              
However nothing shows up under the device:

[image]

xrado, 2025-03-02:
stopped to work for me too. I see "Migration error: Check the logs" ..under integration, but can't find any errors in the logs

Arnoudd, 2025-03-02:
Same issue here. I reverted to 0.7.0 and it works again. 

Version 0.8.0 doesn't seem to work for me.

dragorex71, 2025-03-02:
Same for me with my ABB Terra AC W7. Reverted to 0.7.0.

nils1981, 2025-03-02:
Same for me. Using Webasto Unite (Vestel EVC)

N3rdix, 2025-03-02:
Same here, downgraded to 0.7.0 for now

drc38, 2025-03-02:
When on v0.8.0 try manually rebooting your charger. It seems a number of chargers are not complying with the standard and sending a boot notification when they connect to the server.

drc38, 2025-03-02:
Hold off any revert, I'll add a backstop to the monitor connection for non-compliant chargers

Plawasan, 2025-03-02:
> When on v0.8.0 try manually rebooting your charger. It seems a number of chargers are not complying with the standard and sending a boot notification when they connect to the server.
> 
> [@lbbrhzn](https://github.com/lbbrhzn) we may need to revert [#1527](https://github.com/lbbrhzn/ocpp/pull/1527)

That seems to have helped with my charger at least - still on v0.8.0 and entities are back but I'll wait for the "proper" fix.

markkors, 2025-03-02:
Manualy rebooting my charger (Alfen Single S Line) after installed 0.8.0 (had same Migration error issues) did not help, entities still not there. Reverted back to 0.7.0 for now.

muc2810, 2025-03-03:
Same here, with a Webasto Live Wallbox.
Not a single Entity available

mfuchs1984, 2025-03-03:
@drc38 
>It seems a number of chargers are not complying with the standard and sending a boot notification when they connect to the server.

Are you sure the charger is required to send a BootNotification on connect? I can only find required StatusNotification, the section for BootNotification is only taking about charger booting or rebooting. But maybe I miss something.

The occp 1.6-j document even says
>5.4. Reconnecting
When reconnecting a charge point should not send a BootNotification unless one or more of the elements in the BootNotification have changed since the last connection. For the previous SOAP based solutions this was considered good practice but when using WebsSocket the server can already make the match between the identity and a communciation channel at the moment the connection is established. There is no need for an additional message.

At least in case the charger supports the RemoteTrigger profile, the central system can trigger a BootNotification.

Plawasan, 2025-03-04:
v0.8.1 seems to be working now, I still got a migration error but deleting the previous config, adding back the central component and waiting for the charger to ping it worked OK. The Unlock and Reset buttons were initially unavailable and only became active after an hour:

[image]

Unrelated - glad to see that ocpp.get_configuration now returns a response_variable, that simplified a couple things for me.

PR #1530 (2025-03-03): add post connect backstop, fix ssl references
Fixes #1529 fixes #1531

## Summary by CodeRabbit

## Summary by CodeRabbit

- **New Features**
  - Introduced a fallback mechanism that ensures smoother device connections even when standard startup signals are delayed or missing.
  - Enhanced logging for migration processes, providing clearer visibility on configuration updates and potential issues.

- **Refactor**
  - Updated the connection and reconnection workflows to provide enhanced reliability and consistent performance with a wider variety of devices.
  - Adjusted SSL certificate path refe [... cut]