lbbrhzn/ocpp #1531: Failed migration from 0.7.0 to 0.8.0 with WSS / Certificate

BJReplay, 2025-03-02:
**Describe the bug**
If you have set up a wss configuration with a certificate, the upgrade from 0.7.0 to 0.8.0 fails.

**To Reproduce**
Steps to reproduce the behavior:
1. Accept offered 0.8.0 upgrade
2. Restart HA
3. Note failed migration
4. See error

**Expected behavior**
Upgrade should handle certificates.

**Screenshots**
Error following restart following upgrade:

[image]

**Desktop (please complete the following information):**
 - OS: Win11 22H2 WSL Ubuntu 24.04 Docker
 - Browser: edge
 - Version 135

**Additional context**

See also https://github.com/drc38/ocpp/issues/83 - this error has crept in in the multiple chargers changes.

Logs on re-enabling disabled integration to capture failure:

``` text
homeassistant  | 2025-03-03 08:51:35.626 DEBUG (MainThread) [custom_components.ocpp] Migrating configuration from version 1.1
homeassistant  | 2025-03-03 08:51:35.626 DEBUG (MainThread) [custom_components.ocpp] Migration to configuration version 2.0 successful
homeassistant  | 2025-03-03 08:51:35.629 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry central for ocpp
homeassistant  | Traceback (most recent call last):
homeassistant  |   File "/usr/src/homeassistant/homeassistant/config_entries.py", line 637, in __async_setup_with_context
homeassistant  |     result = await component.async_setup_entry(hass, self)
homeassistant  |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
homeassistant  |   File "/config/custom_components/ocpp/__init__.py", line 104, in async_setup_entry
homeassistant  |     central_sys = await CentralSystem.create(hass, entry)
homeassistant  |                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
homeassistant  |   File "/config/custom_components/ocpp/api.py", line 153, in create
homeassistant  |     localhost_certfile = self.settings.certfile
homeassistant  |                          ^^^^^^^^^^^^^^^^^^^^^^
homeassistant  | AttributeError: 'CentralSystemSettings' object has no attribute 'certfile'
homeassistant  | 2025-03-03 08:51:36.632 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
homeassistant  | Traceback (most recent call last):
homeassistant  |   File "/usr/src/homeassistant/homeassistant/helpers/storage.py", line 515, in _async_callback_delayed_write
homeassistant  |     await self._async_handle_write_data()
homeassistant  |   File "/usr/src/homeassistant/homeassistant/helpers/storage.py", line 540, in _async_handle_write_data
homeassistant  |     await self._async_write_data(self.path, data)
homeassistant  |   File "/usr/src/homeassistant/homeassistant/helpers/storage.py", line 545, in _async_write_data
homeassistant  |     await self.hass.async_add_executor_job(self._write_data, self.path, data)
homeassistant  |   File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
homeassistant  |     result = self.fn(*self.args, **self.kwargs)
homeassistant  |   File "/usr/src/homeassistant/homeassistant/helpers/storage.py", line 552, in _write_data
homeassistant  |     data["data"] = data.pop("data_func")()
homeassistant  |                    ~~~~~~~~~~~~~~~~~~~~~^^
homeassistant  |   File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2426, in _data_to_save
homeassistant  |     "entries": [entry.as_storage_fragment for entry in self._entries.values()]  # type: ignore[misc]
homeassistant  |                 ^^^^^^^^^^^^^^^^^^^^^^^^^
homeassistant  |   File "src/propcache/_helpers_c.pyx", line 80, in propcache._helpers_c.cached_property.__get__
homeassistant  |   File "/usr/src/homeassistant/homeassistant/config_entries.py", line 534, in as_storage_fragment
homeassistant  |     return json_fragment(json_bytes_sorted(self.as_dict()))
homeassistant  |                          ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
homeassistant  | TypeError: Dict key must a type serializable with OPT_NON_STR_KEYS
```

Downgrade allows OCPP to start.

Settings that were used were (defaults shown here - simulating adding a second instance) ticking the Secure Connection checkbox and specifying certificate files that were available (mounted to the HA Docker instance) for the OCPP charger.

[image]

drc38, 2025-03-02:
cheers, added fix to #1530

une18, 2025-03-03:
> cheers, added fix to [#1530](https://github.com/lbbrhzn/ocpp/pull/1530)

Same here, how to apply the fix?

PR #1530 (2025-03-03): add post connect backstop, fix ssl references
Fixes #1529 fixes #1531

## Summary by CodeRabbit

## Summary by CodeRabbit

- **New Features**
  - Introduced a fallback mechanism that ensures smoother device connections even when standard startup signals are delayed or missing.
  - Enhanced logging for migration processes, providing clearer visibility on configuration updates and potential issues.

- **Refactor**
  - Updated the connection and reconnection workflows to provide enhanced reliability and consistent performance with a wider variety of devices.
  - Adjusted SSL certificate path refe [... cut]