lbbrhzn/ocpp #1533: Failed upgrade tp v0.8.0

filmackay, 2025-03-03:
**Describe the bug**
I attempted a 0.8.0 upgrade, which did not work - reverted to v0.7.0. Now this problem persists:

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade to v0.8.0
2. Downgrade to v0.7.0
3. Restart HA
4. Wallbox charger reboots after a few minutes, and device is never available in HA

**Expected behavior**
OCPP start without error.

**Logs**
```
2025-03-03 14:07:26.626 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
2025-03-03 14:07:26.626 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/charger
2025-03-03 14:07:26.626 INFO (MainThread) [custom_components.ocpp] Charger charger connected to 10.0.9.2:9000.
2025-03-03 14:07:30.594 DEBUG (MainThread) [custom_components.ocpp] Received boot notification for charger: {'charge_point_serial_number': 'xxx', 'charge_point_vendor': 'Wall Box Chargers', 'meter_type': 'Internal NON compliant', 'meter_serial_number': '', 'charge_point_model': 'PLP2-0-2-4', 'iccid': '', 'charge_box_serial_number': 'xxx', 'firmware_version': 'xxx', 'imsi': ''}
2025-03-03 14:07:30.595 DEBUG (MainThread) [custom_components.ocpp] Updating device info charger: {'charge_point_serial_number': 'xxx', 'charge_point_vendor': 'Wall Box Chargers', 'meter_type': 'Internal NON compliant', 'meter_serial_number': '', 'charge_point_model': 'PLP2-0-2-4', 'iccid': '', 'charge_box_serial_number': 'xxx', 'firmware_version': 'xxx', 'imsi': ''}
```
[`xxx` is redacted]

dragorex71, 2025-03-03:
Delete the integration (not from HACS but from the Integration in Home Assistant Setting) and install it again. 
The new OCPP instance will find only the "central". Than reboot your charger and it should be discovered again.
This works for me on 0.8.0

teamllamauk, 2025-03-03:
After deleting the integration it finds the charger and adds it but everything is listed as unavailable. 

I've tried multiple reboots of home assistant and the charger and it stays the same.

When I check the charger through it's configuration app it says it is successfully connected to the occp server.

filmackay, 2025-03-03:
Thanks @dragorex71 !

Now I am getting the following when I try to start charging: `Warning: Start transaction failed with response Rejected`
(it started charging automatically)

Have you seen that before?

closed 2025-03-03 by filmackay