lbbrhzn/ocpp #1535: Charger unavailable in 0.8

teamllamauk, 2025-03-03:
**Describe the bug**
After upgrading to 0.8 and readding the charger to fix the failed to migrate error the charger is detected and added but all entities are listed as unavailable.

**To Reproduce**
Add charger to 0.8 and all entities will show as unavailable.

**Expected behavior**
Entities should be available

Home assistant running on raspberry pi 4. Same experience when using office HA app on android or via edge, chrome and firefox on windows 10.

**Additional context**
Charger is a Hydra Cubis
Ocpp integration lists it as: AC_7KW_001_H_B
by LINCHR

MartinAirN, 2025-03-03:
I have the exact same issue, re-moving / re-adding however shows nothing. Also, the logs are empty with regards of OCPP.

teamllamauk, 2025-03-03:
I have left it for the last hour and it has just finally populated the entities and is now showing as available

rhpijnacker, 2025-03-03:
Same problem here.
I tried removing the current device and rebooted HA.
This left HA in an unbootable state :o
I had to restore core.device_registry from a backup before HA would boot again.

alainsch, 2025-03-03:
Same problem, HA didn't boot anymore after this update. Had to restore

teamllamauk, 2025-03-03:
> I did not test deleting the integration, as too many automations are linked that would need a full reconfigure.
> 

If it helps, I did a full unusual of both the integration and also from HACs and then reinstalled with the same charger name and all my automations still work without modification. I am using node red though for those automations.

MartinAirN, 2025-03-03:
> If it helps, I did a full unusual of both the integration and also from HACs and then reinstalled with the same charger name and all my automations still work without modification. I am using node [...]

Indeed, this at least fixed it for now. Now just the worry that a reboot still breaks it

closed 2025-05-11 by github-actions[bot]