lbbrhzn/ocpp #1539: update v0.8.0 to v0.8.1 migration error "Could not find charger id during migration"

hreinhar, 2025-03-03:
## Version of the custom_component

OCPP v0.8.1
HA Core 2025.2.5
Supervisor 2025.02.4
Operating System 14.2

## Configuration
I have two ocpp instances configured for two wallboxes

## Describe the bug

After the last update from v0.8.0 to v.0.8.1 I got a migration error and Entities are no longer available.
All other updates before are fine.

## Debug log

```text

2025-03-03 15:38:24.198 WARNING (MainThread) [custom_components.ocpp] Could not find charger id during migration, try a clean install
2025-03-03 15:38:24.200 WARNING (MainThread) [custom_components.ocpp] Could not find charger id during migration, try a clean install
2025-03-03 15:40:10.927 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.wattpilot_parkplatz_connection_charged from integration wattpilot has state class total_increasing, but its state is negative. Triggered by state -0.001 with last_updated set to 2025-03-03T14:38:31.131386+00:00. Please report it to the author of the 'wattpilot' custom integration
2025-03-03 15:40:10.929 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.wattpilot_garage_connection_charged from integration wattpilot has state class total_increasing, but its state is negative. Triggered by state -0.001 with last_updated set to 2025-03-03T14:38:31.150856+00:00. Please report it to the author of the 'wattpilot' custom integration

```

BaukeDeVries, 2025-03-03:
Same problem here.

BaukeDeVries, 2025-03-03:
Same problem here.

jesperlowe, 2025-03-03:
Same issue here
2025-03-03 16:00:04.910 WARNING (MainThread) [custom_components.ocpp] Could not find charger id during migration, try a clean install

nils1981, 2025-03-03:
- same for me

toast1234, 2025-03-03:
My Wallbox
PLP2-M-2-4
Firmware: 6.6.12
Goes in to endless reboot after integration upgrade. It is not enough to rollback to v7 reboot HA and Wallbox. I have to factory reset the Wallbox for it to connect again with OCPP integration.

Arnoudd, 2025-03-03:
Same here with an ABB charger.
I reverted back to 0.7.0 as the last working version for me.

toast1234, 2025-03-03:
As hinted in other reports I gave it one more try and this works for my Wallbox.
1. On working version 0.7. Settings -> Devices -> OCPP -> Delete Integration "central"
2. HACS -> OCPP -> Upgrade/Reinstall to v0.8.1
3. Settings -> Devices -> Add integration OCPP
4. Restart HA
5. Reuse the same naming central/charger and the v0.8.1 integration now works

In the v0.7 Integration I had a lot of entries disable as they were not valid for my Wallbox if that might have had anything to with failing upgrade.

Dixet, 2025-03-03:
> [@Dixet](https://github.com/Dixet) After reverting to 0.7.0 have you tried to restart your wallbox? My "Juice Charger Me" wasnt reachable via network until I have rebooted it.

Reverting to 0.7 gave the same error in home assistant. By recovering a previous backup (also with 0.7.0) the home assistant part worked again. I then had to completely factory reset the wallbox to get the connection working.

After that, updating to 0.8.1 the way @toast1234 described worked for me. So finally everything up-and-running again.

drc38, 2025-03-03:
> As hinted in other reports I gave it one more try and this works for my Wallbox.
>
> 1. On working version 0.7. Settings -> Devices -> OCPP -> Delete Integration "central"
> 2. HACS -> OCPP -> [...]

If this doesn't work for your charger please report the last known value in sensor.charger_id noting if you changed charger to something else eg xxx in the setup it will be sensor.xxx_id

beardyname, 2025-03-03:
Just piped in to say the same thing.
Will not have time to test the instructions from @toast1234 until this weekend.

BJReplay, 2025-03-03:
> If this doesn't work for your charger please report the last known value in sensor.charger_id noting if you changed charger to something else eg xxx in the setup it will be sensor.xxx_id

Similar issue here, @drc38:

My charger is named sn10052312237994 as that's the serial number of the charger, and what it appends to the address of the ocpp server

Here are the entries from core.entity_registry (the entities are not available in the UI after upgrade):

`grep sensor.sn10052312237994_id core.entity_registry`

``` json
      {"config_entry_id":null,"created_at":"2025-02-01T11:15:20.592057+00:00","entity_id":"sensor.sn10052312237994_id","id":"5db7c764b254c061dece1a15a8a59f24","modified_at":"2025-03-03T22:01:34.308734+00:00","orphaned_timestamp":1741039294.3461263,"platform":"ocpp","unique_id":"ocpp.SN10052312237994.id.sensor"},
      {"config_entry_id":null,"created_at":"2025-02-01T11:15:20.588607+00:00","entity_id":"sensor.sn10052312237994_id_tag","id":"f0bc8771493b43a9314ac2888fd71d30","modified_at":"2025-03-03T22:01:34.309956+00:00","orphaned_timestamp":1741039294.3461263,"platform":"ocpp","unique_id":"ocpp.SN10052312237994.id.tag.sensor"}
```

As for 0.7.0 to 0.8.0, I got the same error message reported in the UI as I noted in #1531 

[image]

I don't know if this error is related to OCPP, but it appears immediately before a series of OCPP errors in the log:

```
2025-03-04 00:00:29.753 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/storage.py", line 515, in _async_callback_delayed_write
    await self._async_handle_write_data()
  File "/usr/src/homeassistant/homeassistant/helpers/storage.py", line 540, in _async_handle_write_data
    await self._async_write_data(self.path, data)
  File "/usr/src/homeassistant/homeassistant/helpers/storage.py", line 545, in _async_write_data
    await self.hass.async_add_executor_job(self._write_data, self.path, data)
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/storage.py", line 552, in _write_data
    data["data"] = data.pop("data_func")()
                   ~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2426, in _data_to_save
    "entries": [entry.as_storage_fragment for entry in self._entries.values()]  # type: ignore[misc]
                ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "src/propcache/_helpers_c.pyx", line 80, in propcache._helpers_c.cached_property.__get__
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 534, in as_storage_fragment                                                                                                                                                                                           return json_fragment(json_bytes_sorted(self.as_dict()))
                         ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
TypeError: Dict key must a type serializable with OPT_NON_STR_KEYS
```

Followed immediately by four similar failures for:

- ocpp.sensor
- ocpp.switch
- ocpp.number
- ocpp.button

```
2025-03-04 00:02:22.286 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry central for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 637, in __async_setup_with_context                                                                                                                                                                                    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                                                                                                                                File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 96, in async_setup_entry
    return await hass.data[DATA_COMPONENT].async_setup_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 182, in async_setup_entry
    raise ValueError(
    ...<2 lines>...
    )
ValueError: Config entry central (01JMEXZJ2EZPXMGQRZVZDEY1Y7) for ocpp.sensor has already been setup!
2025-03-04 00:02:22.289 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry central for switch
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 637, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/switch/__init__.py", line 79, in async_setup_entry
    return await hass.data[DATA_COMPONENT].async_setup_entry(entry)
[... 13 lines cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 182, in async_setup_entry
    raise ValueError(
    ...<2 lines>...
    )
ValueError: Config entry central (01JMEXZJ2EZPXMGQRZVZDEY1Y7) for ocpp.number has already been setup!
[... cut ...] ath":"/ocppcerts/fullchain.cer","ssl_keyfile_path":"/ocppcerts/ocppmydomain.key","websocket_close_timeout":10,"websocket_ping_interval":20,"websocket_ping_timeout":20,"websocket_ping_tries":2},"disabled_by":"user","discovery_keys":{},"domain":"ocpp","entry_id":"01JMEXZJ2EZPXMGQRZVZDEY1Y7","minor_version":1,"modified_at":"2025-02-19T11:02:39.950856+00:00","options":{},"pref_disable_new_entities":false,"pref_disable_polling":false,"source":"user","title":"central","unique_id":null,"version":1}
```

drc38, 2025-03-03:
@BJReplay what is the last known good value of this sensor in the UI `sensor.sn10052312237994_id`?

BJReplay, 2025-03-04:
@drc38 SN10052312237994

See annotated timeline (will edit from desktop)

drc38, 2025-03-04:
Ok, so you're now up and running on 0.8.1 ? 

To do a clean install does require removal of the integration, reboot and then install 0.8.1 otherwise the HA registries don't get cleaned.

BJReplay, 2025-03-04:
> Ok, so you're now up and running on 0.8.1 ?

Yes.  Couldn't migrate, couldn't cleanly uninstall (remove integration, removed from HACS, rebooted, HA wouldn't start): 

``` text
homeassistant  | 2025-03-04 09:17:06.497 ERROR (MainThread) [root] Uncaught exception
homeassistant  | Traceback (most recent call last):
homeassistant  |   File "<frozen runpy>", line 198, in _run_module_as_main
homeassistant  |   File "<frozen runpy>", line 88, in _run_code
homeassistant  |   File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module>
homeassistant  |     sys.exit(main())
homeassistant  |              ~~~~^^
homeassistant  |   File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main
homeassistant  |     exit_code = runner.run(runtime_conf)
homeassistant  |   File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run
homeassistant  |     return loop.run_until_complete(setup_and_run_hass(runtime_config))
homeassistant  |            ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
homeassistant  |   File "/usr/local/lib/python3.13/asyncio/base_events.py", line 720, in run_until_complete
homeassistant  |     return future.result()
homeassistant  |            ~~~~~~~~~~~~~^^
homeassistant  |   File "/usr/src/homeassistant/homeassistant/runner.py", line 122, in setup_and_run_hass
homeassistant  |     hass = await bootstrap.async_setup_hass(runtime_config)
homeassistant  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
homeassistant  |   File "/usr/src/homeassistant/homeassistant/bootstrap.py", line 343, in async_setup_hass
homeassistant  |     await async_from_config_dict(config_dict, hass) is not None
homeassistant  |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
homeassistant  |   File "/usr/src/homeassistant/homeassistant/bootstrap.py", line 475, in async_from_config_dict
homeassistant  |     await async_load_base_functionality(hass)
homeassistant  |   File "/usr/src/homeassistant/homeassistant/bootstrap.py", line 444, in async_load_base_functionality
homeassistant  |     await asyncio.gather(
homeassistant  |     ...<13 lines>...
homeassistant  |     )
homeassistant  |   File "/usr/src/homeassistant/homeassistant/helpers/device_registry.py", line 1262, in async_load
homeassistant  |     await async_get(hass).async_load()
homeassistant  |   File "/usr/src/homeassistant/homeassistant/helpers/device_registry.py", line 1156, in async_load
homeassistant  |     tuple(iden)  # type: ignore[misc]
homeassistant  |     ~~~~~^^^^^^
homeassistant  | TypeError: unhashable type: 'dict'
homeassistant  | [09:17:07] INFO: Home Assistant Core finish process exit code 1
homeassistant  | [09:17:07] INFO: Home Assistant Core service shutdown

```

I figured the issue was something left over in device registry, so I edited, looked for ocpp, and deleted a left-over entry.  I should have copied it first but was needed to get the integration running one way or the other to fully charge today ahead of a trip, so neglected to save the line that I removed.

drc38, 2025-03-04:
Very weird, glad it's sorted in time charge!

erepeo, 2025-03-04:
Warning: Do Not Update to V0.8.1!

Do not update the plugin to version V0.8.1 – it has a migration bug similar to the problematic V0.8.0. This new version does not fix previous issues and might even make things worse.

In my case, after a failed migration, it somehow set the charging current to the maximum 32A, overloaded my inverter, and caused a total power outage in my house.

Stay away from this update – do not install or even touch it!

toast1234, 2025-03-04:
When my charger did a reset it defaulted to max power. This was not reflected in the integration (still showing the last known setting). So when charging started it was at max but then my power monitoring automation adjusted it down to not blow the main fuse. So maybe not directly related to v0.7 or v0.8.

erepeo, 2025-03-04:
> When my charger did a reset it defaulted to max power. This was not reflected in the integration (still showing the last known setting). So when charging started it was at max but then my power [...]

I don’t have any automation that reduces the charging power, so the inverter got overloaded. I also couldn’t manually change the charging power.  

I rolled back to **V0.7.0**, and everything is working fine now.

hreinhar, 2025-03-04:
Yes, v0.8.2 is also not working for me. Still the migration error.
I've now replaced the directory /homeassistant/custom_components/ocpp with the folder of the backup from v0.7.0
After reboot it looks good, but I've not testet to update again and I'm now running the old version.

closed 2025-05-12 by github-actions[bot]

netmikey, 2025-05-12:
It seems like quite a few of us have had to roll back and use an old version. With this seeing no activity and being closed, is there any update or guideline for the people having this issue? /cc @drc38

drc38, 2025-05-12:
Not without people posting a more detailed debug. Given there are approx 5k installs using 0.8.4 any issue is likely to be related to your specific circumstances/charger/firmware.

Tomakava, 2025-05-12:
I also had this problem when upgrading and solved it by removing integration and adding it again.

But now tested more with new empty Home Assistant 2025.5.1 and installed OCPP integration v0.7.0.
Created "charger" with all default settings without even connecting real charger. Then upgraded to v0.8.4 and got error:
```
2025-05-12 22:58:29.592 INFO (MainThread) [custom_components.ocpp] config = {}
2025-05-12 22:58:29.592 INFO (MainThread) [custom_components.ocpp] Migrating configuration from version 1.1
2025-05-12 22:58:29.592 WARNING (MainThread) [custom_components.ocpp] Could not find charger id during migration, try a clean install
```
It could be race condition that hass.states.get is not yet available.
If I disable OCPP integration and enable some time after restart:
```
2025-05-12 23:02:13.445 INFO (MainThread) [custom_components.ocpp] config = {}
2025-05-12 23:02:13.445 INFO (MainThread) [custom_components.ocpp] Migrating configuration from version 1.1
2025-05-12 23:02:13.445 INFO (MainThread) [custom_components.ocpp] Migration to configuration version 2.0 successful
2025-05-12 23:02:14.449 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/storage.py", line 515, in _async_callback_delayed_write
    await self._async_handle_write_data()
  File "/usr/src/homeassistant/homeassistant/helpers/storage.py", line 540, in _async_handle_write_data
    await self._async_write_data(self.path, data)
  File "/usr/src/homeassistant/homeassistant/helpers/storage.py", line 545, in _async_write_data
    await self.hass.async_add_executor_job(self._write_data, self.path, data)
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/storage.py", line 552, in _write_data
    data["data"] = data.pop("data_func")()
                   ~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2722, in _data_to_save
    "entries": [entry.as_storage_fragment for entry in self._entries.values()]  # type: ignore[misc]
                ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "src/propcache/_helpers_c.pyx", line 82, in propcache._helpers_c.cached_property.__get__
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 646, in as_storage_fragment
    return json_fragment(json_bytes_sorted(self.as_dict()))
                         ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
TypeError: Dict key must a type serializable with OPT_NON_STR_KEYS
```
If I now restart Home Assistant it no longer starts with this error:
```
2025-05-12 23:08:09.978 ERROR (MainThread) [root] Uncaught exception
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module>
    sys.exit(main())
             ~~~~^^
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
           ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 719, in run_until_complete
    return future.result()
           ~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/runner.py", line 122, in setup_and_run_hass
    hass = await bootstrap.async_setup_hass(runtime_config)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/bootstrap.py", line 337, in async_setup_hass
    await async_from_config_dict(config_dict, hass) is not None
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/bootstrap.py", line 474, in async_from_config_dict
    await async_load_base_functionality(hass)
  File "/usr/src/homeassistant/homeassistant/bootstrap.py", line 443, in async_load_base_functionality
    await asyncio.gather(
    ...<13 lines>...
    )
  File "/usr/src/homeassistant/homeassistant/helpers/device_registry.py", line 1468, in async_load
    await async_get(hass).async_load()
  File "/usr/src/homeassistant/homeassistant/helpers/device_registry.py", line 1300, in async_load
    tuple(iden)  # type: ignore[misc]
    ~~~~~^^^^^^
TypeError: unhashable type: 'dict'
```
In device registry I see bad configuration:
```
    "identifiers": [
        [
            "ocpp",
            {
                "entity_id": "sensor.charger_id",
                "state": "unavailable",
                "attributes": {
                    "restored": true,
                    "icon": "mdi:ev-station",
                    "friendly_name": "ID",
                    "supported_features": 0
                },
                "last_changed": "2025-05-12T20:01:56.578511+00:00",
                "last_reported": "2025-05-12T20:01:56.578511+00:00",
                "last_updated": "2025-05-12T20:01:56.578511+00:00",
                "context": {
                    "id": "01JV31DYB2C8JW1PFJQ9C4J4ZK",
                    "parent_id": null,
                    "user_id": null
                }
            }
        ],
        [
            "ocpp",
            "charger"
        ]
    ],
```
If I try migration again and replace hass.states.get in `__init__.py` with hardcoded random string everything works correctly.
May be it is necessary to call `.state`?