lbbrhzn/ocpp #1542: 8.2 -> Still migration error

ssieppi, 2025-03-04:
OCPP v0.8.2
HA Core 2025.2.5
Supervisor 2025.03.0
Operating System 14.2

Configuration
One Ha installation conncted to ABB terra w22

Describe the bug
After the last update from v0.64 to v.0.8.2 (also tested from v0.70 to v.0.82) got a migration error and Entities are no longer available. Deleting device "central" before update dont fix this issue (this also tested)
Also tested remove central+charger devices after 0.8.2 installation but then charger status remain unknown.
All other updates before 0.80 are fine.

Debug log
**Describe the bug**
A clear and concise description of what the bug is.

ssieppi, 2025-03-04:
This is weird.. After upgrade to 0.8.2 -> migration error -> then delete "central" and "charger" then I connect from ABB mobile app to my charger -> now everything works..

erepeo, 2025-03-04:
I don't know which version is safe anymore. I'm not touching anything until something certain is released because after one update it set the charging to 32A and overloaded the inverter... I am grateful for the Author's work but this error was already big because it affects devices at home. Of course I could update when the car was not connected but it didn't even occur to me that something like this could happen. It has never happened.

emm-mag, 2025-03-04:
**Solved this way :**

Still migration error with 8.0.2
I made a whole backup of HA then removed "central" and rebooted HA
I added back ocpp and discovered charger with default settings (except the charger name) and everything seems fine !

emm-mag, 2025-03-04:
32A is fine for my use and seems to work on graphic.
I hope it'll work on next VE charge because I'm using it for dynamic charge !

erepeo, 2025-03-04:
It worked. I'm not sure about the procedure but I think deleting 'central' and 'charger' and restarting HA and adding it again helped.

Thanks again for the work to the Author. This integration is very helpful.

BikeMikeAU, 2025-03-06:
> I'm using it for dynamic charge !

Do not use the max charge current for dynamic charge. It's value is written to storage in the charger and you will potentially wear it out with frequent updates. Use a charge profile instead. https://github.com/lbbrhzn/ocpp/wiki/Charge_automation#dynamically-adjusting-ev-charge-current-a-smart-approach

emm-mag, 2025-03-06:
@BikeMikeAU 
Thanks a lot for this advice !
I've already setup automation based on charge status and current house load to change max current charge. 
I'll use your sample and reuse my calculation.

Side note : My ABB Terra AC should handle dynamic charge by requesting current load to a modbus meter, 
It works but charging stops and restart many times without explanation. I don't know if this may damage the car charger.

ABB support doesn't reply to my emails anymore - some (usefull) settings can't be changed in terra config application

BikeMikeAU, 2025-03-06:
No, charge profile will not be written to flash storage, it is temporary in RAM. However it is dependent on your charger and vehicle how quickly it responds to changes. In my experience, that's about once per minute maximum.

erepeo, 2025-03-06:
OK, I understand. I gave the time "second" as an example of very frequent changes so that we would understand each other. I wanted to be sure that Tx is COMPLETELY safe and I can send any number of messages without worrying about the durability of the device. Thank you for the information.

BikeMikeAU, 2025-03-10:
I made some improvements in my automation and created a separate discussion, since not related to this issue discussion
https://github.com/lbbrhzn/ocpp/discussions/1566

closed 2025-05-16 by github-actions[bot]