lbbrhzn/ocpp #1545: set_charge_rate limit_watts value ignored

molikk, 2025-03-05:
**Describe the bug**
when performing set_charge_rate action with limit_watts:

```
action: ocpp.set_charge_rate
data:
  devid: charger_1
  limit_watts: 2300
```
to charger is send 32 A limits

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Developer Tools -> Actions'
2. Choose: 'ocpp.set_charge_rate'
3. Put to action data (change devid to yours):
```
devid: charger_1
limit_watts: 2300
```
4. Click 'Perform Action'

This should set charging limit to 2300W.
Instead it sets 32A

```
2025-03-05 13:48:32.173 INFO (MainThread) [ocpp] EVB-P21378481: send [2,"13c1b364-d8bf-4219-8660-e943c74f5c6c","GetConfiguration",{"key":["ChargeProfileMaxStackLevel"]}]
2025-03-05 13:48:32.328 INFO (MainThread) [ocpp] EVB-P21378481: receive message [3,"13c1b364-d8bf-4219-8660-e943c74f5c6c",{"configurationKey":[{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"6"}]}]
2025-03-05 13:48:32.331 INFO (MainThread) [ocpp] EVB-P21378481: send [2,"7b8d54f5-c19e-4940-b2ba-2290fedbbca2","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":6,"chargingProfileKind":"Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":32}]}}}]
2025-03-05 13:48:32.383 INFO (MainThread) [ocpp] EVB-P21378481: receive message [3,"7b8d54f5-c19e-4940-b2ba-2290fedbbca2",{"status":"Rejected"}]
2025-03-05 13:48:32.387 INFO (MainThread) [ocpp] EVB-P21378481: send [2,"609ba29d-269e-447e-a1ce-9407e0d89158","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":5,"chargingProfileKind":"Relative","chargingProfilePurpose":"TxDefaultProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":32}]}}}]
2025-03-05 13:48:34.243 INFO (MainThread) [ocpp] EVB-P21378481: receive message [3,"609ba29d-269e-447e-a1ce-9407e0d89158",{"status":"Accepted"}]
```

drc38, 2025-03-05:
In the log you should see:
"Charger supports setting the following units: " 
If your charger does not report it supports setting W then you need to use limit_amps instead.

molikk, 2025-03-05:
@drc38 
I don't see such entry in my logs but
ocpp_key: ChargingScheduleAllowedChargingRateUnit returns value: Current,Power

drc38, 2025-03-06:
Enable debugging and provide the full log from when you send the action, it should provide more detail on what's happening.

molikk, 2025-03-07:
@drc38 

I think the issue might be with that: 
[custom_components.ocpp] Charger supports setting the following units: Current,Power
[custom_components.ocpp] If more than one unit supported default unit is **Amps**

Full log:

```
2025-03-07 11:57:40.959 INFO (MainThread) [homeassistant.helpers.script.websocket_api_script] websocket_api script: Running websocket_api script
2025-03-07 11:57:40.960 INFO (MainThread) [homeassistant.helpers.script.websocket_api_script] websocket_api script: Executing step call service
2025-03-07 11:57:40.962 INFO (MainThread) [ocpp] EVB-P21378481: send [2,"1e48d783-1780-4175-b3ea-49bc8c6ed32c","GetConfiguration",{"key":["ChargingScheduleAllowedChargingRateUnit"]}]
2025-03-07 11:57:40.962 DEBUG (MainThread) [websockets.server] > TEXT '[2,"1e48d783-1780-4175-b3ea-49bc8c6ed32c","GetC...wedChargingRateUnit"]}]' [113 bytes]
2025-03-07 11:57:41.036 DEBUG (MainThread) [websockets.server] < TEXT '[3,"1e48d783-1780-4175-b3ea-49bc8c6ed32c",{"con...ue":"Current,Power"}]}]' [155 bytes]
2025-03-07 11:57:41.036 INFO (MainThread) [ocpp] EVB-P21378481: receive message [3,"1e48d783-1780-4175-b3ea-49bc8c6ed32c",{"configurationKey":[{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current,Power"}]}]
2025-03-07 11:57:41.038 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargingScheduleAllowedChargingRateUnit: Current,Power
2025-03-07 11:57:41.038 INFO (MainThread) [custom_components.ocpp] Charger supports setting the following units: Current,Power
2025-03-07 11:57:41.038 INFO (MainThread) [custom_components.ocpp] If more than one unit supported default unit is Amps
2025-03-07 11:57:41.040 INFO (MainThread) [ocpp] EVB-P21378481: send [2,"f637db8b-e941-4652-b5df-89b2a640e2a5","GetConfiguration",{"key":["ChargeProfileMaxStackLevel"]}]
2025-03-07 11:57:41.041 DEBUG (MainThread) [websockets.server] > TEXT '[2,"f637db8b-e941-4652-b5df-89b2a640e2a5","GetC...rofileMaxStackLevel"]}]' [100 bytes]
2025-03-07 11:57:41.183 DEBUG (MainThread) [websockets.server] < TEXT '[3,"f637db8b-e941-4652-b5df-89b2a640e2a5",{"con...y":true,"value":"6"}]}]' [130 bytes]
2025-03-07 11:57:41.183 INFO (MainThread) [ocpp] EVB-P21378481: receive message [3,"f637db8b-e941-4652-b5df-89b2a640e2a5",{"configurationKey":[{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"6"}]}]
2025-03-07 11:57:41.186 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargeProfileMaxStackLevel: 6
2025-03-07 11:57:41.188 INFO (MainThread) [ocpp] EVB-P21378481: send [2,"123a5359-9702-4996-a71b-b34833a7917b","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":6,"chargingProfileKind":"Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":32}]}}}]
2025-03-07 11:57:41.188 DEBUG (MainThread) [websockets.server] > TEXT '[2,"123a5359-9702-4996-a71b-b34833a7917b","SetC...iod":0,"limit":32}]}}}]' [323 bytes]
2025-03-07 11:57:41.238 DEBUG (MainThread) [websockets.server] < TEXT '[3,"123a5359-9702-4996-a71b-b34833a7917b",{"status":"Rejected"}]' [64 bytes]
2025-03-07 11:57:41.240 INFO (MainThread) [ocpp] EVB-P21378481: receive message [3,"123a5359-9702-4996-a71b-b34833a7917b",{"status":"Rejected"}]
2025-03-07 11:57:41.244 DEBUG (MainThread) [custom_components.ocpp] ChargePointMaxProfile is not supported by this charger, trying TxDefaultProfile instead...
2025-03-07 11:57:41.250 INFO (MainThread) [ocpp] EVB-P21378481: send [2,"917a2cdf-425a-4c14-91d7-2f73218f59bf","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":5,"chargingProfileKind":"Relative","chargingProfilePurpose":"TxDefaultProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":32}]}}}]
2025-03-07 11:57:41.250 DEBUG (MainThread) [websockets.server] > TEXT '[2,"917a2cdf-425a-4c14-91d7-2f73218f59bf","SetC...iod":0,"limit":32}]}}}]' [318 bytes]
2025-03-07 11:57:43.032 DEBUG (MainThread) [websockets.server] > PING c0 00 31 e1 [binary, 4 bytes]
2025-03-07 11:57:43.060 DEBUG (MainThread) [websockets.server] < PONG c0 00 31 e1 [binary, 4 bytes]
2025-03-07 11:57:43.060 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'EVB-P21379157': ping=1.0 ms, pong=27.0 ms
2025-03-07 11:57:43.093 DEBUG (MainThread) [websockets.server] < TEXT '[3,"917a2cdf-425a-4c14-91d7-2f73218f59bf",{"status":"Accepted"}]' [64 bytes]
2025-03-07 11:57:43.093 INFO (MainThread) [ocpp] EVB-P21378481: receive message [3,"917a2cdf-425a-4c14-91d7-2f73218f59bf",{"status":"Accepted"}]
```

drc38, 2025-03-07:
The logic can be improved to accommodate both by setting the default values to -1 and checking each value if >=0 and unit supported, if neither then exit with warning.

closed 2025-05-14 by github-actions[bot]