lbbrhzn/ocpp #1546: Update from 0.8.1 to 0.8.2 Home Assistant won’t restart.

BitsLeftOver, 2025-03-05:
**Describe the bug**
Upgrading from 0.8.1 to 0.8.2, home assistant server no longer boots after restart.
No further information available. 

Steps:
Receive notice that 0.8.2 is available
Click update
Click restart
Dead haas.

Intel NUC i5, 1TB SSD
HomeAssistant OS

BJReplay, 2025-03-06:
I had the same problem as @miqm and the same solution - removed entries with `ocpp` in them.

SimoneBWS, 2025-03-06:
I had the same problem!

literalaids, 2025-03-09:
I'm having this problem now. But being not very techliterate I'm having trouble getting into the actual json file whilst the instance isnt running since it fails to boot. Can anyone point me in the right direction please?

closed 2025-05-16 by github-actions[bot]