lbbrhzn/ocpp #1548: Missing state_class, device_class and unit_of_measurement for Temperature sensor

dreimer1986, 2025-03-05:
## Version of the custom_component
0.8.2

## Configuration
Working fine config. Selected auto detection of allowed measurands. Rest is default.

```yaml
Add your logs here.
```

## Describe the bug

Well, I keep things clean a bit as we don't really need any logs or configs here. I just checked my temperature sensors as I have many custom ones and one was missing these by accident:

```
state_class: measurement
device_class: temperature
```

I thought maybe I missed these on another one, too. In the end it was just one I forgot to add these lines, but then I saw that the temperature sensor coming from my EVBox ELVI was completely unaware about what it is at all, even the unit_of_measurement was missing. For now I fixed it on my side with the help of the good ol' customize.yaml by adding:

```
sensor.evbox_temperature
  state_class: measurement
  device_class: temperature
  unit_of_measurement: °C
```
But maybe these should find it's way into the integration to be provided by default?

## Debug log

```text

Add your logs here.

```

closed 2025-03-07 by lbbrhzn

smoki3, 2025-03-07:
Temperatur still has no unit for me

smoki3, 2025-03-07:
```
Logger: homeassistant.components.sensor
Quelle: components/sensor/__init__.py:709
Integration: Sensor (Dokumentation, Probleme)
Erstmals aufgetreten: 13:43:50 (2 Vorkommnisse)
Zuletzt protokolliert: 13:48:54

Entity sensor.charger_temperature (<class 'custom_components.ocpp.sensor.ChargePointMetric'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('temperature') it is using; expected one of ['K', '°F', '°C']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/lbbrhzn/ocpp/issues
```