lbbrhzn/ocpp #1550: No Values since 0.8.xx

smoki3, 2025-03-06:
**Describe the bug**
I upgraded from 0.70 to 0.8.x

Since the upgrade I am not able to get all data from my Huawei Charger.

0.7.0

[image]

0.8.0

[image]

I downgraded back and its working again.

ArtjomE, 2025-03-06:
i have same issue by Vestel EVC04 after updating from 0.7.x to 0.8.2

[image]

dreimer1986, 2025-03-06:
Do you see a warning or error under Devices on the OCPP entry? If so you likely have the same problem as many others and must remove the adding completely and reinstall it cleanly.

smoki3, 2025-03-06:
> Do you see a warning or error under Devices on the OCPP entry? If so you likely have the same problem as many others and must remove the adding completely and reinstall it cleanly.

You mean removing the charger and readd it? Or you mean to uninstall the integration and reinstall ist? 

I already delete my charger and readded it a few times. This doesn't help.

dreimer1986, 2025-03-06:
> You mean removing the charger and readd it? Or you mean to uninstall the integration and reinstall ist?
>
> I already delete my charger and readded it a few times. This doesn't help.

Deleting means that you have no Open Charge Point Protocol (OCPP) entry anymore? Did you reboot after removing?

ArtjomE, 2025-03-06:
I have uninstalled and installed OCPP. Restarted the wallbox. I can still only see some of the values.

[image]

[image]

dreimer1986, 2025-03-06:
As long as you see values on a few entities, everything works as expected. Ignore the values not working and after a while these will be shown as not provided by integration and can be removed. This is what the automatic detection does. None of the wall boxes support OCPP 100% as no one of the vendors is able to read a simple standard documentation. So there is not big in the behavior.

ArtjomE, 2025-03-06:
It seems that after the update some entities no longer work and others are called differently.

smoki3, 2025-03-06:
I don't use automatic detection. I select them manually. And ob 0.7 they receive value on 0.8 not

smoki3, 2025-03-06:
So what I did now:

- Removed my device
- Removed the integration in HACS
- Reboot HA
- Installed Integration 0.8.2 in HACS
- Reboot HA
- Added my Huawei Charger

--> Result: Not working. Still missing the most values.

- Removed my device
- removed integration in HACS
- Reboot HA
- Installed Integration 0.7.0
- Reboot HA
- Added my  Huawei Charger

--> Result: All values are available

Looks excatly like on my first screenshot!

smoki3, 2025-03-06:
My logs:
```
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.ocpp
Quelle: custom_components/ocpp/api.py:237
Integration: Open Charge Point Protocol (OCPP) (Dokumentation, Probleme)
Erstmals aufgetreten: 13:45:14 (23 Vorkommnisse)
Zuletzt protokolliert: 13:45:59

Failed to setup charger HW-123: ChargerSystemSettings.__init__() missing 1 required positional argument: 'monitored_variables'

```

smoki3, 2025-03-06:
An I get one 

```
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.ocpp
Quelle: custom_components/ocpp/ocppv16.py:206
Integration: Open Charge Point Protocol (OCPP) (Dokumentation, Probleme)
Erstmals aufgetreten: 13:46:11 (2 Vorkommnisse)
Zuletzt protokolliert: 14:00:26

Failed with response: Rejected
```

smoki3, 2025-03-06:
So I found the bug now:

As soon you choose "charger" as name, everthing is working es expected

[image]

As soon I choose a different name its not working!

dreimer1986, 2025-03-06:
Interesting. Worked fine here... I named it evbox from the very beginning

smoki3, 2025-03-06:
is it maybe a "space"?

ArtjomE, 2025-03-06:
> So, ich habe den Fehler jetzt gefunden:
> 
> Sobald Sie im Konfigurationsablauf "Ladegerät" als Namen auswählen, funktioniert alles wie erwartet
> 
> [image]
> 
> Sobald ich einen anderen Namen wähle, funktioniert es nicht!
> 
> Ich kann das Ladegerät später umbenennen, dann funktioniert es noch

Yes, that's it. I had installed BackUp, earlier I had ‘charger’ as name and now ‘vestel_evc04’

kfussmann, 2025-03-11:
Hello @smoki3 ,
I have in my setup also Huawei Fusion Solar Charger, but two of them. I am also having difficulties in getting them detected. Could you please tell me how exactly you defined which setting at startup to get them detected by the OCPP Server? I then would like to try to first detect one of the two and after that even try to get both of them detected. That would be extremly helpfull of yours. Thanks a lot.

kr0ner, 2025-04-04:
> So I found the bug now:
> 
> As soon you choose "charger" as name in the config flow, everthing is working es expected
> 
> [image]
> 
> As soon I choose a different name its not working!
> 
> I can rename the charger later, than its still working

How did you manage to get this to work with the latest release? I still had to downgrade to 0.7. to get any values :-/