lbbrhzn/ocpp #1558: Charger being lost and rediscovered approximately daily

Mark-Hetherington, 2025-03-07:
**Describe the bug**
After upgrading to 0.8.2, our EV Charger was rediscovered with home assistant displaying a notification that devices had been discovered. After clicking the link in the notification, and using the same name ("charger") the device reappears with all automations still intact.

**To Reproduce**
Steps to reproduce the behavior:
1. Set up charger and automations on 0.7.x
2. Upgrade to 0.8.2 (maybe with some tinkering to resolve issues)
3. Rediscover and set up charger
4. Login after 24 hours to see charger is missing, and "new device discovered"

**Expected behavior**
Charger remains set up and is not rediscovered.

**Desktop (please complete the following information):**
 - OS: HA running in docker on Debian bookwork. Accessed from either Linux or Windows desktop
 - Browser Chrome
 - Version 134

I would appreciate guidance on debugging the issue or how to collect relevant logging to further diagnose this.

mikael-andersson, 2025-03-07:
I get more or less the same problem but the connection goes down and the charger is rediscovered immediately after a HA Core restart.

*To Reproduce**
Steps to reproduce the behavior:
1. Setup OCPP integration with default settings
2. Connection to charger (ABB Terra) is established
3. Restart HA
4. OCPP integration does not get any values from the charger although it is in the list of OCPP devices
5. OCPP says it has discovered a new device.
6. After having added the charger again, the connection to it is established. Although the device is already in the list I do not get a second device, but it seems to replace the existing one.

**Desktop (please complete the following information):**
 - OS: Ubuntu 24.04.2 LTS
 - brave
 - [Version 1.75.178 Chromium: 133.0.6943.98 (Official Build) (64-bit)](https://brave.com/latest/)

**Additional context**

I get the same problem with both OCPP version 0.8.1 and 0.8.2. I have not tried any others.

mikael-andersson, 2025-03-07:
I have now tried with version 0.7.0 and it seems to works fine.

drc38, 2025-03-07:
There may be some issue with the config entry being saved by HA to the hard drive. The entry is used to recognise the charger and which sensors to show. Check that the ocpp details are in the config entries file after the discovery step and connection of charger, then reboot and see what's there before going through the discovery. If there is a record post the detail.

If there is no ocpp record in either case enable debugging before going through discovery and post the log.

mikael-andersson, 2025-03-07:
Here is the record in core.config_entries after "after the discovery step and connection of charger":

```
      {
        "created_at": "2025-03-07T16:55:12.500671+00:00",
        "data": {
          "cpids": [],
          "csid": "central",
          "host": "0.0.0.0",
          "port": 9000,
          "ssl": false,
          "ssl_certfile_path": "/config/fullchain.pem",
          "ssl_keyfile_path": "/config/privkey.pem",
          "websocket_close_timeout": 10,
          "websocket_ping_interval": 20,
          "websocket_ping_timeout": 20,
          "websocket_ping_tries": 2
        },
        "disabled_by": null,
        "discovery_keys": {},
        "domain": "ocpp",
        "entry_id": "01JNRRGJVM1B51F21X0PBCJRF0",
        "minor_version": 0,
        "modified_at": "2025-03-07T16:55:12.500675+00:00",
        "options": {},
        "pref_disable_new_entities": false,
        "pref_disable_polling": false,
        "source": "user",
        "subentries": [],
        "title": "central",
        "unique_id": null,
        "version": 2
      }

```

Before restarting HA Core, I do:

cp core.config_entries core.config_entries.bak

And after HA has restarted the files are still identical, but the OCPP still asks me to add the charger again. I do not understand this.

drc38, 2025-03-07:
"cpids": [] should not be empty. After the discovery step it should be populated with the details entered for the charger. Are you manually setting the measurands?

Mark-Hetherington, 2025-03-09:
> I get more or less the same problem but the connection goes down and the charger is rediscovered immediately after a HA Core restart.

I have had a look and realised that yes, indeed, even with the notification of a discovered device, the existing device is still shown within the OCPP plugin, but with values unavailable. Also the prompt of device discovery happens after a restart - I'm restarting home assistant daily to take a backup.

Following the information reported by @mikael-andersson, I had a look for core.config_entries to see what it contains and similarly cpids is empty.
`{"created_at":"2025-03-04T02:47:32.910349+00:00","data":{"cpids":[],"csid":"central","host":"0.0.0.0","port":9000,"ssl":false,"ssl_certfile_path":"/config/fullchain.pem","ssl_keyfile_path":"/config/privkey.pem","websocket_close_timeout":10,"websocket_ping_interval":20,"websocket_ping_timeout":20,"websocket_ping_tries":2},"disabled_by":null,"discovery_keys":{},"domain":"ocpp","entry_id":"01JNFGTA9E03JWT10M9DN6FE23","minor_version":0,"modified_at":"2025-03-04T02:47:32.910354+00:00","options":{},"pref_disable_new_entities":false,"pref_disable_polling":false,"source":"user","title":"central","unique_id":null,"version":2}
`

Running through the discovery I see the following logging:
`2025-03-09 14:04:32.739 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
2025-03-09 14:04:32.739 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/HomeCharger
2025-03-09 14:04:32.739 INFO (MainThread) [custom_components.ocpp] Charger match found for charger:HomeCharger
2025-03-09 14:04:32.739 DEBUG (MainThread) [custom_components.ocpp] Central settings: CentralSystemSettings(csid='central', host='0.0.0.0', port=9000, ssl=False, ssl_certfile_path='/config/fullchain.pem', ssl_keyfile_path='/config/privkey.pem', websocket_close_timeout=10, websocket_ping_interval=20, websocket_ping_timeout=20, websocket_ping_tries=2, cpids=[{'HomeCharger': {'cpid': 'charger', 'max_current': 32, 'monitored_variables_autoconfig': True, 'meter_interval': 60, 'idle_interval': 900, 'skip_schema_validation': False, 'force_smart_charging': False, 'monitored_variables': 'Current.Export,Current.Import,Current.Offered,Energy.Active.Export.Interval,Energy.Active.Export.Register,Energy.Active.Import.Interval,Energy.Active.Import.Register,Energy.Reactive.Export.Interval,Energy.Reactive.Export.Register,Energy.Reactive.Import.Interval,Energy.Reactive.Import.Register,Frequency,Power.Active.Export,Power.Active.Import,Power.Factor,Power.Offered,Power.Reactive.Export,Power.Reactive.Import,RPM,SoC,Temperature,Voltage'}}], subprotocols=['ocpp1.6', 'ocpp2.0.1'])
2025-03-09 14:04:32.742 INFO (MainThread) [custom_components.ocpp] Charger charger:HomeCharger connected to 0.0.0.0:9000.
2025-03-09 14:04:32.742 INFO (MainThread) [custom_components.ocpp] 1 charger(s): {'charger': 'HomeCharger'} now connected to central system:central.
2025-03-09 14:04:32.873 INFO (MainThread) [ocpp] HomeCharger: receive message [2,"hj1rd1zxrc4rr28lu1l55xhm","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2025-03-09T03:04:32Z"}]
2025-03-09 14:04:32.887 INFO (MainThread) [ocpp] HomeCharger: send [3,"hj1rd1zxrc4rr28lu1l55xhm",{}]
2025-03-09 14:04:33.255 INFO (MainThread) [ocpp] HomeCharger: receive message [2,"7qjy87c1rlyxisqenia0wjsm","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Charging","timestamp":"2025-03-09T03:04:32Z"}]
`

From that, I assume (without looking at the code) that we should see the cpid data above saved into the core.config_entries?

Mark-Hetherington, 2025-03-09:
As a further test I manually edited the core.config_entities to include the data from the discovery. I can now restart Home Assistant and the charger has available data and is not rediscovered.

drc38, 2025-03-09:
> As a further test I manually edited the core.config_entities to include the data from the discovery. I can now restart Home Assistant and the charger has available data and is not rediscovered.

That will fix the issue, however it's still unclear why HA has not saved the entry to disk though.

Mark-Hetherington, 2025-03-09:
I added some logging to `async_step_cp_user.()` At the point where it calls `self.async_update_reload_and_abort` , the `self._data` is set as would be expected:
`{'cpids': [{'HomeCharger': {'cpid': 'charger', 'max_current': 32, 'monitored_variables_autoconfig': True, 'meter_interval': 60, 'idle_interval': 900, 'skip_schema_validation': False, 'force_smart_charging': False, 'monitored_variables': 'Current.Export,Current.Import,Current.Offered,Energy.Active.Export.Interval,Energy.Active.Export.Register,Energy.Active.Import.Interval,Energy.Active.Import.Register,Energy.Reactive.Export.Interval,Energy.Reactive.Export.Register,Energy.Reactive.Import.Interval,Energy.Reactive.Import.Register,Frequency,Power.Active.Export,Power.Active.Import,Power.Factor,Power.Offered,Power.Reactive.Export,Power.Reactive.Import,RPM,SoC,Temperature,Voltage'}}], 'csid': 'central', 'host': '0.0.0.0', 'port': 9000, 'ssl': False, 'ssl_certfile_path': '/config/fullchain.pem', 'ssl_keyfile_path': '/config/privkey.pem', 'websocket_close_timeout': 10, 'websocket_ping_interval': 20, 'websocket_ping_timeout': 20, 'websocket_ping_tries': 2}`

It is still unclear why HA has not saved the entry to disk, but I am pretty sure the issue is within HA rather than this integration.

Mark-Hetherington, 2025-03-09:
Possibly related: https://community.home-assistant.io/t/ha-not-saving-config-entries-every-time-i-reboot/816903/2

mikael-andersson, 2025-03-09:
Sorry for my slow response, I have not been online much.

@Mark-Hetherington, thanks for testing.

> It is still unclear why HA has not saved the entry to disk, but I am pretty sure the issue is within HA rather than this integration.

Is it not strange that this problem does not occur in OCPP version 0.7.0 then? The JSON format looks however different between version 0.7.0 and version 0.8.2.

mikael-andersson, 2025-03-09:
> "cpids": [] should not be empty. After the discovery step it should be populated with the details entered for the charger. Are you manually setting the measurands?

I have only used the default settings where _Automatic detection of OCPP Measurands_ is checked.

> Enable debug, do the discovery step, then post the log.

@drc38 , I suppose it suffice with the log from @Mark-Hetherington?

Mark-Hetherington, 2025-03-09:
That is a good point. The data format has changed as the integration now supports communications with more than one charger. However, I'll have to look closer to see if there is some other subtle difference in the way the home assistant API is called. 

One thing I stumbled across was that if the settings dictionary is mutated, but is still the same dictionary the changes might not be detected. (https://github.com/home-assistant/core/issues/90078). So I plan to test adding a deepcopy of the settings before calling the update.

drc38, 2025-03-09:
@Mark-Hetherington good find with the discussion on deepcopy. Let us know how you get on and whether it fixes the issue for you.

@mikael-andersson the HA api calls to save the entry are different between 0.7 and >0.8

callifo, 2025-03-10:
I've got the same issue with my MG Chargehub, updated to 0.8.3 and noted it needs to be rediscovered every reboot. Previous updates from 0.7.0 to 0.8.2 required a complete delete of the charger to get it back in.

Mark-Hetherington, 2025-03-10:
Unfortunately taking a deepcopy of the `self._data` to pass to `async_update_reload_and_abort()` has not resolved the problem.

drc38, 2025-03-10:
Try adding hass.config_entries._async_schedule_save()

Mark-Hetherington, 2025-03-10:
Also I debugged into `async_update_reload_and_abort()` enough to see that no change is detected because the config entry already has the cpids. 
So at the point the config flow calls `async_update_reload_and_abort()`, the `self._entry.data` already has the cpids that are in self._data.

Mark-Hetherington, 2025-03-10:
I have found that `self._entry['cpids'] is self._data['cpids'] == True`. Meaning that when `self._data` is updated with the results of the user config, so is `self._entry`, breaking change detection. Although I'm not sure why the deep copy didn't resolve that.

drc38, 2025-03-10:
Where did you do the deepcopy? It should replace
`self._data = {**self._entry.data}` where I had assumed unpacking would do a copy but it looks like it doesn't based on your testing.

Mark-Hetherington, 2025-03-10:
I realised I initially did the deepcopy in a silly spot - after the data had already been updated (in both references). So then I tried replacing the unpacking (`self._data = {**self._entry.data}`). However when that is replaced, the charger gets automatically created without the user setup step. I don't know if it potentially triggers an abort in `_abort_if_unique_id_configured`.

I have realised the best place for it is just before it is updated from user input. 
`self._data = deepcopy(self._data)
 self._data[CONF_CPIDS].append({self._cp_id: user_input})`

This works, and the data is saved. I think it would be good to understand why replacing the unpacking doesn't work.

Mark-Hetherington, 2025-03-10:
From memory I originally replaced the unpack with:
`        self._data = deepcopy(self._entry.data)`
This did not correctly run the config flow.

I've also now tested replacing the unpack with:
`        self._data = {**self._entry.data}
        self._data = deepcopy(self._data)`
This appears to work correctly, both in terms of the flow and the saving of the config entries.

PR #1571 (2025-03-10): Prevent reference sharing between entry and data
Change detection doesn't work when updating the self._data also updates self._entry.data because the reference to the cpid key is the same list. This prevents the data being persisted to disk.

Closes #1558

## Summary by CodeRabbit

- **New Features**
  - Enhanced the configuration process for improved data integrity, ensuring changes are applied without unintended side effects.
- **Bug Fixes**
  - Minor formatting adjustments made for better readability in user input handling.
closed 2025-03-10 by drc38