lbbrhzn/ocpp #1561: Slider does not fit to defined max current

smoki3, 2025-03-08:
I have the Problem that the max current slider is always able to set to 32A. Even if I define in the config flow just 16A. I expecting that it's then limited to 16.

Also if my charger only supports Watts as unit. The slider is always Ampere

smoki3, 2025-03-08:
By the way it was working on 0.7.0

nomad64, 2025-03-09:
I have this same issue. My Grizzl-E Ultimate is set to a max of 48A. When I setup its connection to the OCPP integration, I entered "48" as the max charging amps. After being successfully added to HA, the slider maxes out at 32A.

Interestingly, the slider doesn't seem to have any effect. Changing its value doesn't change the value on the charger. Also, changing the value in the charger isn't reflected in the slider.

For now, I just disabled it since I don't need to adjust the charging amps.

PR #1563 (2025-03-09): fix slider max value
fixes #1561

## Summary by CodeRabbit

- **Refactor**
	- Streamlined how charging current settings are applied by sourcing values directly from the charger configuration, ensuring more accurate and robust performance.
closed 2025-03-09 by lbbrhzn