lbbrhzn/ocpp #1565: SetChargingProfile issue

npirtube, 2025-03-09:
**Describe the bug**
When I try to change the intensity of my charger it sends an error due to the type of profile sent.

**To Reproduce**
Try to change the maximum amperage value via the web interface.

**Expected behavior**
You should send an “Absolute” profile instead of “Relative”.
Also, it would be simpler to send a message by changing the key instead of a SetChargingProfile

**Screenshots**

[image]

`2025-03-09 20:11:37.829 INFO (MainThread) [ocpp] Plaza_1: send [2,"9706e144-0ceb-4349-8ed6-e83888e0d4c0","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":1,"chargingProfileKind":"Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"W","chargingSchedulePeriod":[{"startPeriod":0,"limit":22000}]}}}]
2025-03-09 20:11:38.087 INFO (MainThread) [ocpp] Plaza_1: receive message [4,"9706e144-0ceb-4349-8ed6-e83888e0d4c0","PropertyConstraintViolation","When chargingProfilePurpose =ChargingStationMaxProfile, chargingProfileKind SHALL NOT be Relative",{}]
2025-03-09 20:11:38.088 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140528426986384] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2837, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 122, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/ocpp/number.py", line 131, in async_set_native_value
    resp = await self.central_system.set_max_charge_rate_amps(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self.cpid, num_value
        ^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/ocpp/api.py", line 345, in set_max_charge_rate_amps
    return await self.charge_points[cp_id].set_charge_rate(limit_amps=value)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/ocppv16.py", line 323, in set_charge_rate
    resp = await self.call(req)
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 421, in call
    response = await self._get_specific_response(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        call.unique_id, self._response_timeout
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/ocpp/chargepoint.py", line 341, in _get_specific_response
    raise resp.to_exception()
ocpp.exceptions.PropertyConstraintViolationError: PropertyConstraintViolationError: When chargingProfilePurpose =ChargingStationMaxProfile, chargingProfileKind SHALL NOT be Relative,  {}`

**Additional context**
When using the slider to select the maximum amperage it would be best to send the following command.
`action: ocpp.configure
metadata: {}
data:
  devid: number.plaza_120_maximum_current
  ocpp_key: MaxChargeCurrentLimit
  value: "16"
response_variable: OCPP_Response`

`2025-03-09 20:43:59.891 INFO (MainThread) [ocpp] Plaza_1: send [2,"11d019a5-00de-4cf2-8048-dd1fddc6007c","GetConfiguration",{"key":["MaxChargeCurrentLimit"]}]
2025-03-09 20:44:01.661 INFO (MainThread) [ocpp] Plaza_1: receive message [3,"11d019a5-00de-4cf2-8048-dd1fddc6007c",{"configurationKey":[{"key":"MaxChargeCurrentLimit","readonly":false,"value":"16"}]}]`

Would it be possible to implement both modifications?

flashg1, 2025-03-10:
https://github.com/lbbrhzn/ocpp/blob/main/docs/Charge_automation.md

rbb-dev, 2025-03-10:
1 - ChargingStationMaxProfile does not forbid Relative in 1.6 spec.
      You can check it yourself: [](https://openchargealliance.org/my-oca/ocpp)
      It's perfectly valid.
      Having said that, you can create your automation that will send the correct message for your charger using set_charge_rate
      see here [](https://github.com/lbbrhzn/ocpp/discussions/1566)
2 - It looks like MaxChargeCurrentLimit is specific to MaxiCharger chargers only, so it most likely won't be implemented.
      This can also be solved with the correct automation.

npirtube, 2025-03-14:
I am testing and have managed to send the message to the charger. The problem I have now is that I have two chargers and I can't send the message to the second charger

This is the message I am sending

`action: ocpp.set_charge_rate
metadata: {}
data:
  devid: sensor.plaza_120_id
  custom_profile: |
    {
      "chargingProfileId": 8,
      "stackLevel": 0,
      "chargingProfilePurpose": "ChargePointMaxProfile",
      "chargingProfileKind": "Absolute",
      "chargingSchedule": {
        "chargingRateUnit": "A",
        "chargingSchedulePeriod": [
          {"startPeriod": 0, "limit": {{ states('input_number.amperaje_plaza_1') | int }}}
        ],
        "startSchedule": "{{ now().strftime('%Y-%m-%dT%H:%M:%SZ') }}"
      }
    }
  conn_id: 0`

closed 2025-05-21 by github-actions[bot]