lbbrhzn/ocpp #1572: Upgrading from 7.0 to 0.8.4 still fails with Migration error

Spangsberg, 2025-03-11:
[image]

Aekung, 2025-03-11:
Me too. I'm still usgin 0.7.0 because I can't get pass to version 0.8 without migration error.

TAGVL, 2025-03-11:
I have similar issues since updating from 0.7.0 to whatever version newer. I cannot get my integration to start working again, I downgrade via HACS, I removed the complete integration from HACS and added again, added the central and charger multiple times with different settings again in 0.7.0 but nothing seems to work to get the integration up and running again. My Wallbox just doesn't connect to the OCPP anymore, my charger which was preivously labled as Brand Wallbox now keeps all information labled as Unknown. Hopefully there will be a proper fix or documentation available soon to get this solved. For now I use OCPP enabled to 'lock' the charger and once charging needs to happen I disable OCPP on the charger itself but this isn't ideal since I don't have direct access to my Wallbox 24/7 and need to be within bluetooth range to change this

stevemcan, 2025-03-12:
I too am having issues upgrading past 0.7.0, just get a migration error.

d-l-h, 2025-03-13:
same for me. I even had some strange DNS issues in HA after upgrade. Had to got back to 0.7

taste66, 2025-03-13:
I had the same problem but resolved it in the following way:
1. removed the full OCPP integration
2. Added it again and created the central
3. Waited until the Charger was discovered
4. Added the charger from the discovery
5. Rebooted the full machine
6. Waited 5 minutes after it came back online

after #5 it was up and running again with the same entities

TAGVL, 2025-03-14:
> I had the same problem but resolved it in the following way:
>
> 1. removed the full OCPP integration
> 2. Added it again and created the central
> 3. Waited until the Charger was discovered
> 4. Added the [...]

Do you have to remove the wallbox integration (factory integration of your wallbox supplier) itself also for this and re-add it once discovered?

taste66, 2025-03-14:
> Do you have to remove the wallbox integration (factory integration of your wallbox supplier) itself also for this and re-add it once discovered?

No I left the configuration of the Wall-Box as is and I guess beacuse it will connect to the server it is discoverred again.
I would not expect th ereboot of the server is really needed but since I was struggeling for some time I thought lets do this as clean as possible

Speederer, 2025-03-14:
> I had the same problem but resolved it in the following way:
>
> 1. removed the full OCPP integration
> 2. Added it again and created the central
> 3. Waited until the Charger was discovered
> 4. Added the [...]

I have successfully install the new version (0.8.4) with these steps:
1. removed the full OCPP integration
2. removed OCPP (0.7.0) from HACS
3. installed OCPP (0.8.4) from HACS
4. added OCPP as a new integration (everything was default, except the max charging current (16A))
5. after 1 minute everything came back

So restart wasn't needed. And maybe it is worth to mention that previously I updated from 0.7.0 to 0.8.4 with the default update process and it failed. When I noticed I remove the old non-working version and force to install the 0.7.0 and it went fine.

TAGVL, 2025-03-14:
For me this also worked. Maybe also due to my wallbox was not connected to wifi anymore which makes communication difficult ;) strange thing now is that it is constanly rebooting according to OCPP in HomeAssistant. I thought I read more about this but can't seem to find it why this is happening. Maybe I'll downgrade to 0.7.0 again instead of 0.8.4 for now

Christian904e, 2025-03-15:
I have the same problem and several utility meters connected to my „charger“ entity.
Are they affected by deleting the integration? Or just the central entity?

Do you think that this problem here will be fixed, or is removing and adding the solution?

taste66, 2025-03-15:
@Christian904e if you use the same name for your charger they will return. I used a different name than the default charger but that still worked

wjcloudy, 2025-03-16:
Same issue here (2 wallbox chargers) is there any useful info we can provide to help debug this instead of a reinstall ? I fear issues will keep coming as people upgrade otherwise...

drc38, 2025-03-17:
You can post the debugs, but it is unlikely if you are still getting the migration error it can be addressed in the codebase. You're best to carry out a clean install. Provided you use the same name for the charger on setup and there's no underlying issues with the charger's ocpp implementation the clean install should allow you to upgrade >=0.8.4, take a backup beforehand and if there's a problem you can easily revert to 0.7.0 if needed.

Aekung, 2025-03-17:
> I have successfully install the new version (0.8.4) with these steps:
>
> 1. removed the full OCPP integration
> 2. removed OCPP (0.7.0) from HACS
> 3. installed OCPP (0.8.4) from HACS
> 4. added OCPP as a [...]

Got 0.8.4 working after following the process and restarting the service.

Thank you very much.

matt05d, 2025-03-19:
For those who've had success with this, when you configure central, does it also add the charger device straight away or does that automatically happen later? 

If I reinstall 0.7.0, the charger device is added together with the central device by default with all the previously found entities (but the entities are all unknown). This happens at the end of the configuration flow when I get asked to assign an area to central and charger.

When I follow the steps here about removing the integration completely and doing a fresh install of 0.8.4, central is added but no charger device appears so I am only asked to assign an area to central and then nothing else happens.