lbbrhzn/ocpp #1573: Rejected currentTime message might be interpreted connector count.

BJReplay, 2025-03-11:
## Version of the custom_component

v.0.8.4

<details>
<summary>Configuration</summary>

```yaml
# Loads default set of integrations. Do not remove.
default_config:

# Load frontend themes from the themes folder
frontend:
  themes: !include_dir_merge_named themes

automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml

recorder:
  exclude:
    entities:
      # Disable recorder making entries for water usage
      # It causes miscalculations with the retrospective imports
      - sensor.water_usage_mains

http:
    use_x_forwarded_for: true
    trusted_proxies:
      - 172.21.0.10
      - 192.168.1.150

logger:
  default: warning
  logs:
     custom_components.solcast_solar: debug
     custom_components.epa_victoria_air_quality: debug
     ocpp: debug
     #ocpp: warning
```
</details>

## Describe the bug

I've noticed a warning notification each time Home Assistant starts, as well as after updates to the OCPP integration.

I was running debug logging to determine if the charger would accept a TxDefaultProfile charging schedule (it did), and whether it would accept Watts as well as Amps (it did) so captured the logs from startup of HA until the notification appeared.

Near the bottom of the log (highlighted in the picture below) are the slightly suspect lines.

It looks like the OCPP central station is attempting to set the time of the charger and this is being rejected, but it also looks like the OCPP integration is then interpreting this as the charger has one connection (which it does).

I have raised this as an issue rather than a bug as it might be just that the sequence of messages in the log is misleading, so I have included the entire log below.

Further below the log is the core profile as captured from the charger's own logs.

[image]

<details>
<summary>Debug log</summary>

```text
17:08:02.954 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
17:08:02.954 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/MySerialNumber
17:08:02.955 INFO (MainThread) [custom_components.ocpp] Charger match found for MySerialNumber:MySerialNumber
17:08:02.955 DEBUG (MainThread) [custom_components.ocpp] Central settings: CentralSystemSettings(csid='central', host='0.0.0.0', port=9000, ssl=True, ssl_certfile_path='/ocppcerts/fullchain.cer', ssl_keyfile_path='/ocppcerts/ocpp.sheppard-family.au.key', websocket_close_timeout=10, websocket_ping_interval=20, websocket_ping_timeout=20, websocket_ping_tries=2, cpids=[{'MySerialNumber': {'cpid': 'MySerialNumber', 'force_smart_charging': False, 'idle_interval': 900, 'max_current': 32, 'meter_interval': 10, 'monitored_variables': 'Current.Export,Current.Import,Current.Offered,Energy.Active.Export.Interval,Energy.Active.Export.Register,Energy.Active.Import.Interval,Energy.Active.Import.Register,Energy.Reactive.Export.Interval,Energy.Reactive.Export.Register,Energy.Reactive.Import.Interval,Energy.Reactive.Import.Register,Frequency,Power.Active.Export,Power.Active.Import,Power.Factor,Power.Offered,Power.Reactive.Export,Power.Reactive.Import,RPM,SoC,Temperature,Voltage', 'monitored_variables_autoconfig': True, 'skip_schema_validation': True}}], subprotocols=['ocpp1.6', 'ocpp2.0.1'])
17:08:02.957 INFO (MainThread) [custom_components.ocpp] Charger MySerialNumber:MySerialNumber connected to 0.0.0.0:9000.
17:08:02.957 INFO (MainThread) [custom_components.ocpp] 1 charger(s): {'MySerialNumber': 'MySerialNumber'} now connected to central system:central.
17:08:03.347 INFO (MainThread) [ocpp] MySerialNumber: receive message [2,"250311060803-40","DataTransfer",{"vendorId":"EN+","messageId":"gatewayInfo","data":"{\"SN\":\"MySerialNumber\",\"fwVer\":\"V43.178.555\",\"projectName\":\"EESP32WIIIHome3_7670_V43\",\"branch\":\"dev_home_v3\",\"gateCode\":109,\"measureType\":1,\"disconnectReason\":0,\"maxCurr\":320,\"phase\":1,\"netType\":1,\"chgMode\":0,\"sdCardStatus\":1,\"sdCardFreeSize\":109352,\"rateCurrent\":32}"}]
17:08:03.348 DEBUG (MainThread) [custom_components.ocpp] Data transfer received from MySerialNumber: {'message_id': 'gatewayInfo', 'data': '{"SN":"MySerialNumber","fwVer":"V43.178.555","projectName":"EESP32WIIIHome3_7670_V43","branch":"dev_home_v3","gateCode":109,"measureType":1,"disconnectReason":0,"maxCurr":320,"phase":1,"netType":1,"chgMode":0,"sdCardStatus":1,"sdCardFreeSize":109352,"rateCurrent":32}'}
[... 354 lines cut ...]
  "smartChargingProfile": {
    "ChargeProfileMaxStackLevel": {
      "readonly": true,
      "value": 5
    },
    "ChargingScheduleAllowedChargingRateUnit": {
      "readonly": true,
      "value": "Current, Power"
    },
    "ChargingScheduleMaxPeriods": {
      "readonly": true,
      "value": 5
    },
    "ConnectorSwitch3to1PhaseSupported": {
      "readonly": true,
      "value": false
    },
    "MaxChargingProfilesInstalled": {
      "readonly": true,
      "value": 3
    }
  }
}
```
</details>

rbb-dev, 2025-03-12:
Not related, i'm sure, but your charger is sending an invalid GUID:
```
17:08:34.673 INFO (MainThread) [ocpp] MySerialNumber: receive message [2,"250311060834-42","BootNotification",{"chargePointModel":"48","chargePointSerialNumber":"MySerialNumber","chargePointVendor":"EN+","firmwareVersion":"1.0.25.130"}]
```
**250311060834-42** is not a valid GUID.
See if there is a firmware upgrade for it.

drc38, 2025-03-12:
The messages are async so you need to match the send/receive ids. The rejected message is:
`send [2,"a3148555-744a-49b5-bb26-2c4261ab4358","TriggerMessage",{"requestedMessage":"StatusNotification","connectorId":0}]`

BJReplay, 2025-03-12:
> See if there is a firmware upgrade for it.

I updated to the latest firmware yesterday before logging this issue.

> The messages are async so you need to match the send/receive ids. The rejected message is:

Thanks.  Is that a bug in the chargers OCPP implementation?  If so (especially if you can point me to the expected response), I'll raise both issues with the vendor.

drc38, 2025-03-12:
You can check against the standard here: https://github.com/mobilityhouse/ocpp/tree/master/docs/v16

BJReplay, 2025-03-13:
> Not related, i'm sure, but your charger is sending an invalid GUID:

Thanks.  From the docs at https://github.com/mobilityhouse/ocpp/tree/master/docs/v16 I couldn't find the structure of the message, just the payload.

I did find this reference to the structure: http://www.gir.fr/ocppjs/ocpp_srpc_spec.shtml which refers to callID

From the ocpp library at https://github.com/mobilityhouse/ocpp/blob/master/ocpp/messages.py#L266 it doesn't look like the unique_id has to be an GUID.

To be clear - I'm not suggesting that you're wrong when you say it has to be a GUID - you're the expert - but rather I can't find a reference to this that I can refer back to the vendor.

And, in terms of the original issue, it looks - to my intelligent layman reading - that it's acceptable for the charger to return REJECTED to the StatusNotification request.  Odd, but acceptable.

Changing the title, and closing, because it's not a bug, but if you have more detail on the rules around unique_id that I can share with the vendor, please let me know, otherwise feel free to ignore. 

Thanks

closed 2025-03-13 by BJReplay

rbb-dev, 2025-03-13:
Sorry, I take it back.. It does not stipulate that it must be GUID, but the spec says it must be up to 36 characters to allow for GUID.
So it can be anything, as long as it's unique, but universally, everyone uses GUID.
[image]