lbbrhzn/ocpp #1576: jsonschema.exceptions.ValidationError: 'Hz'

KoalaWerewolf, 2025-03-12:
**Describe the bug**
Chargepoint communication seems to fail due to a validation error.

I'm connecting my ChargeAmps Halo to Home Assistant using OCPP. I know it's not listed as one of the tested chargers but it looks promising overall. I've configured the charger, the measurands appear, the Diagnostic sensors report sensible values.

I've connected my car to see if I can get other values and see how I can control the charger. During this testing I've come up against a ValidationError.

**To Reproduce**
Connect a ChargeAmps Halo charger.

**Expected behavior**
Working communication and sensors reporting correct values.

** Versions **
- OCPP 0.8.4
- HAOS 2025.3.2

**Additional context**
Logger: ocpp
Source: /usr/local/lib/python3.13/site-packages/ocpp/charge_point.py:274
First occurred: 13:35:37 (15 occurrences)
Last logged: 13:54:12

Error while handling request '<Call - unique_id=265468940-1621710057-682692942, action=MeterValues, payload={'connectorId': 1, 'transactionId': 1741782933, 'meterValue': [{'timestamp': '2025-03-12T12:37:37Z', 'sampledValue': [{'value': '14', 'context': 'Sample.Periodic', 'measurand': 'Temperature', 'unit': 'Celsius'}, {'value': '49.97', 'context': 'Sample.Periodic', 'measurand': 'Frequency', 'unit': 'Hz'}, {'value': '10', 'context': 'Sample.Periodic', 'measurand': 'Energy.Active.Import.Register', 'unit': 'Wh'}, {'value': '16.0', 'context': 'Sample.Periodic', 'measurand': 'Current.Offered', 'unit': 'A'}, {'value': '262.56', 'context': 'Sample.Periodic', 'measurand': 'Power.Active.Import', 'unit': 'W'}, {'value': '1.379', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'phase': 'L1', 'unit': 'A'}, {'value': '234.09', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'phase': 'L1', 'unit': 'V'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'phase': 'L2', 'unit': 'A'}, {'value': '234.66', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'phase': 'L2', 'unit': 'V'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'phase': 'L3', 'unit': 'A'}, {'value': '235.83', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'phase': 'L3', 'unit': 'V'}]}]}>'
Error while handling request '<Call - unique_id=1089303919-1647867889-10146282, action=MeterValues, payload={'connectorId': 1, 'transactionId': 1741782933, 'meterValue': [{'timestamp': '2025-03-12T12:37:37Z', 'sampledValue': [{'value': '14', 'context': 'Sample.Periodic', 'measurand': 'Temperature', 'unit': 'Celsius'}, {'value': '49.97', 'context': 'Sample.Periodic', 'measurand': 'Frequency', 'unit': 'Hz'}, {'value': '10', 'context': 'Sample.Periodic', 'measurand': 'Energy.Active.Import.Register', 'unit': 'Wh'}, {'value': '16.0', 'context': 'Sample.Periodic', 'measurand': 'Current.Offered', 'unit': 'A'}, {'value': '262.56', 'context': 'Sample.Periodic', 'measurand': 'Power.Active.Import', 'unit': 'W'}, {'value': '1.379', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'phase': 'L1', 'unit': 'A'}, {'value': '234.09', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'phase': 'L1', 'unit': 'V'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'phase': 'L2', 'unit': 'A'}, {'value': '234.66', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'phase': 'L2', 'unit': 'V'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'phase': 'L3', 'unit': 'A'}, {'value': '235.83', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'phase': 'L3', 'unit': 'V'}]}]}>'
Error while handling request '<Call - unique_id=1394007239-761799734-28452108, action=MeterValues, payload={'connectorId': 1, 'transactionId': 1741782933, 'meterValue': [{'timestamp': '2025-03-12T12:38:37Z', 'sampledValue': [{'value': '14', 'context': 'Sample.Periodic', 'measurand': 'Temperature', 'unit': 'Celsius'}, {'value': '49.99', 'context': 'Sample.Periodic', 'measurand': 'Frequency', 'unit': 'Hz'}, {'value': '11', 'context': 'Sample.Periodic', 'measurand': 'Energy.Active.Import.Register', 'unit': 'Wh'}, {'value': '16.0', 'context': 'Sample.Periodic', 'measurand': 'Current.Offered', 'unit': 'A'}, {'value': '2.29', 'context': 'Sample.Periodic', 'measurand': 'Power.Active.Import', 'unit': 'W'}, {'value': '0.460', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'phase': 'L1', 'unit': 'A'}, {'value': '234.73', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'phase': 'L1', 'unit': 'V'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'phase': 'L2', 'unit': 'A'}, {'value': '234.99', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'phase': 'L2', 'unit': 'V'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'phase': 'L3', 'unit': 'A'}, {'value': '237.71', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'phase': 'L3', 'unit': 'V'}]}]}>'
Error while handling request '<Call - unique_id=1248899787-1482653785-20610958, action=MeterValues, payload={'connectorId': 1, 'transactionId': 1741782933, 'meterValue': [{'timestamp': '2025-03-12T12:38:37Z', 'sampledValue': [{'value': '14', 'context': 'Sample.Periodic', 'measurand': 'Temperature', 'unit': 'Celsius'}, {'value': '49.99', 'context': 'Sample.Periodic', 'measurand': 'Frequency', 'unit': 'Hz'}, {'value': '11', 'context': 'Sample.Periodic', 'measurand': 'Energy.Active.Import.Register', 'unit': 'Wh'}, {'value': '16.0', 'context': 'Sample.Periodic', 'measurand': 'Current.Offered', 'unit': 'A'}, {'value': '2.29', 'context': 'Sample.Periodic', 'measurand': 'Power.Active.Import', 'unit': 'W'}, {'value': '0.460', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'phase': 'L1', 'unit': 'A'}, {'value': '234.73', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'phase': 'L1', 'unit': 'V'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'phase': 'L2', 'unit': 'A'}, {'value': '234.99', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'phase': 'L2', 'unit': 'V'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'phase': 'L3', 'unit': 'A'}, {'value': '237.71', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'phase': 'L3', 'unit': 'V'}]}]}>'
Error while handling request '<Call - unique_id=389571790-1074765951-172066813, action=MeterValues, payload={'connectorId': 1, 'transactionId': 1741782933, 'meterValue': [{'timestamp': '2025-03-12T12:38:37Z', 'sampledValue': [{'value': '14', 'context': 'Sample.Periodic', 'measurand': 'Temperature', 'unit': 'Celsius'}, {'value': '49.99', 'context': 'Sample.Periodic', 'measurand': 'Frequency', 'unit': 'Hz'}, {'value': '11', 'context': 'Sample.Periodic', 'measurand': 'Energy.Active.Import.Register', 'unit': 'Wh'}, {'value': '16.0', 'context': 'Sample.Periodic', 'measurand': 'Current.Offered', 'unit': 'A'}, {'value': '2.29', 'context': 'Sample.Periodic', 'measurand': 'Power.Active.Import', 'unit': 'W'}, {'value': '0.460', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'phase': 'L1', 'unit': 'A'}, {'value': '234.73', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'phase': 'L1', 'unit': 'V'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'phase': 'L2', 'unit': 'A'}, {'value': '234.99', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'phase': 'L2', 'unit': 'V'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'phase': 'L3', 'unit': 'A'}, {'value': '237.71', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'phase': 'L3', 'unit': 'V'}]}]}>'
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/ocpp/messages.py", line 239, in _validate_payload
    validator.validate(message.payload)
    ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/jsonschema/validators.py", line 451, in validate
    raise error
jsonschema.exceptions.ValidationError: 'Hz' is not one of ['Wh', 'kWh', 'varh', 'kvarh', 'W', 'kW', 'VA', 'kVA', 'var', 'kvar', 'A', 'V', 'K', 'Celcius', 'Celsius', 'Fahrenheit', 'Percent', 'Hertz']

Failed validating 'enum' in schema['properties']['meterValue']['items']['properties']['sampledValue']['items']['properties']['unit']:
    {'type': 'string',
     'additionalProperties': False,
     'enum': ['Wh',
              'kWh',
              'varh',
              'kvarh',
              'W',
              'kW',
              'VA',
              'kVA',
[... cut ...] 'Sample.Periodic', 'measurand': 'Voltage', 'phase': 'L1', 'unit': 'V'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'phase': 'L2', 'unit': 'A'}, {'value': '235.04', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'phase': 'L2', 'unit': 'V'}, {'value': '0.000', 'context': 'Sample.Periodic', 'measurand': 'Current.Import', 'phase': 'L3', 'unit': 'A'}, {'value': '236.30', 'context': 'Sample.Periodic', 'measurand': 'Voltage', 'phase': 'L3', 'unit': 'V'}]}]}>}

KoalaWerewolf, 2025-03-13:
> The charger should be sending Hertz, not Hz.
> 
> This is a bug in the charger's implementation.
> 
> You can work around it by checking the "Skip OCPP schema validation" Checkbox when you set up the charger. You would need to remove it and add it again to change this.

Understood. However, considering the values in the enum that it validates against, Hz seems like a perfectly reasonable value. Making sure that the most used string representation for the unit for frequency does not throw an error would probably be a good thing.

drc38, 2025-03-13:
You can propose adding to the upstream library:
https://github.com/mobilityhouse/ocpp

BJReplay, 2025-03-13:
> Hz seems like a perfectly reasonable value.

Yes, it does. But you could argue that kilowatt and kiloWatt also seem perfectly reasonable - and they do.

% seems reasonable, instead of Percent

But there is a standard, and it is explicit on what UOMs are acceptable.

It is a optional element in the specification, but if supplied, must be one of the values from the spec. 

Remember, the charger needs to operate with every OCPP central system if it wants to a) be compliant and b) grow market share.

I would be surprised if the charger manufacturer would not change the UOM once made aware of the error.

The spec https://groups.oasis-open.org/higherlogic/ws/public/download/58944/ocpp-1.6.pdf/latest specifies this at 7.45.

closed 2025-05-20 by github-actions[bot]

KoalaWerewolf, 2025-06-23:
7.45 in that document still does not list Frequency, and it's not included in the json. You can also argue that the text above is correct, the unit is Hertz, but you won't find Hertz widely used in implementations, you will find Hz (and kHz and MHz). I'm sorry to say but you are continuing to argue that if the real world and the map does not match, then it's the map that is correct. The core issue here is interoperability. This is an example where the standard can do a better job, and the implementation of the standard can do an even better job (by smoothing over a specification oversight) to ensure interoperability.

If you continue to feel that Hz shouldn't be validated then consider the fact that "Hertz" is not included in the json and should on the same grounds not be validated either. Skip validation completely for Frequency and all meter values will be considered as Hertz anyway. This way chargers that implement Hz would work without bypassing schema validation. It makes more sense to add Hz to the enumeration.

The erratas for that version also argue a similar point "We advise implementers of an OCPP 1.6 JSON Central System to add the correct spelling (Celsius) to the enumeration. This way your implementation will support both the correct and the incorrect spellings." An example where the implementation needs to do some error handling. OCPP not specifying Hz as a unit for Frequency is an oversight, not a design choice.

BJReplay, 2025-06-23:
Look, I don't have a dog in this fight. I just happened to have been reading the spec for a different reason when you raised your issue, and recalled that I had seen UoMs and Frequency defined. My failure was attempting to find a URL to the latest spec that wasn't a zip (I failed - I provided the original 1.6 version) and not pointing out both clauses. 

The spec may not be perfectly written, but 7.31 notes - in the latest version of 1.6 - the gap in 7.45, and clearly defines the UoM for frequency. 

That is what the spec says - badly in your opinion, I dare say - and so that is the implied if not explicitly defined in 7.45 UoM for any charger that wants to claim compliance with 1.6.

You can complain here about processes, errata, standards, etc., but this is a community implementation of a spec.

Complaints about the spec should be addressed to the specification body.