lbbrhzn/ocpp #1590: ABB Terra AC wallbox

pierinho, 2025-03-19:
OCPP Integration v0.8.4
I read the documents and i find a problem with my charger (ABB Terra AC chargers with firmware version 1.8.21 and earlier fail to respond correctly when OCPP measurands are automatically detected by the OCPP integration.), but when try to add the integration, where is "Disable Automatic detection of OCPP Measurands” option?
In the previous version 0.7.0 i find it, in the new 0.8.4 no.
Thanks

d-l-h, 2025-03-20:
The process has changed a bit with v0.8. 
You will first just set-up the central. After a while, when your charger connects to the central, a new wizard will be available to configure the discovered charger and in that wizard you'll have the option to "Disable automatic detection of OCPP measurands"

pierinho, 2025-03-25:
Ok, working fine.
One question, the reset button (button.wallbox_reset) found in integration, restart the wallbox or reset wallbox settings?

matteocopelli, 2025-04-29:
Good morning, I have OCPP version 0.8.4 and an AC ground wallbox with SW 1.8.32 version.

The station is detected by the server but following the procedure in the documentation, no sensors are read (all sensors are "not available".

Any solution/test to carry out?

pierinho, 2025-04-29:
> Good morning, I have OCPP version 0.8.4 and an AC ground wallbox with SW 1.8.32 version.
> 
> The station is detected by the server but following the procedure in the documentation, no sensors are read (all sensors are "not available".
> 
> Any solution/test to carry out?

Try with full reboot of station. For me resolve the not available sensor.

matteocopelli, 2025-04-30:
I tried to reboot from the WB electrical panel but nothing changes.

I also tried to change the port of the ocpp server (reading a closed issue on our wallbox, someone solved it like that) but nothing...

I noticed that the connection between the WB and the server works, because when events happen to the WB, notifications called "Security Notifications" arrive, but the entities always remain "unavailable".

I'm thinking of trying to downgrade ocpp, what do you think?

pierinho, 2025-04-30:
Try to downgrade occp. Or you can try to delete the integration, reboot Home Assistant, and reinstall fresh OCPP Integration.
One question, are you Italian?

JoseEugenioGonzalez, 2025-05-19:
Hello, I have just installed a Terra Ac w7 G5 with firmware version 1.8.32

The charger is connected to the WiFi but I can't change the Ocpp server in the abb apps (terra config and charger sync) and it is fixed to an ocpp URL from Toyota.

I have installed the hacks integration and I figured central but my charger is not detected although I can see it connected to my router? 

Any idea about what could I do?

Thanks:)

matteocopelli, 2025-05-19:
I solved it by downgrading ocpp.
Yes, I am Italian.

d-l-h, 2025-05-20:
@JoseEugenioGonzalez 
You need to login first into the web portal of TerraConfig (with your TerraConfig username and password): https://abb.installer.chargedot.com/

In the portal you can add your own OCPP server.

closed 2025-07-26 by github-actions[bot]