lbbrhzn/ocpp #1594: Consider setting `default_authorization_status` to `invalid`

sshaikh, 2025-03-20:
## Version of the custom_component

0.8.4

## Configuration

None/Empty

## Describe the bug

At the moment with a default (ie empty) configuration, any RFID tag (including a phone) can authorise the charger for charging. This leaves it open to anyone to use if they gave it a try :)

## Debug log

N/A - code working as expected.

sshaikh, 2025-03-20:
I believe the offending line to be __init__.py:77:

```
CONFIG_SCHEMA = vol.Schema(
    {
        vol.Optional(
            CONF_DEFAULT_AUTH_STATUS, default=AuthorizationStatus.accepted.value
        ): cv.string,
        vol.Optional(CONF_AUTH_LIST, default={}): vol.Schema(
            {cv.string: AUTH_LIST_SCHEMA}
        ),
    },
    extra=vol.ALLOW_EXTRA,
)
```

sshaikh, 2025-03-21:
Yes, the suggestion is to make it the default (ie when there is no configuration), and to instead have users be explicit if they want to make it as permissive as it currently is.

drc38, 2025-03-21:
Changing the default is not the preferred approach as most home users will physically secure access to their charger so have no need to manage authorizations.

sshaikh, 2025-03-21:
I suspect it's geographical (in the UK I believe most chargers are outdoors), but the decision makes sense in the garage context.

Ideally it would be configurable via the initialisation flow. I have more ideas about this so will raise it in another ticket.

closed 2025-03-21 by sshaikh