lbbrhzn/ocpp #1599: SSL Not Seen To Be Working

sshaikh, 2025-03-22:
**Describe the bug**

I am able to connect to ocpp without SSL enabled. When I enable SSL, no connection is made.

**To Reproduce**
Steps to reproduce the behavior:
1. set up ocpp without SSL
2. see a successful connection
3. remove and readd ocpp, this time setting SSL on and configuring paths to use the same certificates as HA
4. no connection formed

**Expected behavior**
for the charger to connect without issue

**Additional context**
I am using the same cert and key as HA which works as expected. There doesn't seem to be any discernable errors (or indeed ocpp output at all) in the logs (when an insecure connection is made, the log runs).

sshaikh, 2025-03-22:
Connecting to the webservice via a browser works in both cases (albeit with a warning that it's a webservice), and each test reflects the secure status correctly, so this may be a charger issue - however I doubt this as it is able to connect to the vendor OCPP server securely.

BJReplay, 2025-03-23:
> so this may be a charger issue - however I doubt this as it is able to connect to the vendor OCPP server securely

Almost certainly a charger issue:
- If you have a self signed cert, the charger may not be able to validate the cert
- Even with a signed cert, the charger may not be able to validate it if it doesn't have access to the root certs or intermediate certs
- A workaround may be to use the full chain cert in the OCPP config, which might allow the charger to validate the cert

sshaikh, 2025-03-23:
I'll try again with debug logs on for both ocpp and webservices, but I suppose what we need are logs from the charger. I notice that the ocpp spec provides for this, is it possible to obtain them via this integration?

BJReplay, 2025-03-23:
> Fwiw it's a let's encrypt certificate.

FWIW my charger works with a let's encrypt cert.

But it is well known that many ESP32 devices struggle with certs, particularly if the firmware hasn't been compiled with the right options, and many chargers use ESP32 SOCs. 

ChargeHQ warns against SSL as it simply doesn't work for a lot of chargers. 

What charger are you using? You may find that it is one of the ones that has a broken SSL implementation.

Try it with ChargeHQ and determine if you can get it working with SSL there.

sshaikh, 2025-03-25:
>Try it with ChargeHQ and determine if you can get it working with SSL there.

Thanks for the tip. ws worked but wss didn't, so I agree that the charger may only have a certificate (or other config) for the vendor OCPP server.

closed 2025-03-25 by sshaikh

BJReplay, 2025-03-25:
> the charger may only have a certificate (or other config) for the vendor OCPP server

@sshaikh what brand / model is your charger? The fact that wss doesn't work is worth sharing for the community.

sshaikh, 2025-03-26:
It's the Cord EV one, aka the Caro:

https://github.com/lbbrhzn/ocpp/discussions/1591

BJReplay, 2025-03-26:
Interesting - I have a rebranded Caro - Switch EV, and it works with wss fine.  Worked with both older and current firmware.

You might need to update your firmware (if you haven't yet).

sshaikh, 2025-03-26:
According to Cord I have the latest firmware, so I presume it is one built specifically for Monta (their outsourced platform)

BJReplay, 2025-03-26:
> According to Cord I have the latest firmware

It's not.

> so I presume it is one built specifically for Monta (their outsourced platform).

Monta is just an OCPP central station.  It's unlikely.

> Fingers crossed for a generic firmware for all these variants!

Just run the base firmware - request it as described https://github.com/lbbrhzn/ocpp/wiki/supported-devices#en-caro-series-home-wallbox

sshaikh, 2025-03-26:
I presumed that was for a specific vendor, and each white label would have their own firmware. I'd rather not flash something officially unsupported this early on in the life of the charger, just in case there's some differences we don't know about.