lbbrhzn/ocpp #1600: Grizzle Ultimate 3.09.9 entities show as 'unavailable'.

hpeyerl, 2025-03-23:
I got the integration installed and managed to point my charger at my HA port using ws://......:9000/

It looks like My Ultimate-48A charger is connected to the OCPP integration and the logs show periodic messages coming from the charger:

```
2025-03-23 07:19:06.374 INFO (MainThread) [ocpp] GRU-L0000348767: receive message [2,"15671","MeterValues",{
	"connectorId":	1,
	"transactionId":	1058136,
	"meterValue":	[{
			"timestamp":	"2025-03-23T13:19:03Z",
			"sampledValue":	[{
					"value":	"0.00",
					"unit":	"A",
					"measurand":	"Current.Import"
				}, {
					"value":	"240.00",
					"unit":	"V",
					"measurand":	"Voltage"
				}, {
					"value":	"0.00",
					"unit":	"kW",
					"measurand":	"Power.Active.Import"
				}, {
					"value":	"15627",
					"unit":	"Wh",
					"measurand":	"Energy.Active.Import.Register"
				}]
		}]
}]
2025-03-23 07:19:06.374 WARNING (MainThread) [custom_components.ocpp] Unknown transaction detected with id=1058136
2025-03-23 07:19:06.376 INFO (MainThread) [ocpp] GRU-L0000348767: send [3,"15671",{}]
```

I noticed that there was a documentation update in supported-devices.md suggesting that Grizzl-E Ultimate should work mostly without issue other than amperage changes not accepted or applied.  

I'm not familiar with how this SHOULD work but  in my HA /config/devices/device/<hex string> screen, it looks like it has correctly identified a  "GRU 48A 2024" charger with "Firmware: ULGRW000A-03.09.9" so there seems to be at least some communication from the Grizzl-e.  All of the Sensors, Configuration, and Diagnostic entities show as "Unavailable". 

[image]

So have I misconfigured it or something, or is there something I've missed.  

fwiw, when the GRU-48A is configured to point to HA's OCPP server, the charger won't charge the car.  The car (Blazer EV) says "Charger is connected but not delivering power" and the status on the Grizzl-E web page says "Standby".  

Sorry if this is a basic 'user' question.  I'm not sure how to proceed.

lewtopia, 2025-03-28:
I'm not the only one!!

I have the exact same experience.  For a few brief seconds immediately after adding the integration, the controls become available, but then quickly everything becomes Unavailable.

I have found older threads referring to the same or a similar issue.  The fixes don't appear to be relevant, as the current code has changed significantly.

[https://github.com/lbbrhzn/ocpp/issues/442#issuecomment-1237651231](url)

I purchased the Grizzl-e solely on the facts that it is OCPP compliant (apparently not?) AND that it is "Made in Canada".

Sliding that DIP into dumb mode, charges instantly.  But also would charge any car, not just my own.  And I also wouldn't be able to get metrics and such.

I'm still looking for a solution to this.  It shouldn't be too far away.

hpeyerl, 2025-03-28:
I'm disappointed with Grizzl-E.  After a week of iterating with their tech-support department, the end result is they only support charging when connected to their cloud or when disconnected completely.  The fact that it wouldn't charge my car when configured to talk to HomeAssistant via OCPP and their response was "that's not a supported configuration. There's something wrong at HomeAssistant.  We only support using the Grizzle OCPP server."

lewtopia, 2025-03-28:
As I was further messing about with this, trying different methods of connecting, I accidentally stumbled back into the Grizzl-e Connect app, which informed me a firmware update is available!

[image]

[image]

So far, I have yet to be able to get the Ultimate 48A to connect to the internet. The webUI factory reset does anything but reset.

I will need to do the power off at panel, toggle to dumb mode, power on, power off, toggle back to smart, power on. 

Or something similar. I read it somewhere... like all DIPs must be 1 and then turn on power to actually reset.

I've run out of spare time today to continue fussing about with this thing. 

However, I did read on the repository notes that the 3.11.x firmware is supposed to correct these OCPP issues.

hpeyerl, 2025-03-28:
I did see a note about 3.11.x firmware fixing some OCPP issues but my charger didn't think that update was available.  So I guess you got to be the guinea pig.  

I'm out of town so I eagerly await to hear how you make out with it.

loc, 2025-04-29:
FWIW, I had this same issue with the entities showing up as Unavailable and I restarted Home Assistant (full restart) and it started working. Not sure that will work for everyone but it did for me.

hpeyerl, 2025-04-29:
Thanks.  My HomeAssistant gets fully restarted around once a week now since the latest update unfortunately.  Didn't help my Grizzl-e integration.

loggin722, 2025-05-29:
> FWIW, I had this same issue with the entities showing up as Unavailable and I restarted Home Assistant (full restart) and it started working. Not sure that will work for everyone but it did for me.

Same in my case. After restarting HA, OCPP works also with Schneider Charge

emitclever, 2025-07-11:
restart HA fix the issue after adding.