lbbrhzn/ocpp #1602: Webasto Next doesnt work (anymore)

renebiemans, 2025-03-24:
**Describe the bug**
After installation of the Webasto Next I tried first to connect with this intergration. This did work and I also reveived the Diagnostic information. After updating to the latest firmware of the Webasto Next, I was not able anymore to receive the data. I can add the intergration and also the device information (device WB_NEXT_EVSE_EU_32A) is received (so there is a communication). At first installation the V2 firmware was installed in the charger which is unfortunatly not available for download, therefore downgrade is also not possible. The notification in home assistant says: "Charger NEXT-WS111143 rebooted".

**To Reproduce**
Steps to reproduce the behavior: 
Setup the connection, by adding

**Expected behavior**
Getting data

**Screenshots**
 ##
[image]
[image]
[image]

**Desktop (please complete the following information):**
- Home assistant core: 2025.3.4
- Home assistant supervisor: 2025.03.3
- Operating System: 15.0
- OCPP 0.8.4 

**Additional context**

[home-assistant_ocpp_2025-03-24T23-07-32.018Z.log](https://github.com/user-attachments/files/19439172/home-assistant_ocpp_2025-03-24T23-07-32.018Z.log)

In the logging I found this: 2025-03-25 00:05:57.477 INFO (MainThread) [ocpp] NEXT-WS111143: receive message [2,"c4fe0189-fff0-4bf0-8cbf-9895fcbfccd0","BootNotification",{"chargePointModel":"WB_NEXT_EVSE_EU_32A","chargePointVendor":"Webasto","chargePointSerialNumber":"NEXT-WS111143","firmwareVersion":"3.1.16"}]
2025-03-25 00:05:57.479 INFO (MainThread) [ocpp] NEXT-WS111143: send [3,"c4fe0189-fff0-4bf0-8cbf-9895fcbfccd0",{"currentTime":"2025-03-24T23:05:57Z","interval":3600,"status":"Accepted"}]
2025-03-25 00:05:57.883 INFO (MainThread) [ocpp] NEXT-WS111143: receive message [3,"76a29817-6b27-49ec-a7f1-9bf54b531e14",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,RemoteTrigger,SmartCharging"}],"unknownKey":[]}]
2025-03-25 00:05:57.884 INFO (MainThread) [ocpp] NEXT-WS111143: send [2,"dd9d88ec-1fcc-4669-8297-76f3f65477ae","GetConfiguration",{"key":["NumberOfConnectors"]}]
2025-03-25 00:06:01.344 INFO (MainThread) [ocpp] NEXT-WS111143: receive message [2,"de177207-0fe7-4bd8-831e-42841e265d58","BootNotification",{"chargePointModel":"WB_NEXT_EVSE_EU_32A","chargePointVendor":"Webasto","chargePointSerialNumber":"NEXT-WS111143","firmwareVersion":"3.1.16"}]
2025-03-25 00:06:01.346 INFO (MainThread) [ocpp] NEXT-WS111143: send [3,"de177207-0fe7-4bd8-831e-42841e265d58",{"currentTime":"2025-03-24T23:06:01Z","interval":3600,"status":"Accepted"}]
2025-03-25 00:06:03.855 ERROR (Recorder) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/queues.py", line 186, in get
    await getter
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 421, in call
    response = await self._get_specific_response(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        call.unique_id, self._response_timeout
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/ocpp/chargepoint.py", line 338, in _get_specific_response
    resp = await super()._get_specific_response(unique_id, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 455, in _get_specific_response
    response = await asyncio.wait_for(self._response_queue.get(), timeout)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 506, in wait_for
    async with timeouts.timeout(timeout):
               ~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/ocpp/chargepoint.py", line 228, in post_connect
    num_connectors: int = await self.get_number_of_connectors()
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... 95 lines cut ...]
  File "/config/custom_components/ocpp/chargepoint.py", line 227, in post_connect
    await self.fetch_supported_features()
  File "/config/custom_components/ocpp/chargepoint.py", line 218, in fetch_supported_features
    self._attr_supported_features = await self.get_supported_features()
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/ocppv16.py", line 162, in get_supported_features
    resp = await self.call(req)
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 425, in call
    raise asyncio.TimeoutError(
    ...<2 lines>...
    )
TimeoutError: Waited 10s for response on [2,"921c1e25-6fbe-4562-9237-9ac5a921bd34","GetConfiguration",{"key":["SupportedFeatureProfiles"]}].
2025-03-25 00:06:04.942 INFO (MainThread) [ocpp] NEXT-WS111143: receive message [2,"f3bf25fe-a111-47ed-8de7-e666efaa711b","BootNotification",{"chargePointModel":"WB_NEXT_EVSE_EU_32A","chargePointVendor":"Webasto","chargePointSerialNumber":"NEXT-WS111143","firmwareVersion":"3.1.16"}]
2025-03-25 00:06:04.944 INFO (MainThread) [ocpp] NEXT-WS111143: send [3,"f3bf25fe-a111-47ed-8de7-e666efaa711b",{"currentTime":"2025-03-24T23:06:04Z","interval":3600,"status":"Accepted"}]
2025-03-25 00:06:07.885 INFO (MainThread) [ocpp] NEXT-WS111143: send [2,"531f7dcb-dd26-4f83-b7a6-739d91b6ac8a","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2025-03-25 00:06:08.329 INFO (MainThread) [ocpp] NEXT-WS111143: receive message [2,"5e348ab7-86ce-470a-ba32-7b4ad5d7deca","BootNotification",{"chargePointModel":"WB_NEXT_EVSE_EU_32A","chargePointVendor":"Webasto","chargePointSerialNumber":"NEXT-WS111143","firmwareVersion":"3.1.16"}]
2025-03-25 00:06:08.331 INFO (MainThread) [ocpp] NEXT-WS111143: send [3,"5e348ab7-86ce-470a-ba32-7b4ad5d7deca",{"currentTime":"2025-03-24T23:06:08Z","interval":3600,"status":"Accepted"}]
2025-03-25 00:06:08.696 INFO (MainThread) [ocpp] NEXT-WS111143: receive message [3,"531f7dcb-dd26-4f83-b7a6-739d91b6ac8a",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,RemoteTrigger,SmartCharging"}],"unknownKey":[]}]
2025-03-25 00:06:08.696 INFO (MainThread) [ocpp] NEXT-WS111143: send [2,"0b851856-d708-41fd-823f-081b6779bccf","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2025-03-25 00:06:12.150 INFO (MainThread) [ocpp] NEXT-WS111143: receive message [2,"dc6519fa-2e24-4ed4-a3e9-76e82e7c149b","BootNotification",{"chargePointModel":"WB_NEXT_EVSE_EU_32A","chargePointVendor":"Webasto","chargePointSerialNumber":"NEXT-WS111143","firmwareVersion":"3.1.16"}]
2025-03-25 00:06:12.152 INFO (MainThread) [ocpp] NEXT-WS111143: send [3,"dc6519fa-2e24-4ed4-a3e9-76e82e7c149b",{"currentTime":"2025-03-24T23:06:12Z","interval":3600,"status":"Accepted"}]
2025-03-25 00:06:15.644 INFO (MainThread) [ocpp] NEXT-WS111143: receive message [2,"60bfd6ad-797f-4c5a-b6d4-7c4cdc4382f4","BootNotification",{"chargePointModel":"WB_NEXT_EVSE_EU_32A","chargePointVendor":"Webasto","chargePointSerialNumber":"NEXT-WS111143","firmwareVersion":"3.1.16"}]

Hopefully somebody can find a fix for this :)

drc38, 2025-03-25:
You can change line 91 of `ocppv16.py` to return 1 but the firmware may have other issues. Best approach longer term is to raise it with the manufacturer and have them fix it.

renebiemans, 2025-03-25:
Hi, I think it is not related to the version of the firmware. Somtimes after a reboot I do get the diagnostic information, but all other not. And then after a minute it does say it keeps rebooting. With the change in line 91 I still get the same issue, also after a reboot. 

I tried:
return await self.get_configuration(1)

and also return 1

drc38, 2025-03-25:
There may be an issue with your network/websocket connection given the messages are timing out.