lbbrhzn/ocpp #1607: Measurands no longer update during charging session in 0.8.x

cascer1, 2025-03-30:
**Describe the bug**
In versions prior to 0.8.0, I received measurand updates during a charging session. Specifically, the following values would update regularly:

* `sensor.charger_current_offered`
* `sensor.charger_energy_active_import_register`
* `sensor.charger_energy_session`
* `sensor.charger_power_active_import`

I'm sure other values perform similarly but I don't remember if they did or didn't update in previous versions.

The following values do update, but only after a charging session has ended:

* `sensor.charger_energy_session`
* `sensor.charger_energy_active_import_register`

**To Reproduce**
Steps to reproduce the behavior:
1. Install any 0.8.x version of the integration
2. Setup a Central System Configuration
3. Wait for charger auto discovery
4. Setup charger
4.1. Either choose to automatically discover measurands or manually specify them
4.2. Either enable or disable smart charging, both work
5. Observe non-updating values

**Expected behavior**
Values such as Current Offered, Power, or total energy register, should update frequently.

**Screenshots**

1: Active import register only updates after a charging session ends.
[image]

2: Active Power does not update at all during charging session, and remains at 0 kW.

[image]

3: On versions before 0.8.0, values would update during a charging session. For example, the active power import:

[image]

**Desktop (please complete the following information):**
 - OS: Windows 11, Fedora Workstation 42
 - Browser: Firefox 
 - Version: 136.0.4

**Smartphone (please complete the following information):**
 - Device: Google Pixel 9 Pro
 - OS: Android 15
 - Browser Firefox
 - Version 136.0.2

**Additional context**

I've tried several methods of setting up, like automatically detecting measurands or manually configuring them, as well as enabling or disabling smart charging profile.

A result of the values only updating after a session completes means that, among other things, the power usage is no longer properly accounted in the Home Assistant energy dashboard.

drc38, 2025-03-30:
Try reverting this PR https://github.com/lbbrhzn/ocpp/issues/1466 and see if it fixes your issue

cascer1, 2025-03-30:
I think I've actually discovered the origin of this issue and I believe it has to do with the automatic measurand discovery.

I can't find the exact logs yet, but this is how I understand the problem.

1. During automatic measurand discovery, the integration tries to set every measurand individually, to see if the charger responds with with success or error.
2. Then, it combines all the successful values into one request and asks the charger to send those
3. In my case, it found more than 9 possible values. However, the charger only supports setting up to 9. So, that request failed
4. The charger remembers the last successful request. In my case, this was `Voltage`, so only that value was sent by the charger in updates.

To fix this, I went into the charger settings and manually configured it to send the values I'm interested in.

[image]