lbbrhzn/ocpp #1612: Documenting Compleo eBox (innogy)

StefanLobbenmeier, 2025-04-05:
Hope it is okay to raise this as an issue, plan is create an issue of all information, then create a PR with any relevant information.

# Setup via bluetooth (if not connected to wifi)
1. Download compleo eConfig app from app store
2. Go to wallbox, long press on the bluetooth button for 5s to initiate pairing
3. Pair to the device in iPhone settings
4. turn off wifi and mobile data (background is that the bluetooth tethering connection opens a network and the app might get confused if there are other networks)
5. Open the eConfig app
6. At the very button, click on the “Configure <device name>
7. Enter the PUK, you should find it in the box that the device came with or possibly at the back of the manual
8. You might see the current settings, its a good idea to screenshot them
9. Possibly reset the device, it takes around 3 minutes to reboot afterwards, then follow 2-6 to reconnect
10. close the dialog that shows device configuration, click on configure
11. setup Amps and phases like previously screenshotted
12. enable phase shifting recognition if required in your country (e.g. in germany and austria)
13. Connect to Wifi, configure IP via DHCP
14. Connect to hass ocpp by ip address (not hostname), e.g. `ws://192.168.178.78:9000`

# Setup via network (if already connected to wifi)
1. Navigate to device configuration page, in my case https://icom/
2. enter credentials, username `admin`, password is your PUK
3. in OCPP settings, set endpoints uri to `ws://192.168.178.78:9000`
4. restart the device, e.g. via System -> Reset -> App Restart -> OK Confirm Restart

Also you can get log data from the wallbox via the web interface ECU -> Logging -> Download, which might help you debug any issues. E.g. now it should say that is successfully connected, then disconect immediately and retry that every second.

# Accept connection in Home Assistant OCPP Plugin
1. In the OCPP Plugin, there should now be a button to configure the charger that is trying to connect
2. Accept that connection, i named my device eBox
3. Possibly restart home assistant? In my case all entities were still disabled, but became available after the restart, YMMV

# Start charging
To enable charging, toggle on eBox Charge Control
The toggle eBox Availability does not seem to be doing anything for me.

# Adjust charging speed
eBox maximum current does NOT work, but you are also not supposed to use that. I followed this guide: https://github.com/lbbrhzn/ocpp/wiki/Charge_automation and that worked.

However, because it is a three phase charger, it does not quite behave as nicely. Tested with a hyundai Kona It seems that it refuses to charge with any phase below 2.8 kW. I was able to get the most options by setting the car amps limit to 60%, which gives a bit more granularity as we can only set the current limit to a whole number of Amps (right?)

I tested various current limits and those were my results:

60% Amp limit on the car
- 5A no charging
- 6A-10A charging at 2.7kW
- 11A 3.0kW
- 12A 3.2kW
- 13A 3.5kW
- 14A 5.6kW (I suspect it enabled a second phase here)
- 15A 6.1kW
- 16A 6.4kW
- 17A 6.8kW
- 18A 7.2kW
- 19A 7.6kW
- 20A or more 8.0kW

90% limit on the car
- 5A no charging
- 6A charging at 2.7kW
- 7A 2.8kW
- 8A 3.2kW
- 9A 3.5kW up to 5kW, it fluctuates a lot in this value so I would not recommend this
- 10A 6.0kW
- 11A 6.6kW
- 12A 7.2kW
- 13A 7.8kW
- 14A 8.5kW
- 15A 9.1kW
- 16A 9.8kW
- 17A 10.3kW
- 18A or more 10.9kW

# Issues I faced
- Tried connecting via hostname at first, this did not work
- Resetting the wallbox was very scary the first two times I did it, but it really takes 2-3 minutes so relax have some tea, it will come back up haha
- The very first charging session the car was unable to stop the charging session, it could only be stopped via the server. I suspect this was a fluke with the car though, the server should not have this kind of power (right?). Super weird that this can even happen (imagine this happening at a public charging spot, being held hostage there with no way to disconnect the car sounds super scary)

# Thoughts on home assistant automation
I get data on 
- how much power the solar panels generate
- how much electricity is consumed by the home / how much is put into the grid
- current charge level of the home battery

So the automation rule will likely involve those, something like
- attempt to use all the power that would otherwise be “wasted” by going into the grid
- when that value cannot be reached, get next lowest value from the amp table (see above)
- stop charging when there is less than 2.8kW coming from the roof and home battery has less than X%
- possibly also start charging when home battery reaches 100%

nylocx, 2025-05-04:
Thanks a lot this helped with setting up automations for my ebox smart cable (updated to 2.51).

StefanLobbenmeier, 2025-05-04:
Glad I could help 😄 

Another update from my side, the eBox I first tested with broke due to unrelated issue, so it’s now replaced with an eBox smart S, firmware 1.2.3

With that device the web interface briefly worked and now it is no longer reachable even after restarts, so I will probably update the firmware via ocpp soon. How was your experience with the updates?

StefanLobbenmeier, 2025-05-13:
Well tried the upgrade via OCPP again, but unfortunately not successfully:

```
2025-05-13 22:05:49.624 INFO (MainThread) [ocpp] LP00XXXX: send [2,"uuid","UpdateFirmware",{"location":"https://media-cdn.atlassian.com/eu-central-1/v1/cdn/file/71997c24-1eda-4980-818a-8134f78b740c/binary?","retrieveDate":"2025-05-13T20:05:49Z"}]
2025-05-13 22:07:41.929 INFO (MainThread) [ocpp] LP00XXXX: receive message [2,"xxxxxxxxxxx","FirmwareStatusNotification",{"status":"Downloading"}]
2025-05-13 22:08:55.533 DEBUG (MainThread) [custom_components.ocpp] Received boot notification for LP00XXXX: {'charge_box_serial_number': 'LP00XXXX', 'charge_point_model': 'eBox smart S', 'charge_point_vendor': 'innogy', 'firmware_version': '1.2.3', 'meter_serial_number': 'N/A', 'meter_type': 'N/A'}
2025-05-13 22:09:06.008 INFO (MainThread) [ocpp] LP007630: receive message [2,"eKGa0w2nDp5olPpg2qlt5eMomxLzUK18eIie","FirmwareStatusNotification",{"status":"DownloadFailed"}]
```

Not 100% sure if it is restarting because of failed download or if the download fails because of the restarts in the middle.

StefanLobbenmeier, 2025-06-03:
Update on the firmware upgrade situation, you can trigger it via ocpp but url needs to be sftp (or possibly ftp works as well)

You can also check the configuration key
```json
			{
				"key": "SupportedFileTransferProtocols",
				"readonly": true,
				"value": "ftp,sftp"
			},
```

Possibly this integration could check if the url matches the configuration key and warn if they do not match

stefanloerwald, 2026-02-23:
Over the last week, I tried to integrate my Compleo eBox smart into home assistant via OCPP and perform firmware updates. Many things were more difficult than necessary. I'm sharing a few details that stood in my way here, in the hope that it helps others:

My device was on firmware version 1.2.3. That firmware doesn't support connecting to SFTP servers for firmware updates.
I had to go via the webconfig interface. The trouble with that was that it was unreachable from the local network (via WiFi). So the cable option was the only possibility. In contrast to the manual, I had to use the supposedly deactivated LAN 1 instead of LAN 2 to establish a connection to a laptop. In the webconfig interface I then first installed the FW updates in sequence and then fiddled with the network settings (I think enabling WAN) until the webconfig interface was also reachable via local network instead of direct LAN connection.