lbbrhzn/ocpp #1617: Charger Stopes Receiving Measurands No control

lubricatedcactus, 2025-04-17:
## Version of the custom_component
v0.8.4

## Configuration

```yaml
Add your logs here.
```

## Describe the bug

Tieson Smart Mini
Charger stops reporting Import current despite charging (as evident from house meter sensors).
set_charge_rate seems to have no effect on the actual charge rate. 
Maximum Current slider does not work

## Debug log

```text

2025-04-17 11:18:37.635 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration ocpp which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-17 11:18:37.657 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration dynamic_ocpp_evse which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-17 11:19:02.122 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"ea5c0661-3268-427a-a865-b906b313bae4","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2025-04-17 11:19:02.403 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "ea5c0661-3268-427a-a865-b906b313bae4", {"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,LocalAuthListManagement,Reservation,SmartCharging,RemoteTrigger"}]}]
2025-04-17 11:19:02.409 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"7b146b58-c709-4fb8-9e48-6a631cb3d2eb","GetConfiguration",{"key":["NumberOfConnectors"]}]
2025-04-17 11:19:02.611 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "7b146b58-c709-4fb8-9e48-6a631cb3d2eb", {"configurationKey":[{"key":"NumberOfConnectors","readonly":true,"value":"1"}]}]
2025-04-17 11:19:02.615 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"4f938b72-6306-48fc-98de-0aae4ff54dfe","GetConfiguration",{"key":["HeartbeatInterval"]}]
2025-04-17 11:19:02.911 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "4f938b72-6306-48fc-98de-0aae4ff54dfe", {"configurationKey":[{"key":"HeartbeatInterval","readonly":false,"value":"3600"}]}]
2025-04-17 11:19:02.916 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"f1189273-9eec-4acc-89ec-799bea33a715","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Export"}]
2025-04-17 11:19:03.116 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "f1189273-9eec-4acc-89ec-799bea33a715", {"status":"Accepted"}]
2025-04-17 11:19:03.120 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"30f7496b-6243-42af-8504-c8f3313e2607","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Import"}]
2025-04-17 11:19:03.321 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "30f7496b-6243-42af-8504-c8f3313e2607", {"status":"Accepted"}]
2025-04-17 11:19:03.324 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"d75ede70-bee0-40ce-8dac-3e57a02ea502","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Offered"}]
2025-04-17 11:19:03.531 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "d75ede70-bee0-40ce-8dac-3e57a02ea502", {"status":"Accepted"}]
2025-04-17 11:19:03.534 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"e3d47a38-d693-4336-8675-da0fc2d3dd66","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Export.Interval"}]
2025-04-17 11:19:03.757 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "e3d47a38-d693-4336-8675-da0fc2d3dd66", {"status":"Accepted"}]
2025-04-17 11:19:03.761 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"ec1df9c5-047e-42a9-b91a-ac88fd04fab1","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Export.Register"}]
2025-04-17 11:19:03.935 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "ec1df9c5-047e-42a9-b91a-ac88fd04fab1", {"status":"Accepted"}]
2025-04-17 11:19:03.939 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"8b6a8775-5209-4a96-9b47-5fd79e79f753","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Interval"}]
2025-04-17 11:19:04.140 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "8b6a8775-5209-4a96-9b47-5fd79e79f753", {"status":"Accepted"}]
2025-04-17 11:19:04.143 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"73c4e261-b513-493e-a035-a9edc95db3db","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register"}]
2025-04-17 11:19:04.268 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "73c4e261-b513-493e-a035-a9edc95db3db", {"status":"Accepted"}]
2025-04-17 11:19:04.272 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"fab32c4a-4b1d-46a3-b822-2f4d9521ea96","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Reactive.Export.Interval"}]
2025-04-17 11:19:04.447 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "fab32c4a-4b1d-46a3-b822-2f4d9521ea96", {"status":"Accepted"}]
2025-04-17 11:19:04.451 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"169a7872-4ecd-4f93-b5b4-a8f02222a8bb","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Reactive.Export.Register"}]
2025-04-17 11:19:04.679 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "169a7872-4ecd-4f93-b5b4-a8f02222a8bb", {"status":"Accepted"}]
2025-04-17 11:19:04.683 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"9873372f-c9db-4e33-8699-2c4d7db22a2a","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Reactive.Import.Interval"}]
2025-04-17 11:19:04.856 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "9873372f-c9db-4e33-8699-2c4d7db22a2a", {"status":"Accepted"}]
2025-04-17 11:19:04.860 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"602cef84-45ea-46d5-8416-696eca6866c2","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Reactive.Import.Register"}]
2025-04-17 11:19:04.965 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "602cef84-45ea-46d5-8416-696eca6866c2", {"status":"Accepted"}]
2025-04-17 11:19:04.968 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"af30f8d4-81a7-45c3-928f-afc344059bb1","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Frequency"}]
2025-04-17 11:19:05.163 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "af30f8d4-81a7-45c3-928f-afc344059bb1", {"status":"Accepted"}]
2025-04-17 11:19:05.167 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"adda112e-26b3-42f5-b883-f599bec5755d","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Export"}]
2025-04-17 11:19:05.265 INFO (MainThread) [ocpp] 0312209102324480672: receive message [2, "08000148-6800-5708-da0c-581a76f42b72", "BootNotification", {"chargePointModel":"TeisonMe@7KW","chargePointSerialNumber":"0312209102324480672","chargePointVendor":"TEISON","firmwareVersion":"V1.5@25238bf+100,F8.gl,ws,ESP32,RN8213@V31","iccid":"","imsi":""}]
2025-04-17 11:19:05.274 INFO (MainThread) [ocpp] 0312209102324480672: send [3,"08000148-6800-5708-da0c-581a76f42b72",{"currentTime":"2025-04-17T01:19:05Z","interval":3600,"status":"Accepted"}]
2025-04-17 11:19:05.675 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "adda112e-26b3-42f5-b883-f599bec5755d", {"status":"Accepted"}]
2025-04-17 11:19:05.676 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"4890c012-e4e2-4be2-bf2c-09aae166c2d9","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2025-04-17 11:19:05.783 INFO (MainThread) [ocpp] 0312209102324480672: receive message [2, "08000149-6800-5709-0707-198b5c6187e7", "StatusNotification", {"connectorId":0,"errorCode":"NoError","status":"Available"}]
2025-04-17 11:19:05.900 INFO (MainThread) [ocpp] 0312209102324480672: send [3,"08000149-6800-5709-0707-198b5c6187e7",{}]
2025-04-17 11:19:06.187 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "4890c012-e4e2-4be2-bf2c-09aae166c2d9", {"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,LocalAuthListManagement,Reservation,SmartCharging,RemoteTrigger"}]}]
2025-04-17 11:19:06.188 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"6e429849-b5d7-4fee-bbf2-817026db1b5d","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Import"}]
2025-04-17 11:19:06.332 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "6e429849-b5d7-4fee-bbf2-817026db1b5d", {"status":"Accepted"}]
[... cut ...] riodic","measurand":"Voltage","location":"Outlet","unit":"V"}]}]}]
2025-04-17 13:19:40.316 INFO (MainThread) [ocpp] 0312209102324480672: send [3,"08000019-6800-734b-28b3-69434c5e2f63",{}]
2025-04-17 13:20:12.161 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '0312209102324480672': ping=0.0 ms, pong=48.0 ms
2025-04-17 13:20:32.231 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '0312209102324480672': ping=0.0 ms, pong=68.0 ms

```

lubricatedcactus, 2025-04-18:
This has happened again twice today. After sending "limit_amps”: 0  the sensor.current_import stops responding.
Sending "limit_amps”: 6 (or any other number) shows sensor.current_import  as 0.
Reboot of the charger or the OCPP intergration does not fix it but rebooting Home Assistant does.

`2025-04-18 13:59:13.772 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration ocpp which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-18 13:59:13.794 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration dynamic_ocpp_evse which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-18 13:59:38.170 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"ade410b6-3689-44f9-a06a-cea655d2727b","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2025-04-18 13:59:38.386 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "ade410b6-3689-44f9-a06a-cea655d2727b", {"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,LocalAuthListManagement,Reservation,SmartCharging,RemoteTrigger"}]}]
2025-04-18 13:59:38.392 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"30244693-aeae-47f9-aeac-96221eea3e5b","GetConfiguration",{"key":["NumberOfConnectors"]}]
2025-04-18 13:59:38.693 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "30244693-aeae-47f9-aeac-96221eea3e5b", {"configurationKey":[{"key":"NumberOfConnectors","readonly":true,"value":"1"}]}]
2025-04-18 13:59:38.698 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"d59e72d8-5904-4f51-98b9-37156ff16131","GetConfiguration",{"key":["HeartbeatInterval"]}]
2025-04-18 13:59:38.904 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "d59e72d8-5904-4f51-98b9-37156ff16131", {"configurationKey":[{"key":"HeartbeatInterval","readonly":false,"value":"3600"}]}]
2025-04-18 13:59:38.909 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"73163b19-21ee-407f-b40a-4f0f0957c438","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Export"}]
2025-04-18 13:59:39.159 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "73163b19-21ee-407f-b40a-4f0f0957c438", {"status":"Accepted"}]
2025-04-18 13:59:39.204 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"391bdc82-e1aa-438e-aa97-c971cbcdd916","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Import"}]
2025-04-18 13:59:39.410 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "391bdc82-e1aa-438e-aa97-c971cbcdd916", {"status":"Accepted"}]
2025-04-18 13:59:39.414 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"b3a333e4-d796-4fbc-ad4e-07b4b058d718","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Offered"}]
2025-04-18 13:59:39.615 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "b3a333e4-d796-4fbc-ad4e-07b4b058d718", {"status":"Accepted"}]
2025-04-18 13:59:39.618 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"aa35e6af-21f4-4084-809e-cbd8fc631907","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Export.Interval"}]
2025-04-18 13:59:39.827 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "aa35e6af-21f4-4084-809e-cbd8fc631907", {"status":"Accepted"}]
2025-04-18 13:59:39.831 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"ce45b4c7-57ca-4acc-895e-b50afe4efb22","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Export.Register"}]
2025-04-18 13:59:40.024 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "ce45b4c7-57ca-4acc-895e-b50afe4efb22", {"status":"Accepted"}]
2025-04-18 13:59:40.029 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"dc5e6677-4975-4d02-b5d3-388c1453a7b6","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Interval"}]
2025-04-18 13:59:40.228 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "dc5e6677-4975-4d02-b5d3-388c1453a7b6", {"status":"Accepted"}]
2025-04-18 13:59:40.232 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"f0b867aa-eb83-4d78-bb6b-297222347609","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register"}]
2025-04-18 13:59:40.462 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "f0b867aa-eb83-4d78-bb6b-297222347609", {"status":"Accepted"}]
2025-04-18 13:59:40.465 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"4bae12f2-d7c7-44dd-b172-c30d9cd41972","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Reactive.Export.Interval"}]
2025-04-18 13:59:40.639 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "4bae12f2-d7c7-44dd-b172-c30d9cd41972", {"status":"Accepted"}]
2025-04-18 13:59:40.643 INFO (MainThread) [ocpp] 0312209102324480672: send [2,"750646b9-ada2-42cb-8c9c-52d0abbf798e","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Reactive.Export.Register"}]
2025-04-18 13:59:40.854 INFO (MainThread) [ocpp] 0312209102324480672: receive message [3, "750646b9-ada2-42cb-8c9c-52d0abbf798e", {"status":"Accepted"}]
[... cut ...] ]}]}]
2025-04-18 13:59:43.412 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.farm_charger_power_active_export (<class 'custom_components.ocpp.sensor.ChargePointMetric'>) is using native unit of measurement 'kWh' which is not a valid unit for the device class ('power') it is using; expected one of ['W', 'GW', 'MW', 'mW', 'TW', 'kW']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/lbbrhzn/ocpp/issues

rbb-dev, 2025-04-19:
You charger supports more metervalues that it can handle.
Disable automatic discovery of metavalues and just select the metavalues you need.
Probably 4-5 should be enough :)
Do you really need all those below? No, just use what you need.

Current.Export,Current.Import,Current.Offered,Energy.Active.Export.Interval,Energy.Active.Export.Register,Energy.Active.Import.Interval,Energy.Active.Import.Register,Energy.Reactive.Export.Interval,Energy.Reactive.Export.Register,Energy.Reactive.Import.Interval,Energy.Reactive.Import.Register,Frequency,Power.Active.Export,Power.Active.Import,Power.Factor,Power.Offered,Power.Reactive.Export,Power.Reactive.Import,RPM,SoC,Temperature,Voltage

lubricatedcactus, 2025-04-19:
Thank You!!
That indeed was the problem.
Deleted the plugin and re setup with just a couple of meter values.
Node red is happily turning up and down the charge rate, It seems to have stopped all the notifications as well.

closed 2025-04-19 by lubricatedcactus