lbbrhzn/ocpp #1618: Latest version does not expose any entity

dral3x, 2025-04-18:
**Describe the bug**

Installing the latest version 0.8.4 creates only 1 device (central) but no entity.

However installing the version 0.7.0, it creates 2 devices and several entities.

I tried on 2 separate HomeAssistant instances and got the same result.

**To Reproduce**
Steps to reproduce the behavior:
1. Download OCPP 0.8.4 via Hacs
2. Restart HomeAssistant
3. Add Integration > OCPP > default settings
4. Only 1 device is created, central (see screenshot)

**Expected behavior**

The official guide https://home-assistant-ocpp.readthedocs.io/en/latest/installation.html shows 2 devices and 42 entities. While the specific numbers can be different (because of changes between versions), I would expect at least to have a few entities.

**Screenshots**
If applicable, add screenshots to help explain your problem.

Version 0.8.4
[image]

Version 0.7.0
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] MacOS
 - Browser [e.g. chrome, safari] Google Chrome
 - Version [e.g. 22] 135.0.7049.95

**Smartphone (please complete the following information):**
irrelevant

**Additional context**
Add any other context about the problem here.

rbb-dev, 2025-04-19:
You're supposed to connect the charger after setting up Central.
It will pop up in the devices automatically.

dral3x, 2025-04-23:
@rbb-dev Ok, so the documentation (for the latest version) is incorrect. Thanks for clarifying!

Sonnenwerk, 2025-05-30:
With my instance it found the charger some minutes later. But none of the entities were available. Docs say at least some of the entities should be available for my elvi wb.

henkvanderes, 2025-06-09:
I have the exact same issue for my EVBOX Elvi. As the app support for this Brand will stop - I decided to try the OCPP integration that supposed to support by EVBOX. 

I got this error log - someone who has any clue? I don't have any entities. when I downgraded to version 0.70 many entities became visible. but data was still not coming through. 

Any clue??

[image]

Logger: homeassistant.helpers.frame
Source: helpers/frame.py:335
First occurred: 22:39:18 (4 occurrences)
Last logged: 22:39:18

Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JXB60M11G1Q0JEKJJE29Y1ND'), with device info: {'identifiers': {('ocpp', 'charger')}, 'via_device': ('ocpp', '01JXB60M11G1Q0JEKJJE29Y1ND')} at custom_components/ocpp/sensor.py, line 77: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues
Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JXB60M11G1Q0JEKJJE29Y1ND'), with device info: {'identifiers': {('ocpp', 'charger')}, 'via_device': ('ocpp', '01JXB60M11G1Q0JEKJJE29Y1ND')} at custom_components/ocpp/switch.py, line 75: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues
Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JXB60M11G1Q0JEKJJE29Y1ND'), with device info: {'identifiers': {('ocpp', 'charger')}, 'via_device': ('ocpp', '01JXB60M11G1Q0JEKJJE29Y1ND')} at custom_components/ocpp/number.py, line 69: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues
Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JXB60M11G1Q0JEKJJE29Y1ND'), with device info: {'identifiers': {('ocpp', 'charger')}, 'via_device': ('ocpp', '01JXB60M11G1Q0JEKJJE29Y1ND')} at custom_components/ocpp/button.py, line 57: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues

tobidou, 2025-06-23:
> I have the exact same issue for my EVBOX Elvi. As the app support for this Brand will stop - I decided to try the OCPP integration that supposed to support by EVBOX.
> 
> I got this error log - someone who has any clue? I don't have any entities. when I downgraded to version 0.70 many entities became visible. but data was still not coming through.
> 
> Any clue??
> <img alt="Image" width="1329" src="https://private-user-images.githubusercontent.com/215502670/453198255-274189d5-0ca7-467f-b47b-eb6b6c9f3023.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTA2ODY3NjYsIm5iZiI6MTc1MDY4NjQ2NiwicGF0aCI6Ii8yMTU1MDI2NzAvNDUzMTk4MjU1LTI3NDE4OWQ1LTBjYTctNDY3Zi1iNDdiLWViNmI2YzlmMzAyMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNjIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDYyM1QxMzQ3NDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03ODM3N2M1NDI0OGU4Y2NmZTU3OWM5M2NhZmZhZTAyMWFiOTVkZTMxZDY2ZGJiYjY1NDcyOTJkNWI1YjNkYzVmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.10hxp4Bovsg1rUGwqWt7ByT8JPaTT3yAwy9e_4Cb0vg">
> 
> Logger: homeassistant.helpers.frame Source: helpers/frame.py:335 First occurred: 22:39:18 (4 occurrences) Last logged: 22:39:18
> 
> Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JXB60M11G1Q0JEKJJE29Y1ND'), with device info: {'identifiers': {('ocpp', 'charger')}, 'via_device': ('ocpp', '01JXB60M11G1Q0JEKJJE29Y1ND')} at custom_components/ocpp/sensor.py, line 77: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JXB60M11G1Q0JEKJJE29Y1ND'), with device info: {'identifiers': {('ocpp', 'charger')}, 'via_device': ('ocpp', '01JXB60M11G1Q0JEKJJE29Y1ND')} at custom_components/ocpp/switch.py, line 75: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JXB60M11G1Q0JEKJJE29Y1ND'), with device info: {'identifiers': {('ocpp', 'charger')}, 'via_device': ('ocpp', '01JXB60M11G1Q0JEKJJE29Y1ND')} at custom_components/ocpp/number.py, line 69: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JXB60M11G1Q0JEKJJE29Y1ND'), with device info: {'identifiers': {('ocpp', 'charger')}, 'via_device': ('ocpp', '01JXB60M11G1Q0JEKJJE29Y1ND')} at custom_components/ocpp/button.py, line 57: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues

I can confirm this issue. Charger is also EVBOX Elvi. It took a long time to find out that it depends on the new release and wasn't my fault :/

closed 2025-08-31 by github-actions[bot]