lbbrhzn/ocpp #1619: Grizzle Smart Connect issue with OCPP

grayfox107, 2025-04-20:
**Describe the bug**
When using OCPP to connect with the:

Model: Grizzl-E Smart Connect

Manufacturer: UnitedChargers

Serial Number: GRS-xxxxxxxxxx

Firmware WiFi Module: GWM-07.027-03

Firmware Power Board: GCW-10.22-05.7

Home assistant is able to read the data just fine. However any attempt to click Availability, Charge Control, Maximum Current, or the unlock button causes an error to pop up. 

If any thing else is needed I can try and replicate it. 

Error Logs:

Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:245
integration: Home Assistant WebSocket API ([documentation](https://www.home-assistant.io/integrations/websocket_api), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+websocket_api%22))
First occurred: 5:26:51 AM (1 occurrences)
Last logged: 5:26:51 AM

[546955896160] Unexpected exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/ocpp/messages.py", line 239, in _validate_payload
    validator.validate(message.payload)
    ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/jsonschema/validators.py", line 451, in validate
    raise error
jsonschema.exceptions.ValidationError: 'Accepted' is not one of ['Unlocked', 'UnlockFailed', 'NotSupported']

Failed validating 'enum' in schema['properties']['status']:
    {'type': 'string',
     'additionalProperties': False,
     'enum': ['Unlocked', 'UnlockFailed', 'NotSupported']}

On instance['status']:
    'Accepted'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2837, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/button/__init__.py", line 132, in _async_press_action
    await self.async_press()
  File "/config/custom_components/ocpp/button.py", line 93, in async_press
[... 122 lines cut ...]

Unable to parse message: '[3,"abac2a44-d8d9-48ac-90aa-a62c09e6462a",{"configurationKey":],"unknownKey":["ChargingScheduleAllowedChargingRateUnit"]}]', it doesn't seem to be valid OCPP: FormatViolationError: Payload for Action is syntactically incorrect or structure for Action, {'cause': 'Message is not valid JSON', 'ocpp_message': '[3,"abac2a44-d8d9-48ac-90aa-a62c09e6462a",{"configurationKey":],"unknownKey":["ChargingScheduleAllowedChargingRateUnit"]}]'}
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/ocpp/messages.py", line 87, in unpack
    msg = json.loads(msg)
  File "/usr/local/lib/python3.13/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ~~~~~~~~~~~~~~~~~~~~~~~^^^
  File "/usr/local/lib/python3.13/json/decoder.py", line 345, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/json/decoder.py", line 363, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 63 (char 62)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 260, in route_message
    msg = unpack(raw_msg)
  File "/usr/local/lib/python3.13/site-packages/ocpp/messages.py", line 89, in unpack
    raise FormatViolationError(
        details={"cause": "Message is not valid JSON", "ocpp_message": msg}
    )
ocpp.exceptions.FormatViolationError: FormatViolationError: Payload for Action is syntactically incorrect or structure for Action,  {'cause': 'Message is not valid JSON', 'ocpp_message': '[3,"abac2a44-d8d9-48ac-90aa-a62c09e6462a",{"configurationKey":],"unknownKey":["ChargingScheduleAllowedChargingRateUnit"]}]'}

xblackedemonx, 2025-04-20:
I have the exact same charger with the same firmware and I have the same issue.

Model: Grizzl-E Smart Connect

Manufacturer: UnitedChargers

Serial Number: GRS-xxxxxxxxxx

Firmware WiFi Module: GWM-07.027-03

Firmware Power Board: GCW-10.22-05.7

drc38, 2025-04-20:
The charger is not following the OCPP standard, you'll need to see if there's a newer firmware that fixes the issue. Or search the PRs and discussions here I believe there's a fork with workarounds for the Grizzl-E

xblackedemonx, 2025-04-20:
Hi, I tried the fork and older versions but nothing works.

closed 2025-06-27 by github-actions[bot]