lbbrhzn/ocpp #1620: ValueError: Config entry central (...) for ocpp.sensor has already been setup!

KristianOellegaard, 2025-04-20:
**Describe the bug**
A clear and concise description of what the bug is.

When starting the integration I get

Logger: homeassistant.config_entries
Source: config_entries.py:751
First occurred: 9:33:43 PM (12 occurrences)
Last logged: 9:53:38 PM

Error setting up entry central for button
Error setting up entry OCPP Host for sensor
Error setting up entry OCPP Host for switch
Error setting up entry OCPP Host for number
Error setting up entry OCPP Host for button
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 751, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 97, in async_setup_entry
    return await hass.data[DATA_COMPONENT].async_setup_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 182, in async_setup_entry
    raise ValueError(
    ...<2 lines>...
    )
ValueError: Config entry central (...) for ocpp.sensor has already been setup!

**To Reproduce**
Steps to reproduce the behavior:
1. Add the integration
2. See the charger be discovered
3. Click "Add device"
4. All entities are unavailable and see this error on console

**Additional context**
Add any other context about the problem here.

Some times it seems to work to reboot HA and the charger but then after some hours usually 8 hours or so it fails again

RangerDjanger, 2025-04-22:
I have the same problem...

KristianOellegaard, 2025-04-22:
I managed to get the integration working again yesterday without fixing the underlying problem. I had an EVBox Elvi that also had a simcard and I disabled it. I no longer see the error so I believe it is related to when it has connectivity issues and cannot recover.

I believe there is still a bug that needs to be addressed, but it could be that the exception happens as consequence of something different than what I originally thought

rbb-dev, 2025-04-22:
I think you need to delete occp central entry and create is as another name.
its complaining that its still there.

aaronegger, 2025-05-14:
I have the same issue..

aaronegger, 2025-05-14:
Maybe this correlates with this: [https://github.com/home-assistant/core/pull/116482](https://github.com/home-assistant/core/pull/116482)

Pseudophryne, 2025-06-25:
This error occurred for me on upgrade to HA Core 2025.6.3, reverting back to 2025.5.3 no error experienced.

````
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 99, in async_setup_entry
    return await hass.data[DATA_COMPONENT].async_setup_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 219, in async_setup_entry
    raise ValueError(
    ...<2 lines>...
    )
ValueError: Config entry central (01JYJJ62KZQ4KZYE8BJX0X3Q5D) for ocpp.sensor has already been setup!

Pseudophryne, 2025-06-26:
All, I found after a lot of testing/trail/error that applying v0.8.3 fixed my issue completely, see #1641

closed 2025-09-03 by github-actions[bot]