lbbrhzn/ocpp #1621: "Warning: Set availability failed with response Rejected"

RangerDjanger, 2025-04-22:
I have installed the latest OCPP version.

I have a new EVSwitch (i.e [EN+ Caro Series Home Wallbox].  

Firmware version is 1.4.859.

Home Assistant detected it, and I have added it.

But I noticed a warning in notifications - "Warning: Set availability failed with response Rejected".

All of the entities are unavaialble.

So need some help.

Thanks.

BJReplay, 2025-04-22:
See set up notes [here](https://github.com/lbbrhzn/ocpp/wiki/supported-devices#en-caro-series-home-wallbox)

Note that the detected charger may have a charger ID of charger rather than the serial number - you will need to change it.

Note that you don't have the latest firmware, so probably want to update it first.

RangerDjanger, 2025-04-22:
Ok thanks heaps.  I’ll try changing to just ‘charger’.

What is the latest firmware ? I used the EVchargo app and it said I had the latest version.

BJReplay, 2025-04-22:
> What is the latest firmware ?

> See set up notes here

RangerDjanger, 2025-04-23:
Thanks.   Overnight HA discovered the charger.   I am still getting warnings on not being able to set availability.

I can charge the car.

The firmware in your post is current firmware 1.0.25.130.

The firmware I have is 1.4.859.  This number looks much higher than what is indicated on the guide.

Here is a screen shot of a dump of all the entities.

[image]

BJReplay, 2025-04-23:
> The firmware in your post is current firmware 1.0.25.130.
> 
> The firmware I have is 1.4.859. This number looks much higher than what is indicated on the guide.

Looks can be deceiving.  25 > 4

RangerDjanger, 2025-04-23:
25 is the 3rd digit.    4 is the second digit.

4 > 0.

Am I going crazy here?

BJReplay, 2025-04-23:
> Am I going crazy here?

Dude, I'm trying to help.

You are trying to find as many reasons as you can to **not** follow the instructions on **how to upgrade your firmware**.

Let me quote from the page, since you seem determined not to actually read it there:

_For firmware versions earlier than 1.0.25.130 the only way you can update firmware is by connecting to the evchargo OCPP server at wss://ocpp16.evchargo.com:33033/ and emailing your serial number to support@en-plus.com.cn requesting that your firmware is updated._

Have you done this?

Have you set the OCPP server to `wss://ocpp16.evchargo.com:33033/` and emailed `support@en-plus.com.cn` or have you just figured that you've got the latest firmware?

Or are you the firmware developer, and happen to know what version is the latest?

But you do you.  You choose to complain about the error that is fixed by firmware 1.0.25.130 and you choose to argue that you think that 1.4.859 is newer, even though it isn't.

If you think the instructions are wrong, and the latest firmware is different, update the instructions.

RangerDjanger, 2025-04-23:
Hey BJ, thanks - I do really appreciate the support here. Yes, followed the guide. I was just super confused that 1.0.25.130 is newer than 1.4.859.  It didnt make sense  based on my software dev experience of major.minor.revision.build versioning standards.  

I did request the firmware upgrade 48 hours ago - but no reply.  Do they respond or just do the upgrade.

Note my interface on the phone app does have an option to upgrade firmware. It said it was on the latest. So once again why I was confused by the instructions a little.

RangerDjanger, 2025-04-23:
I’ve reconfigured OCCP back to Evchargo.  Their app has a firmware update section that says it’s already on the latest. Hence my confusion again. I didn’t know how new/old your user guide was.

Anyway I’ll stick to your direct instructions. 

[image]

BJReplay, 2025-04-23:
> I did request the firmware upgrade 48 hours ago - but no reply. Do they respond or just do the upgrade.?

They respond - but unless your charger is configured to their OCPP server, they can't upgrade it.

> Their app has a firmware update section that says it’s already on the latest.

It always says that.

That's why you have to email support.

RangerDjanger, 2025-04-23:
Thanks again for your help.

BTW - Keen see the automations you have created for smart charging / egress solar.  Are you able to share them please?

RangerDjanger, 2025-04-23:
Problem solved.  Support team applied the update. Sorry for the confusion and appreciate the help @BJReplay.

closed 2025-04-23 by RangerDjanger

Delta1977, 2025-05-17:
One quick question:
How long take  support to respond when url is set and email is send ?