lbbrhzn/ocpp #1639: AttributeError: 'WebSocketServerProtocol' object has no attribute 'request'

ghulleman, 2025-06-12:
**Describe the bug**
After start, the logs are filled with an error:

ERROR (MainThread) [websockets.server] connection handler failed
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/websockets/legacy/server.py", line 236, in handler
    await self.ws_handler(self)
  File "/config/custom_components/ocpp/api.py", line 221, in on_connect
    _LOGGER.info(f"Charger websocket path={websocket.request.path}")
                                           ^^^^^^^^^^^^^^^^^
AttributeError: 'WebSocketServerProtocol' object has no attribute 'request'
2025-06-12 09:07:23.945 ERROR (MainThread) [websockets.server] connection handler failed
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/websockets/legacy/server.py", line 236, in handler
    await self.ws_handler(self)
  File "/config/custom_components/ocpp/api.py", line 221, in on_connect
    _LOGGER.info(f"Charger websocket path={websocket.request.path}")

**To Reproduce**
Steps to reproduce the behavior:
Start the device. 

**Expected behavior**
Not give errors in the log

**Screenshots**
n/a

**Desktop (please complete the following information):**
Not relevant

**Smartphone (please complete the following information):**
Not relevant

**Additional context**
The charger is an Alfen s-line

ericsche, 2025-06-13:
I do have exactly the same error.
wonder if it was not introduced with 2025.6

ghulleman, 2025-06-15:
I was unable to set the charging limit on the charger due to this error. It was 0, could not charge at all. 

I am not a python developer, but I managed to get the charger to work again by changing in API.py, line 221:

Change
        _LOGGER.info(f"Charger websocket path={websocket.request.path}")
        cp_id = websocket.request.path.strip("/")

into
        if isinstance(websocket, ServerConnection):
            _LOGGER.info(f"Charger websocket path={websocket.request.path}")
            cp_id = websocket.request.path.strip("/")
        else:
            _LOGGER.info(f"Charger websocket path={websocket.path}")
            cp_id = websocket.path.strip("/")

Keep in mind the indentations. This is in the method, async def on_connect(self, websocket: ServerConnection). The 'websocket' is sometimes of type 'ServerConnection', but also sometimes of type 'WebSocketServerProtocol'.

drc38, 2025-06-15:
Looks like the underlying issue is you are not on the latest version of websockets https://websockets.readthedocs.io/en/stable/ the legacy implementation used in older versions has been deprecated.

Plawasan, 2025-06-16:
How does one get to the latest version? I've also lost control over my charger after upgrading to 2025.6, I've tried the workaround on my test system and I'm still not able to get a working device in HA (all entities are unavailable) even though I see the charger communiating in the server logs and it does pop up as a new endpoint. I'll collect more data and will open a separate issue for this.

ghulleman, 2025-06-16:
> How does one get to the latest version? I've also lost control over my charger after upgrading to 2025.6, I've tried the workaround on my test system and I'm still not able to get a working device [...]

Did you restart HA after the change? Just editing the code in studio code editor gave the same error with a restart of the addon or reload yaml files did not work. If you still have the 'File "/config/custom_components/ocpp/api.py", line 221, in on_connect _LOGGER.info(f"Charger websocket path={websocket.request.path}")' error, I believe this to be the case if you changed the code (the line shifts, so t should not be line 221 anymore). I had to restart HA before changes were applied. 

Please send the log entry you have on your end.

tmenguy, 2025-06-17:
same for me, on latest HA 2025.6.1, same error but has it is my first ocpp setup, I get in the logs too :

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/websockets/legacy/server.py", line 236, in handler
    await self.ws_handler(self)
  File "/config/custom_components/ocpp/api.py", line 221, in on_connect
    _LOGGER.info(f"Charger websocket path={websocket.request.path}")
                                           ^^^^^^^^^^^^^^^^^
AttributeError: 'WebSocketServerProtocol' object has no attribute 'request'

tmenguy, 2025-06-18:
Did the change above on the api.py ... it detects my wallbox but everything (all entities, sensors, button etc) are greyed out (unavailable), and ai get a lot of error like this at the component entry setup: 

ValueError: Config entry central (01JY04Y7PV5XE9ZBSWCZQ00SQ5) for ocpp.sensor has already been setup!

tmenguy, 2025-06-18:
Tried to 6.0.4 version with no luck ... after I have a wallbox pulsar PRO socket not sure it changes how it works

Niek, 2025-07-16:
Same issue here on HA 2025.7.2. Any fix?

tmenguy, 2025-07-18:
I do have a repo I maintain with a fix for this and another fix for the discovery: https://github.com/tmenguy/ocpp I need to do a pr on the main one for this, will do one time :) but if you want a quick fix, you can use it

danmiles86, 2025-07-21:
Thanks @tmenguy 
This resolved it for me.

closed 2025-09-27 by github-actions[bot]