lbbrhzn/ocpp #1641: charger connects, but all buttons greyed out and error in logs

chrismelba, 2025-06-17:
**Describe the bug**
After working fine for nearly 12 months my charger has been offline for 2 days now. I've tried reinstalling OCPP and restarting both HA and the charger, but am getting errors

**To Reproduce**
Unsure if it can be reproduced outside my machine, but the bug in the logs happens when I fresh install OCPP

**Expected behavior**
All controls working

**Screenshots**
Logger: homeassistant.config_entries
Source: config_entries.py:749
First occurred: 6:01:42 PM (8 occurrences)
Last logged: 6:09:57 PM

Error setting up entry central for button
Error setting up entry new-ocpp for sensor
Error setting up entry new-ocpp for switch
Error setting up entry new-ocpp for number
Error setting up entry new-ocpp for button
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 99, in async_setup_entry
    return await hass.data[DATA_COMPONENT].async_setup_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 219, in async_setup_entry
    raise ValueError(
    ...<2 lines>...
    )
ValueError: Config entry central (01JXYEGJMA1VK9KXVN937ZZTYV) for ocpp.sensor has already been setup!

chrismelba, 2025-06-17:
update: Downgraded to 0.6.4 (which i think was the first version I installed) and now it's working. Would be nice to go to the latest version though

mg-arne, 2025-06-18:
Same for me using an Evbox. Downgrade to 0.8.3 workes for me as well.

drc38, 2025-06-19:
Seems odd as there is very little change between those releases: https://github.com/lbbrhzn/ocpp/compare/v0.8.3...v0.8.4

The entry data in HA may have got corrupted, you could try uninstalling, rebooting HA (confirm there is no ocpp entry) then reinstalling integration again.

chrismelba, 2025-06-19:
> v0.8.3...v0.8.4 <https://github.com/lbbrhzn/ocpp/compare/v0.8.3...v0.8.4>

I did try those steps and got the same error

Pseudophryne, 2025-06-26:
I have the same issue as reported by others; to test I did a fresh install of OCPP on my Test system and pointed the charger to my test system and experienced the same issue with all sensors unavailable, reverting to 0.8.3 and all works as expected.

Post testing on my test system I reverted my prod system to 0.8.3 and my charger (ZJ Beny) is back online and working as expected.

Test system config 
````
Installation method = Home Assistant OS
Core = 2025.7.0b0
Supervisor = 2025.06.2
Operating System = 16.0.rc1
Frontend = 20250625.0
````
Prod system config
````
Installation method = Home Assistant OS
Core = 2025.6.3
Supervisor = 2025.06.2
Operating System = 15.2
Frontend = 20250531.4
````

drc38, 2025-06-26:
@Mark-Hetherington any ideas on why the deepcopy is causing this?

ExcelsiorEyer, 2025-07-06:
Have a Wallbox Pulsar. I can confirm that this is not just migration. Deleting and re-adding the integration isn't working either on 0.8.4. 

Set it up fresh on 0.8.4 and all entities showed Unavailable and grayed out. Downgraded to 0.8.3 and everything is back to normal. Didn't even have to delete/readd the device. Started working as-is on 0.8.3

closed 2025-10-19 by github-actions[bot]