lbbrhzn/ocpp #1649: Factory reset or remove OCPP configuration from Autel MaxiCharger

jproveta, 2025-06-30:
Hi there,

I'm using your OCPP server with my Autel MaxiCharger, but now I want to reset the charger back to its factory configuration and remove the current OCPP endpoint .

Someone know the proper way to fully remove the OCPP configuration from the charger?

Also, is there a way via your server to send a factory reset or another command that would clear the OCPP settings?

Thanks in advance!

jproveta, 2025-07-02:
> [@jproveta](https://github.com/jproveta) i dont think you can.

So I'll have the equipment stuck to the OCPP?

Thanks

jproveta, 2025-07-02:
> You have to log in to the charger to reset its settings in its interface.

I've already tried to do a factory reset in the Autel Charge app, but it continues with the connection and configuration to OCPP.

mavnz72, 2025-07-16:
You need to use the Autel phone app to change the occp server to the Autel one.  Just pick it from the list and it will replace the custom setting you currently have.