lbbrhzn/ocpp #1652: Grizzl-E

emitclever, 2025-07-04:
Today died OCP I had to restart the HA. I do my best to report it since I had not seen any report about this charger here. 

25-07-04 10:01:20.797 WARNING (Thread-1 (_monitor)) [homeassistant.util.logging] Module ocpp is logging too frequently. 200 messages since last count
2025-07-04 10:01:21.033 INFO (MainThread) [ocpp] GRS-17000054077: receive message [3,"57630bc2-b959-4cb1-ac13-9e91b9b750d8",{"status":"NotSupported"}]
2025-07-04 10:01:21.035 INFO (MainThread) [ocpp] GRS-17000054077: send [2,"a205ed65-1420-4f40-8279-4b4f04e1a607","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Export"}]
2025-07-04 10:01:21.251 INFO (MainThread) [ocpp] GRS-17000054077: receive message [3,"a205ed65-1420-4f40-8279-4b4f04e1a607",{"status":"NotSupported"}]
2025-07-04 10:01:21.253 INFO (MainThread) [ocpp] GRS-17000054077: send [2,"0981b46e-8623-4298-a96e-33ca484b7ae1","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Import"}]
2025-07-04 10:01:21.542 INFO (MainThread) [ocpp] GRS-17000054077: receive message [3,"0981b46e-8623-4298-a96e-33ca484b7ae1",{"status":"NotSupported"}]
2025-07-04 10:01:21.545 INFO (MainThread) [ocpp] GRS-17000054077: send [2,"8385064c-65b9-46e8-8cb8-2ecf3d8d8f89","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Factor"}]
2025-07-04 10:01:21.751 INFO (MainThread) [ocpp] GRS-17000054077: receive message [3,"8385064c-65b9-46e8-8cb8-2ecf3d8d8f89",{"status":"NotSupported"}]
2025-07-04 10:01:21.753 INFO (MainThread) [ocpp] GRS-17000054077: send [2,"698b9dfd-5241-4d75-a19e-e5c644d9a1d9","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Offered"}]
2025-07-04 10:01:22.020 INFO (MainThread) [ocpp] GRS-17000054077: receive message [3,"698b9dfd-5241-4d75-a19e-e5c644d9a1d9",{"status":"NotSupported"}]
2025-07-04 10:01:22.022 INFO (MainThread) [ocpp] GRS-17000054077: send [2,"dbf9a926-dbef-49f4-b618-792575b70a5a","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Reactive.Export"}]
2025-07-04 10:01:22.717 INFO (MainThread) [ocpp] GRS-17000054077: receive message [3,"dbf9a926-dbef-49f4-b618-792575b70a5a",{"status":"NotSupported"}]
2025-07-04 10:01:22.719 INFO (MainThread) [ocpp] GRS-17000054077: send [2,"e57d49da-088b-430a-b7db-9f9126107575","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Reactive.Import"}]
2025-07-04 10:01:23.018 INFO (MainThread) [ocpp] GRS-17000054077: receive message [3,"e57d49da-088b-430a-b7db-9f9126107575",{"status":"NotSupported"}]
2025-07-04 10:01:23.020 INFO (MainThread) [ocpp] GRS-17000054077: send [2,"44c05adb-2b8e-4c6f-aaf6-7287171a8eab","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"RPM"}]
2025-07-04 10:01:23.290 INFO (MainThread) [ocpp] GRS-17000054077: receive message [3,"44c05adb-2b8e-4c6f-aaf6-7287171a8eab",{"status":"NotSupported"}]
2025-07-04 10:01:23.292 INFO (MainThread) [ocpp] GRS-17000054077: send [2,"49b10f21-793f-4a57-93bd-4d7f83d60d7c","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"SoC"}]
2025-07-04 10:01:23.558 INFO (MainThread) [ocpp] GRS-17000054077: receive message [3,"49b10f21-793f-4a57-93bd-4d7f83d60d7c",{"status":"NotSupported"}]
2025-07-04 10:01:23.560 INFO (MainThread) [ocpp] GRS-17000054077: send [2,"6936df7c-a400-4c09-91c9-b6f0fced9035","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Temperature"}]
2025-07-04 10:01:23.658 INFO (MainThread) [ocpp] GRS-17000054077: receive message [3,"6936df7c-a400-4c09-91c9-b6f0fced9035",{"status":"NotSupported"}]
2025-07-04 10:01:23.660 INFO (MainThread) [ocpp] GRS-17000054077: send [2,"eccf80fe-163e-4c9f-a222-6aba8b1efb6f","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Voltage"}]
2025-07-04 10:01:23.790 INFO (MainThread) [ocpp] GRS-17000054077: receive message [3,"eccf80fe-163e-4c9f-a222-6aba8b1efb6f",{"status":"NotSupported"}]
2025-07-04 10:01:23.792 INFO (MainThread) [ocpp] GRS-17000054077: send [2,"4e4fbc62-5007-4ca4-9aa4-10927d891546","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2025-07-04 10:01:23.908 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/queues.py", line 186, in get
    await getter
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 421, in call
    response = await self._get_specific_response(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        call.unique_id, self._response_timeout
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/ocpp/chargepoint.py", line 347, in _get_specific_response
    resp = await super()._get_specific_response(unique_id, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 455, in _get_specific_response
    response = await asyncio.wait_for(self._response_queue.get(), timeout)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 506, in wait_for
    async with timeouts.timeout(timeout):
               ~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/config/custom_components/ocpp/chargepoint.py", line 233, in post_connect
    accepted_measurands: str = await self.get_supported_measurands()
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/ocppv16.py", line 136, in get_supported_measurands
    chgr_measurands = await self.get_configuration(key)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/ocppv16.py", line 498, in get_configuration
    resp = await self.call(req)
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 425, in call
    raise asyncio.TimeoutError(
    ...<2 lines>...
    )
TimeoutError: Waited 10s for response on [2,"3775df4f-a15a-4f46-878a-f9d36bdea9b0","GetConfiguration",{"key":["MeterValuesSampledData"]}].
[... 43 lines cut ...]
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 455, in _get_specific_response
    response = await asyncio.wait_for(self._response_queue.get(), timeout)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 506, in wait_for
    async with timeouts.timeout(timeout):
               ~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/config/custom_components/ocpp/chargepoint.py", line 233, in post_connect
    accepted_measurands: str = await self.get_supported_measurands()
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/ocppv16.py", line 136, in get_supported_measurands
    chgr_measurands = await self.get_configuration(key)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/ocppv16.py", line 498, in get_configuration
    resp = await self.call(req)
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 425, in call
    raise asyncio.TimeoutError(
    ...<2 lines>...
    )
TimeoutError: Waited 10s for response on [2,"4e4fbc62-5007-4ca4-9aa4-10927d891546","GetConfiguration",{"key":["MeterValuesSampledData"]}].

emitclever, 2025-07-10:
Frequent reboot was do to a network issue. Low signal, the esp controller is enclosed into a complete metallic box - not need to be an engineer, so  more stable now but the errors are still there I suppose from what I've been read, issues with grizlee ocp implementation.

emitclever, 2025-08-16:
I do not know the menu for grizzlee ultimate 48. Mine was is Grizlee-E or Smart Grizlee 40 A. So is not the same code behind.
once you are on main menu, send me the source code by right click on the page in browser.

emitclever, 2025-08-16:
the python code take advantage of this part in html source code, if you have it in the main page source code may work. Only the ocppUrl has to be changed. Keep a backup of the original exact path if you want to reverse.
Even today the firmware was not fixed, so I think these guys are not serious until proven. I may never update this thing, since work fine with these tricks.

		applyChangesButton.addEventListener('click', () => {
			const formData = new FormData()
			formData.append('ocppUrl', ocppUrlInput.value);
			formData.append('authKey', authKeyInput.value);
			formData.append('stationId', stationIdInput.value);
			sendReq('/ocpp', (data) => {
				closeOCPPModal.style.display = 'none'
				alert('Changes applied')
			}, {
				method: 'POST',
				payload: new URLSearchParams(formData).toString(),
				contentType: 'application/x-www-form-urlencoded'
			})
			initialBEValues.ocppUrl = ocppUrlInput.value;
			initialBEValues.authKey = authKeyInput.value;
			initialBEValues.stationId = stationIdInput.value;
			closeOCPPModalHandler();
		})

shortyjacobs, 2025-08-16:
I'm sorry, I'm no expert, so I've been trying to get ChatGPT to help me out.  I attached the pagesource as emitclever asked.  Some stumbles, but not bad overall....here's what we've done:

What we’re working with [(full chat log for masochists here)](https://chatgpt.com/share/68a0fded-d7a4-8001-8756-718b3603e868)

<html>
<body>

[view-source_192.168.181.166.html](https://github.com/user-attachments/files/21818034/view-source_192.168.181.166.html)

<html><head></head><body><h1>Grizzl-E OCPP Debug Session — Findings &amp; Attempts (No New Steps)</h1>
<blockquote>
<p><strong>TL;DR:</strong> On Wi-Fi FW <code inline="">ULGRW000A-03.14.1</code>, config writes for Custom/Local OCPP <strong>succeed and persist</strong> (<code inline="">ocppVendor=255</code> or <code inline="">254</code>), but the charger <strong>never initiates a socket</strong> to the local OCPP server. Packet captures show <strong>no traffic</strong> to the HA listener, even with cloud egress blocked.</p>
</blockquote>
<hr>
<h2>Device &amp; Environment</h2>
<ul>
<li>
<p><strong>Charger IP:</strong> <code inline="">192.168.181.166</code></p>
</li>
<li>
<p><strong>Firmware</strong></p>
<ul>
<li>
<p><strong>Wi-Fi:</strong> <code inline="">ULGRW000A-03.14.1</code></p>
</li>
<li>
<p><strong>EVSE:</strong> <code inline="">GRU077L-01.07.3</code></p>
</li>
</ul>
</li>
<li>
<p><strong>Home Assistant OCPP Central:</strong> <code inline="">192.168.181.14:7274</code></p>
<ul>
<li>
<p>Known-good reference from another charger:<br>
<code inline="">ws://192.168.181.14:7274/ocpp1.6?station=rivian</code></p>
</li>
<li>
<p>Target for this unit:<br>
<code inline="">ws://192.168.181.14:7274/ocpp1.6?station=polestar</code></p>
</li>
</ul>
</li>
<li>
<p><strong>Connectivity check (HA listener):</strong></p>
<pre><code class="language-powershell">Test-NetConnection 192.168.181.14 -Port 7274
# =&gt; TcpTestSucceeded : True
</code></pre>
</li>
</ul>
<hr>
<h2>What <code inline="">/main</code> Reports</h2>
<ul>
<li>
<p><strong>Fields seen:</strong> <code inline="">stationId</code>, <code inline="">ocppEnabled</code>, <code inline="">ocppconnected</code>, <code inline="">ocppVendor</code>, <code inline="">ocppOfflineAva</code>, plus EVSE telemetry.</p>
</li>
<li>
<p><strong>Notably absent:</strong> any <code inline="">ocppUrl</code>/host/port/path readback fields.</p>
</li>
<li>
<p><strong>Baseline values observed over the session:</strong></p>
<ul>
<li>
<p><code inline="">stationId</code>: started as <code inline="">GRU-L0000361539</code>; we also set/tested <code inline="">polestar</code>.</p>
</li>
<li>
<p><code inline="">ocppEnabled</code>: toggles between <code inline="">0</code> and <code inline="">1</code> as commanded.</p>
</li>
<li>
<p><code inline="">ocppVendor</code>: initially <code inline="">1</code>; we successfully set many different values (see below).</p>
[... 224 lines cut ...]
<p><strong>No traffic</strong> observed from the charger to <code inline="">192.168.181.14:7274</code> during attempts.</p>
</li>
</ul>
</li>
<li>
<p>Network:</p>
<ul>
<li>
<p>WAN egress for the charger is blocked (app shows <strong>offline</strong>).</p>
</li>
</ul>
</li>
<li>
<p>HA:</p>
<ul>
<li>
<p>OCPP central running; known-good with another charger; no handshake attempts from this unit observed.</p>
</li>
</ul>
</li>
</ul>
<hr>
<p><em>End of recap. Happy to attach this as a file if needed for a support ticket or GitHub issue trail.</em></p></body></html>
</body>
</html>

emitclever, 2025-08-17:
woau, impressive report. I am not expert either but the source shared it's a complete different source page for this unit. Compare with mine (attached) where all the js scripts <script> are on the same page (at the end of the page) after HTML part and "accessible" to take advantage. So, it's obvious the previous python script will not work. I have updated my comment there to specify the exact applicable model.
May answers lies here in this file <script src="[./js/script.js](http://192.168.181.166/js/script.js)"></script> on your source code
in your source code the part of java scripts is a separate file.
here is mine HTML and js script in the same file
[Grizzle-e 40A page source.txt](https://github.com/user-attachments/files/21820032/Grizzle-e.40A.page.source.txt)

emitclever, 2025-08-17:
Reading the chatgpt log I saw this ? 
[4] Setting OCPP URL to ws://192.168.181.14:7274/ocpp1.6?station=polestar ...
Why the port 7274 abd alll the path after 

so you manage to updated the ocpp address from united chargers domain to an internal one ?
on my setup is the only thing required on ocpp setup to communicate with HA is this, that's all
ocpp_url_value = ' ws://192.168.XXX.XXX:9000/'

closed 2025-12-23 by github-actions[bot]