lbbrhzn/ocpp #1703: After 0.10.0 update, SetChargingProfile rejected, charger no longer accepts profile

haroldevelop, 2025-09-06:
After updating to the latest version of the OCPP custom integration in HACS, my charger started rejecting SetChargingProfile requests. This used to work flawlessly until the update.

Details:
Home Assistant version: 2025.9.1
OCPP integration version: Version v0.10.0
Previous working version: Version v0.9.0
Charger model: EV-box
Protocol: OCPP 1.6 JSON

Observed behavior:
Before the update, changing the entity number.charger_maximum_current correctly sent a SetChargingProfile and the charger accepted it.
After the update, HA still sends the message, but the charger responds with Rejected.

Log excerpt:
2025-09-05 17:26:56.913 INFO (MainThread) [ocpp] EVB-P21280000: send [2,"b81b0376-0d78-4d2e-b0f9-8166c2844c4e","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":6,"chargingProfileKind":"Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":10.0}]}}}]
2025-09-05 17:26:56.964 INFO (MainThread) [ocpp] EVB-P21280000: receive message [3,"b81b0376-0d78-4d2e-b0f9-8166c2844c4e",{"status":"Rejected"}]

Expected behavior:
Charger should accept the profile, as it did with the previous integration version.

Workaround:
Rolling back to the previous OCPP integration version resolves the issue.

bleara, 2025-09-06:
I have tried to revert to 0.9.0 but that has failed with :-
2025-09-06 09:58:25.022 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
2025-09-06 09:58:25.022 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/
2025-09-06 09:58:25.023 ERROR (MainThread) [custom_components.ocpp] Failed to setup charger : ChargerSystemSettings.__init__() got an unexpected keyword argument 'num_connectors'
I guess I'll have to remove OCPP and do a temporary new re-install of 0.9.0 to get it working

Debug from 0.10.0 was :
2025-09-06 08:23:48.725 INFO (MainThread) [ocpp] : send [2,"a4b1eee8-657f-4c68-b6c7-f3662d8f925b","GetConfiguration",{"key":["ChargingScheduleAllowedChargingRateUnit"]}]
2025-09-06 08:23:48.804 INFO (MainThread) [ocpp] : receive message [3,"a4b1eee8-657f-4c68-b6c7-f3662d8f925b",{"configurationKey":[{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current"}]}]
2025-09-06 08:23:48.807 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargingScheduleAllowedChargingRateUnit: Current
2025-09-06 08:23:48.812 INFO (MainThread) [ocpp] : send [2,"afefb069-77a5-4b10-95b5-c312e282aa97","GetConfiguration",{"key":["ChargeProfileMaxStackLevel"]}]
2025-09-06 08:23:48.901 INFO (MainThread) [ocpp] : receive message [3,"afefb069-77a5-4b10-95b5-c312e282aa97",{"configurationKey":[{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"0"}]}]
2025-09-06 08:23:48.903 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargeProfileMaxStackLevel: 0
2025-09-06 08:23:48.906 INFO (MainThread) [ocpp] : send [2,"7270fb69-900e-4bbb-8128-2691cf681069","ClearChargingProfile",{"chargingProfilePurpose":"ChargePointMaxProfile"}]
2025-09-06 08:23:49.001 INFO (MainThread) [ocpp] : receive message [4,"7270fb69-900e-4bbb-8128-2691cf681069","InternalError","",""]
  File "/config/custom_components/ocpp/number.py", line 218, in async_set_native_value
  File "/config/custom_components/ocpp/api.py", line 532, in set_max_charge_rate_amps
  File "/config/custom_components/ocpp/ocppv16.py", line 391, in set_charge_rate
  File "/config/custom_components/ocpp/ocppv16.py", line 323, in clear_profile
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 421, in call
  File "/config/custom_components/ocpp/chargepoint.py", line 451, in _get_specific_response
ocpp.exceptions.InternalError: InternalError: ,

haroldevelop, 2025-09-06:
I think you need this part:

2025-09-06 10:28:00.308 INFO (MainThread) [ocpp] EVB-P21280000: send [2,"dd355c72-eab9-4ad2-aafa-f5e51866ee02","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":5,"chargingProfileKind":"Relative","chargingProfilePurpose":"TxDefaultProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":1.0}]}}}]
2025-09-06 10:28:02.206 INFO (MainThread) [ocpp] EVB-P21280000: receive message [3,"dd355c72-eab9-4ad2-aafa-f5e51866ee02",{"status":"Accepted"}]

troystanway, 2025-09-07:
Also experiencing this issue- using Schneider EVLink. since .10 unable to set current offered to 0 without getting a rejected message.

moorgrove, 2025-09-07:
With a Ctek Chargestorm 2 I can not change the maximum current entity.

closed 2025-09-08 by drc38

jeremycoppey, 2025-09-14:
> Also experiencing this issue- using Schneider EVLink. since .10 unable to set current offered to 0 without getting a rejected message.

@troystanway Did you ever get it working on your Schneider EVLink? Even before the 0.10.0 version?
I have this one: https://www.se.com/be/en/product/EVH3S22P0CK/evlink-wallbox-plus-t2-attached-cable-3-phase-32a-22kw/
But i don't get it to adjust the number.charger_maximum_current in the past. 

@drc38 is this already fix included in 0.10.2?

troystanway, 2025-09-16:
@jeremycoppey I have evb1a22pcri and on v0.9.0 I am able to vary the charging maximum current.
Perhaps there is a config you need to change or you need to firmware update.
Be very careful with firmware update, it’s easy to brick. 
The clearest version changes the webgui from http to https

PR #1708 (2025-09-08): Issue fixes for initial release of multiple connector support.
Update EAIR continuously (Fixes #1707). Handle clear profile exceptions when changing max current (Fixes #1706 and fixes #1703). Use proper connector id for extra attr (phase data) (Fixes #1704). Clear IdTag and reset transaction id to 0 on stop/unplug (Fixes #1702). Add more tests. Test coverage settings.

## Summary by CodeRabbit

- New Features
  - Optimistic UI for adjusting current limit (immediate slider response).
  - Improved multi-connector handling, including per-connector phase attributes and energy accounting.
- Bug Fix [... cut]