lbbrhzn/ocpp #1705: LINCHR H-Series charger, debug logs show data however not appearing in HA

arh-au, 2025-09-06:
## Version of the custom_component

Using v0.10.0

## Describe the bug

The meterValues Voltage, Current.Import and Energy.Active.Import aren't retrieved withi HA (show unavailable).

Both entities have message: "This entity is no longer being provided by the ocpp integration. If the entity is no longer in use, delete it in settings."

## Debug log

2025-09-07 08:15:04.220 INFO (MainThread) [ocpp] LINCHR7123050003103101050209A00: receive message [2,"d151265e-07fd-4b7b-847e-c4cb38786fe2","MeterValues",{"connectorId":1,"meterValue":[{"sampledValue":[{"context":"Sample.Periodic","format":"Raw","location":"Inlet","measurand":"Voltage","phase":"L1","unit":"V","value":"241.31"},{"context":"Sample.Periodic","format":"Raw","location":"Outlet","measurand":"Current.Import","phase":"L1","unit":"A","value":"6.39"},{"context":"Sample.Periodic","format":"Raw","location":"Outlet","measurand":"Energy.Active.Import.Register","unit":"Wh","value":"3906335"},{"context":"Sample.Periodic","format":"Raw","location":"Outlet","measurand":"Power.Active.Import","unit":"W","value":"1355"},{"context":"Sample.Periodic","format":"Raw","location":"Outlet","measurand":"Current.Offered","unit":"A","value":"32.00"}],"timestamp":"2025-09-06T22:15:00Z"}],"transactionId":1757196092}]

jthun, 2025-09-07:
How many connectors (outlets) does the charger have? What happens if you reload the integration?

arh-au, 2025-09-07:
Charger has only 1 connector, however below is the log sequence when probed by OCPP, see last log entry Forcing number of connectors to -1.  Note that charging can be started via HA in this state  by toggling "charge control" and vehicle starts charging.

2025-09-07 14:43:01.823 DEBUG (MainThread) [custom_components.ocpp] 'LINCHR7123050003103101050209A00' post connection setup completed successfully
2025-09-07 14:43:01.825 INFO (MainThread) [ocpp] LINCHR7123050003103101050209A00: send [2,"e4150dfd-d20e-47f0-95d7-470519e4da70","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
2025-09-07 14:43:01.831 INFO (MainThread) [ocpp] LINCHR7123050003103101050209A00: receive message [3,"e4150dfd-d20e-47f0-95d7-470519e4da70",{"status":"Accepted"}]
2025-09-07 14:43:01.832 DEBUG (MainThread) [custom_components.ocpp] trigger status notification for connector=0
2025-09-07 14:43:01.833 INFO (MainThread) [ocpp] LINCHR7123050003103101050209A00: send [2,"180121cc-ca8f-482b-8d6a-e854f6952474","TriggerMessage",{"requestedMessage":"StatusNotification","connectorId":0}]
2025-09-07 14:43:01.839 INFO (MainThread) [ocpp] LINCHR7123050003103101050209A00: receive message [3,"180121cc-ca8f-482b-8d6a-e854f6952474",{"status":"Rejected"}]
2025-09-07 14:43:01.841 WARNING (MainThread) [custom_components.ocpp] Failed with response: Rejected
2025-09-07 14:43:01.841 WARNING (MainThread) [custom_components.ocpp] Forcing number of connectors to -1, charger returned 1

jthun, 2025-09-08:
Could you please test the next release and get back with debug logs if you still have issues?

jthun, 2025-09-09:
Have you checked the charger device for newly created sensors for Current, Voltage, ...? They might be inactivated by default. Could be that your unavailable sensors are old sensors that the integration no longer uses. 

The log message "Forcing number of connectors to -1" is probably just a bad log message (I'll fix that in the code). The integration keeps your num_connectors to 1.

arh-au, 2025-09-10:
Hi @jthun, I checked and there are no additional sensors for current, voltage, etc for the charger.