lbbrhzn/ocpp #1707: After updating to 0.10 "Energy Active Import Register" doesn't update while charging, only after charging has finished

tronsense, 2025-09-07:
**Describe the bug**
Before the update the "Energy Active Import Register" (energy_active_import_register) updated while loading so i was able to see the exact amount of loaded energy within the Energy-Dashboard.

After updating to 0.10 the energy_active_import_register only updates after the charging has finished, it jumps to the loaded amount:

**Expected behavior**
The energy_active_import_register should update itself while loading.

**Screenshots**
[image]

Energy Dashboard:

[image]

jthun, 2025-09-07:
Fixed in the next release. Is it possible to use the energy_session sensor meanwhile?

tronsense, 2025-09-08:
> Fixed (hopefully, please verify) in the next release. Is it possible to use the energy_session sensor meanwhile?

Unfortunately not, because for the energy dashboard I need a sensor that only raises (i don't know how to explain it otherwise), the energy_session session resets to 0 for every charge.
(Or i'm missing something for the energy dashboard settings)

jthun, 2025-09-08:
> in 0.10 energy_session also do snot work well.

Did it not update during the charging session? Or completely missing?

tronsense, 2025-09-08:
> > Did it not update during the charging session? Or completely missing?
> 
> For OCPP 1.6 chargers Energy.Active.Import.Register, session_energy and energy_meter_start are updated to 0 at transaction start. After some of my local hacks session_energy is updated ate the end of the session, but it does not update continuously.

Cant reproduce this, even before 0.10 energy_sessions resets to 0 when a new charging-cycle starts and updates continiously

[image]

jthun, 2025-09-08:
> Also I do not see any sensor attributes set, however, the code is there.

Probably related to #1704. Please check the next release.

altserg, 2025-09-08:
> Cant reproduce this, even before 0.10 energy_sessions resets to 0 when a new charging-cycle starts and updates continiously

Not in my case... Maybe I need to remove OCPP, clean the HASS registry and reinstall it completely...

altserg, 2025-09-08:
> > Also I do not see any sensor attributes set, however, the code is there.
> 
> Probably related to [#1704](https://github.com/lbbrhzn/ocpp/issues/1704). Please check the next release.

This patch will help attributes for Current, but I do not see any attributes for any of sensors. However, I'm pretty sure that the attribute value setting code is executed at least in chargepoint.py/process_measurands()

closed 2025-09-08 by drc38

altserg, 2025-09-09:
> Meter start is the another one may has issue. Its value was lagging behind the active import, suddenly jumped to the same value of the active import sensor at 15:00.. the charge session itself ended at 6:00.

Maybe it will be fixed with https://github.com/lbbrhzn/ocpp/pull/1710, together with issue #1709

painkillerluo, 2025-09-10:
No change for 0.10.1

painkillerluo, 2025-09-10:
Oh, just showed up after my last charge session. I'll try and report back.

CRCinAU, 2025-09-13:
Just for confirmation, this is a charge with 0.10.2. Numbers certainly don't add up.

[image]

PR #1708 (2025-09-08): Issue fixes for initial release of multiple connector support.
Update EAIR continuously (Fixes #1707). Handle clear profile exceptions when changing max current (Fixes #1706 and fixes #1703). Use proper connector id for extra attr (phase data) (Fixes #1704). Clear IdTag and reset transaction id to 0 on stop/unplug (Fixes #1702). Add more tests. Test coverage settings.

## Summary by CodeRabbit

- New Features
  - Optimistic UI for adjusting current limit (immediate slider response).
  - Improved multi-connector handling, including per-connector phase attributes and energy accounting.
- Bug Fix [... cut]