lbbrhzn/ocpp #1713: "SetChargingProfile failed (last status=None)." every 10 minutes

chkr1011, 2025-09-09:
**Describe the bug**
After upgrade to version 0.10.1 I get the notification "SetChargingProfile failed (last status=None)." every 10 minutes in HA

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade version
2. Restart HA
3. Look at notifications in HA

**Expected behavior**
No notification every 10 minutes. Ideally none at all because the error seems to be an internal issue the user cannot fix.

jthun, 2025-09-09:
Is the charger working as it should otherwise?

jthun, 2025-09-09:
Thanks for the log. This looks like the charger itself briefly losing power or rebooting than anything caused by the integration. You can see three reconnects (around 08:06, 08:18, and 08:29), and immediately after the latter two the charger reports ResetFailure with “Unintended Reset - Power Outage?” and even sends a StopTransaction with reason="PowerLoss". Those should be the charger’s own diagnostics, indicating a supply dip or some kind of internal watchdog reset.

I've fixed the notification spamming, so that will come in the next release, most likely.

chkr1011, 2025-09-10:
After reverting to version 0.9.0 everything works fine. See screenshot.

[image]

The availability is off but this is ok. So in my opinion there is a bug in the integration and the charger works fine.

jthun, 2025-09-10:
Ok. Please test the coming release to see if that fixes it for you.

jthun, 2025-09-10:
Another question: do you have any automations that runs regularly, like every 10 minutes, that does a SetChargingProfile?

chkr1011, 2025-09-10:
Bad news, I tried 0.10.2 but the behavior is basically the same. Unavailable every now and then.
My charger shows that it is trying to connect but cannot even if HA and integration is running properly. Every now and then it can connect but gets disconnected after a while. The charger is also reboot if connecting does not work for a certain amount of time or retries.

Reverting back to 0.9.0 fixed the issue and everything works as expected. I also removed all chargers from the integration and added everything from scratch but no success (when trying 0.10.2).

The only automation I have sets the limit of charger every 10 or so minutes. But no charging profile.

jthun, 2025-09-10:
What happens if you disable the automation, does it still become unavailable?

chkr1011, 2025-09-13:
What is the reason of changing this in 0.10? Are there chargers which require is?

jthun, 2025-09-13:
0.10 introduced support for chargers with multiple connectors, requiring quite a few code changes (and unfortunately also introducing a few issues).

commit 7a2d711c (2025-09-10): Average Current Values. More robust handling of meter values. (#1710)

* Average Current values. Handle cases where meter values bucket contains several EAIR with different contexts (Fixes #1709). Set energy_session initally to 0 to avoid spike. More tests.

* More EAIR handling, also covering ABB chargers with 0 on session start. More efficient measurand handling on init. Improved log message/handling in trigger_status_notification. Remove excessive logging and notification of SetChargingProfile fails (Fixes #1713). Handle Power.Factor. More tests.

---------