lbbrhzn/ocpp #1725: OCPP integration with Wallbox Pulsar Max – missing instant power sensor

CorceiroPT, 2025-09-15:
### The problem

I am using a Wallbox Pulsar Max with the OCPP integration in Home Assistant.  
The charger connects correctly and several entities are created. I can:

- Start and stop charging sessions  
- Adjust the charging current  
- See the energy consumed in a session  
- See the total accumulated energy  

However, the **instantaneous power sensor** always shows as "unknown" or "0".  
This means I cannot monitor real-time power usage, which is one of the most important metrics during charging.

### What I expected to happen

Along with session energy and total energy, I expected the OCPP integration to expose real-time instantaneous power values reported by the charger.

### What actually happened

The power sensor entity exists, but it remains stuck at `0,00 kW` during the whole charging session, even when the car is actively charging.

### Steps to reproduce

1. Set up OCPP integration with Wallbox Pulsar Max  
2. Start a charging session  
3. Observe entities created by the integration  

**Result:**  
Energy values update, but instantaneous power remains unknown.

### Environment

- Home Assistant version: Home Assistant OS 2025.9.3
- OCPP integration version: v0.10.2
- Charger model: Wallbox Pulsar Max  
- Charger firmware version: 6.6.12  

### Additional information

Other sensors (like current) also remain at `0,00 A`, which may indicate that the Pulsar Max is not sending full `MeterValues` over OCPP.  
It would be useful to confirm whether this is a limitation of the integration mapping or if additional charger configuration is required (e.g., enabling telemetry, but I can't find than option on the Wallbox application).  

Please advise if this is expected behavior, a missing feature, or a bug.

jostd, 2025-09-16:
I have had similar issues and edited the OCCP config, which can be found in the file /config/.storage/core.config_entries
I would suggest adding Power.Active.Import to monitored_variables
In my configuration I have:
"monitored_variables":"Current.Import,Energy.Active.Import.Register,Energy.Active.Import.Interval,Frequency,Power.Active.Import,Power.Offered"
and "monitored_variables_autoconfig":false

Maybe there is a way to do this via the configuration.yaml file rather than messing directly in core.config_entries

drewB, 2025-09-16:
I am having the same issue with my pulsar plus chargers.  Unfortunately, @jostd solution didn’t fix it for me.

jthun, 2025-09-16:
Can you please attach a debug log from a charging session? This is a value that normally comes from the charger.

jthun, 2025-09-16:
Your charger rejects the proposed MeterValuesSampleData configuration:

```
2025-09-16 11:18:10.858 INFO (MainThread) [ocpp] OCPP_Leandro: send [2,"2e0d3e1b-0e1e-4616-85b1-e9a82aaa9ab7","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Export,Current.Import,Current.Offered,Energy.Active.Export.Interval,Energy.Active.Export.Register,Energy.Active.Import.Interval,Energy.Active.Import.Register,Energy.Reactive.Export.Interval,Energy.Reactive.Export.Register,Energy.Reactive.Import.Interval,Energy.Reactive.Import.Register,Frequency,Power.Active.Export,Power.Active.Import,Power.Factor,Power.Offered,Power.Reactive.Export,Power.Reactive.Import,RPM,SoC,Temperature,Voltage"}]

2025-09-16 11:18:10.870 INFO (MainThread) [ocpp] OCPP_Leandro: receive message [3,"2e0d3e1b-0e1e-4616-85b1-e9a82aaa9ab7",{"status": "Rejected"}]
```

It then tells you what it will report:

```
2025-09-16 11:18:10.874 INFO (MainThread) [ocpp] OCPP_Leandro: send [2,"aa98601a-8833-4ee7-afd6-27e8e641b24a","GetConfiguration",{"key":["MeterValuesSampledData"]}]

2025-09-16 11:18:10.885 INFO (MainThread) [ocpp] OCPP_Leandro: receive message [3,"aa98601a-8833-4ee7-afd6-27e8e641b24a",{"unknownKey": [],"configurationKey": [{"readonly": false,"key": "MeterValuesSampledData","value": "Energy.Active.Import.Register"} ]}]
```
When the charging starts, the MeterValues message from the charger only contains the `Energy.Active.Import.Register` data.

Have you tried selecting just a few measurands manually, as per the [user guide](https://github.com/lbbrhzn/ocpp/wiki/user-guide)? Maybe your charger is rejecting some of the values in the big list?

CorceiroPT, 2025-09-16:
> Have you tried selecting just a few measurands manually, as per the [user guide](https://github.com/lbbrhzn/ocpp/wiki/user-guide)? Maybe your charger is rejecting some of the values in the big list?

I read the user guide but I didn't quite understand how can i select justo a few measurements manually, @jthun. Should I just deactivate the sensors on the OCPP Home Assistant Integration or I need to configure it in some file inside Home Assistant as @jostd said?

jthun, 2025-09-17:
Please set `ocpp` log level to debug as well (now you have the `custom_components.ocpp` log level set to debug).

[image]

CorceiroPT, 2025-09-17:
> Remove the integration and re-add it. Deselect automatic detection of measurands when configuring, and instead select a few manually. @jostd  had a good list of measurands above.

Following your suggestion and using only some measurands like @jostd indicated, I manage to make it work!
Thank you very much for your help, @jthun.

A wonderful day for you all!

[image]

drewB, 2025-09-17:
For clearing profile, when I try, I keep getting an unknown error.  In the logs, it is saying invalid id.  For data, I have tried `devid: charger` `devid: <cpid>` `devid: <ha sensor prefix id>`

drewB, 2025-09-17:
That is my cpid.  Got the same error when I tried that.

drewB, 2025-09-17:
I reset the charger as you suggested and that fixed the issue.  Now I am getting all the measures I would expect!  Thank you for your help!

jthun, 2025-09-20:
@CorceiroPT, please close the issue.

closed 2025-09-20 by CorceiroPT