lbbrhzn/ocpp #1734: Maximum Current not changing

ricardo88faria, 2025-09-19:
Hello,

I am using a FoxESS A7300P1-E-2-WO wallbox connected to Home Assistant through this OCPP integration.
The wallbox is also connected to a FoxESS H1-6.0-E-G2 inverter via RS485 (for solar linkage), but according to FoxESS support this should not block OCPP parameters.

The integration is working partially:

✅ The wallbox connects successfully to Home Assistant via OCPP.

✅ Telemetry data (status, power, energy, etc.) is received.

✅ Start/Stop charging works correctly.

❌ I cannot set or change the Max Charging Current from Home Assistant.

According to FoxESS support, the wallbox complies with OCPP and exposes the Max Charging Current parameter through OCPP (same parameter shown in the FOXSwitch app). They confirmed that the parameter is controlled by the charger itself, not by the inverter, and should be available through the OCPP Smart Charging profile.

My question:

Does the current implementation of this HA OCPP integration fully support the MaxChargingCurrent from OCPP 1.6J?

[home-assistant_ocpp_2025-09-19T10-13-02.771Z.log](https://github.com/user-attachments/files/22425445/home-assistant_ocpp_2025-09-19T10-13-02.771Z.log)

[home-assistant_ocpp_2025-09-19T10-13-02.771Z.log](https://github.com/user-attachments/files/22425448/home-assistant_ocpp_2025-09-19T10-13-02.771Z.log)

If yes, how can I expose or send this parameter to the FoxESS charger?
If not, would it be possible to add support for this, since it is a critical control for dynamic charging?

Additional information:

HA core: 2025.9.3
OCPP integration version: v0.10.3
Wallbox: FoxESS A7300P1-E-2-WO
Inverter: FoxESS H1-6.0-E-G2

Logs attached

Thanks a lot for your work on this integration!

ricardo88faria, 2025-09-19:
This is very odd everything seems ok, but this is not effectively changing the maximum current.
check the logs running the ocpp.clear_profile action befor and after changing the maximum current:

2025-09-19 16:00:57.663 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargingScheduleAllowedChargingRateUnit: Current,Power
2025-09-19 16:00:57.664 INFO (MainThread) [ocpp] central: send [2,"24fa74ac-dccb-43c0-851c-45a72ea01ca1","GetConfiguration",{"key":["ChargeProfileMaxStackLevel"]}]
2025-09-19 16:00:57.702 INFO (MainThread) [ocpp] central: receive message [3,"24fa74ac-dccb-43c0-851c-45a72ea01ca1",{"configurationKey":[{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"9"}]}]
2025-09-19 16:00:57.703 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargeProfileMaxStackLevel: 9
2025-09-19 16:00:57.705 INFO (MainThread) [ocpp] central: send [2,"9338bce5-ab63-4cad-b667-c056cc1be6e9","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":1000,"stackLevel":9,"chargingProfileKind":"Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":15.0}]}}}]
2025-09-19 16:00:57.776 INFO (MainThread) [ocpp] central: receive message [3,"9338bce5-ab63-4cad-b667-c056cc1be6e9",{"status":"Accepted"}]
2025-09-19 16:01:04.314 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'central': ping=0.0 ms, pong=81.0 ms
2025-09-19 16:01:13.407 INFO (MainThread) [ocpp] central: send [2,"d2d32eb6-9b7f-4cc3-a6ca-532d93dfca69","ClearChargingProfile",{}]
2025-09-19 16:01:13.442 INFO (MainThread) [ocpp] central: receive message [3,"d2d32eb6-9b7f-4cc3-a6ca-532d93dfca69",{"status":"Accepted"}]
2025-09-19 16:01:17.265 INFO (MainThread) [ocpp] central: send [2,"9aa23f16-4bc0-44a6-9bb0-d924f6e6c012","GetConfiguration",{"key":["ChargingScheduleAllowedChargingRateUnit"]}]
2025-09-19 16:01:17.323 INFO (MainThread) [ocpp] central: receive message [3,"9aa23f16-4bc0-44a6-9bb0-d924f6e6c012",{"configurationKey":[{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current,Power"}]}]
2025-09-19 16:01:17.324 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargingScheduleAllowedChargingRateUnit: Current,Power
2025-09-19 16:01:17.325 INFO (MainThread) [ocpp] central: send [2,"d863c80f-757a-4ff7-ae2d-92fc43853cbc","GetConfiguration",{"key":["ChargeProfileMaxStackLevel"]}]
2025-09-19 16:01:17.374 INFO (MainThread) [ocpp] central: receive message [3,"d863c80f-757a-4ff7-ae2d-92fc43853cbc",{"configurationKey":[{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"9"}]}]
2025-09-19 16:01:17.375 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargeProfileMaxStackLevel: 9
2025-09-19 16:01:17.387 INFO (MainThread) [ocpp] central: send [2,"07a4515a-bdac-459b-a56b-61d7c8d57d8c","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":1000,"stackLevel":9,"chargingProfileKind":"Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":25.0}]}}}]
2025-09-19 16:01:17.443 INFO (MainThread) [ocpp] central: receive message [3,"07a4515a-bdac-459b-a56b-61d7c8d57d8c",{"status":"Accepted"}]

jthun, 2025-09-19:
What current do you get when starting a charging session? None?

ricardo88faria, 2025-09-19:
No, I actually get the current number actually charging.

ricardo88faria, 2025-09-19:
its a 60A >> than the 32A wallbox limit