lbbrhzn/ocpp #1739: Maximum current unavailable

emitclever, 2025-09-19:
Upgrade from 0.9.X to 10.0.2 and tried again 0.9.X to 10.0.4 on both instances I had 2 maximum current entities after upgrade

number.charger_maximum_current      -- unavailable
number.charger_maximum_current_2  -- available

So I tough there is a new one I deleted the old one and renamed the second one
from number.charger_maximum_current_2 to number.charger_maximum_current

After complete restart of home assistant the number.charger_maximum_current is unavailable now

It's confusing, I do not know what is happening. I will reverse back to 0.9.X seems best fit for my purposes.
Time session is does not contain anymore minutes.

jthun, 2025-09-22:
Have you tried reloading the integration? Or remove - restart HA - reinstall the integration?

emitclever, 2025-09-22:
everything except re-install. for now I stick with 0.9.0. I want to keep the history. If I upgrade everything is different.