lbbrhzn/ocpp #1740: The unit of sensor.charger_time_session (None)

emitclever, 2025-09-19:
Error log

The unit of sensor.charger_time_session (None) cannot be converted to the unit of previously compiled statistics (min). Generation of long term statistics will be suppressed unless the unit changes back to min or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this

StroppyOz, 2025-09-25:
This may also be related. My session time was reported as follows:

[image]

No errors in the logs that I can find. It fixed itself somehow after I added session time to my Dashboard.

nils1981, 2025-09-25:
I have same issue. It seems the error is related to reconnections during charging. In my case my internet connections is renewed in the night and I can see the charge time session always spikes then. 

[image]