lbbrhzn/ocpp #1744: EVBox Elvi loses connection after update to v0.10.0 and newer (works with v0.9.0)

tihenkel, 2025-09-22:
Describe the bug
My EVBox Elvi wallbox works perfectly and is stable with version v0.9.0 of the OCPP integration. However, immediately after updating the integration to any version from v0.10.0 onwards, the wallbox loses its connection to the Home Assistant OCPP server. The wallbox itself indicates an error state with a continuous orange blinking light. Downgrading back to v0.9.0 resolves the issue instantly and the connection is stable again.
To Reproduce
Steps to reproduce the behavior:
Have a working setup with an EVBox Elvi and OCPP integration v0.9.0.
Update the OCPP integration via HACS to the latest version (e.g., v0.10.0 or newer).
Restart Home Assistant.
Observe that the wallbox fails to connect to the server. The ocpp_event log in Home Assistant shows a successful BootNotification followed by connection drops. The wallbox status light starts blinking orange.
Expected behavior
The wallbox should connect to the OCPP server and maintain a stable connection after the update, just as it does with v0.9.0.
Screenshots & Logs
We have observed the following behavior in the logs when using a version newer than v0.9.0. The system receives a BootNotification from the charge point, but the connection seems to drop immediately after.

Log example from a failing version:
Plain Text
2025-09-10 20:38:29,975 INFO ocpp | Charge Point eagle-219 connected to central system.
2025-09-10 20:38:30,223 INFO ocpp.v16.charge_point | Charge point eagle-219 received a BootNotification.
...
<-- After this, the connection becomes unstable and errors appear -->

Configuration:
Wallbox Model: EVBox Elvi
Home Assistant Version: Home Assistant Core Installierte Version 2025.9.4
Home Assistant Installation Type: Home Assistant Operating System Installierte Version 16.2
OCPP Integration Version (working): v0.9.0
OCPP Integration Version (failing): v0.10.0 and all subsequent versions.
Additional context
This issue seems specific to the interaction between the EVBox Elvi and changes introduced in the OCPP integration after v0.9.0. The immediate stability upon downgrading confirms that the base configuration (network, IP, charge point ID) is correct.

tihenkel, 2025-09-22:
2025-09-10 19:13:22,231 - homeassistant.components.ocpp.charge_point - INFO - send [2,"c6e36e9b-20cf-4d3b-9960-cfaf72c90a2","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":1,"stackLevel":1,"chargingProfilePurpose":"ChargePointMaxProfile","chargingProfileKind":"Recurring","recurrencyKind":"Daily","chargingSchedule":{"duration":86400,"startSchedule":"2025-09-10T19:13:22.231Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":16.0}]}}}]
2025-09-10 19:13:22,291 - homeassistant.components.ocpp.charge_point - INFO - receive message [3,"c6e36e9b-20cf-4d3b-9960-cfaf72c90a2",{"status":"Rejected"}]
2025-09-10 19:13:22,292 - homeassistant.components.ocpp.charge_point - WARNING - SetChargingProfile failed (last status=Rejected)
2025-09-10 19:13:22,382 - homeassistant.components.ocpp.charge_point - INFO - receive message [2,"1694373202381","StatusNotification",{"timestamp":"2025-09-10T19:13:22.339Z","connectorId":1,"errorCode":"NoError","status":"SuspendedEVSE"}]

The key finding is that v0.9.0 works perfectly, while any version from v0.10.0 onwards is unstable or completely non-functional with the EVBox Elvi.

Thank you for your understanding and your help!

tihenkel, 2025-09-22:
Hi @jthun,

Thank you for your suggestion. I have followed your instructions precisely, but unfortunately, the problem persists and the situation has become even more unstable. I have now restored a backup of my system to get back to the working version v0.9.0.

Here is a detailed report of what happened after following your plan:

**1. Update to v0.10.4 & Restart:**
I updated the integration to v0.10.4 via HACS and restarted Home Assistant.

**2. Executed `ocpp.clear_profile`:**
After the restart, I successfully executed the `ocpp.clear_profile` service with `devid: charger`.

**3. Initial Success:**
For a brief moment, this seemed to work! The wallbox LED turned green, and I was able to start a charging session manually.

**4. The Failure:**
However, this stable state did not last. After some time (or possibly after the next Home Assistant restart, it's hard to tell), the integration fell back into a broken state.

**5. The Final State (The "Groundhog Day" Scenario):**
The `number.charger_maximum_current` entity was duplicated. One entity was "unavailable" and orphaned, the other one was "live" but completely unresponsive. Moving the slider had no effect. The system was no longer able to control the wallbox at all, even in manual mode. This is the exact same failure state we experienced in our very first attempts.

**6. The Proof from the Logs:**
I was able to capture the logs during this final broken state. They clearly show that the `number.charger_maximum_current` entity is missing or unavailable, and that `SetChargingProfile` commands are still being `Rejected` by the wallbox, which seems to corrupt the integration's state.

Here is the relevant log excerpt from the final failed state on v0.10.4:

WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity number.charger_maximum_current not found.
WARNING (MainThread) [homeassistant.helpers.service] Referenced entities number.charger_maximum_current are missing or not currently available
...
INFO (MainThread) [ocpp] EVB-P21322974: send [2,"50793a38-5785-4328-ada5-f3a6eb6195d2","SetChargingProfile",{"connectorId":0,...}]
INFO (MainThread) [ocpp] EVB-P21322974: receive message [3,"50793a38-5785-4328-ada5-f3a6eb6195d2",{"status":"Rejected"}]
DEBUG (MainThread) [custom_components.ocpp] ChargePointMaxProfile not accepted (Rejected); will continue.

**Conclusion:**
The `ocpp.clear_profile` service seems to be a temporary fix that does not survive a restart or a certain amount of time. The root cause of the incompatibility between the v0.10.x branch and the EVBox Elvi persists. The only 100% stable and reliable solution is to remain on version **v0.9.0**.

I have now restored a full backup to return to this stable version and will not be performing any further updates until there is a confirmed, fundamental fix for the EVBox compatibility.

Thank you for your time and effort. I hope this detailed feedback helps in finding a permanent solution.

jthun, 2025-09-22:
Hi,

We need a lot of debug logs to be able to see what's going wrong in your setup. No developer has an EVBox Elvi to test with, that I know of, and this behaviour has not been seen with any other box, so unless you can provide us with information about what's not working, we can't do anything.

A final thing to try would be to do a reload (or maybe even a remove - restart HA - reinstall) of the integration. Maybe you have already done that?

jthun, 2025-09-22:
I'm afraid that log doesn't say anything about your problem, more than that there are issues with your entities. Did you try to reload the integration after upgrading? This is key to get the entities on track. Did you remove - restart HA - reinstall the integration? If you do, please set the logs to debug, as described in the documentation. That will generate more output so that we can see what the integration is doing and potentially why it's not working properly for you.

You might also want to disable any automation you have, that is using the maximum_current entity, while testing.

And when you use the ocpp.clear_profile action, please also try with `devid: EVB-P21322974`

mineadle, 2025-09-25:
Good Day, I faced the same issue with Elvi Charger. Updated from v0.8.2 to v0.10.4 and had same issue related to the 'number.maximum_current' is not avaliable. Returned back to v0.8.2

jthun, 2025-09-29:
Hi @Meetjasper. We are still missing debug logs with some kind of hint of where the problem with EVbox Elvi is, so if you could enable debug logging (as per the [docs](https://github.com/lbbrhzn/ocpp/wiki/debugging)), install the latest version of the integration, maybe change the max current a few times and then paste the logs here. You can also do a reload of the integration from the integration's home page.

Meetjasper, 2025-09-29:
Hi @jthun, 

after charging my car I reconnected the EVbox to the HA platform again (+/- 16:15h). Everything worked until 19:25h. After that the EVbox integration failed. The log is uploaded.
The version of OCPP i'm running is v0.10.4. There are no switches available to change in current state.

I hope this gives you a hint in the right direction. 

[home-assistant.log - Debug ocpp.log](https://github.com/user-attachments/files/22605862/home-assistant.log.-.Debug.ocpp.log)

Meetjasper, 2025-09-30:
Good evening @jthun,

Thanks for the feedback. I reinstalled the charger and manually selected some measurands. I also changed the ping and timeouts according to your advice.
Until now everything works fine (the system has been up for about 5 hours).

What I meant with, 'there are no switches available....', is: I saw the switches and sliders but they were grayed out due to unavailability of the charger. 
At this moment I can change the switches and sliders again.

I'll keep you informed about the status, Probably this will be after the weekend.

Thanks for your help so far!

Meetjasper, 2025-10-07:
I promised to follow up after the weekend on the findings of the OCPP integration with my charger.
Last week, I installed and got the entire charging platform up and running. When I checked again after the weekend, the charger was again fully marked as “unavailable” after about roughly 24 hours of working. The whole device is greyed out and marked unavailable (see screenshot) in the HA portal. The last log entry was as follows:

2025-10-01 15:25:28.609 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'EVB-Pxxxx': ping=0.0 ms, pong=81.0 ms
2025-10-01 15:26:13.975 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'EVB-Pxxxx': ping=0.0 ms, pong=364.0 ms
2025-10-01 15:26:17.039 DEBUG (MainThread) [custom_components.ocpp] Connection closed to 'EVB-Pxxxx': no close frame received or sent

After that, it was impossible to reconnect to the charger (until tonight). So I tried the following steps (not necessarily in order):

- Removed the OCPP integration from HA
- Re-added the OCPP integration to HA
- Downgraded OCPP to v0.9.0
- Connected the EVBox Elvi to the old platform — it was visible there
- Set the EVBox Elvi to Offline mode
- Set the EVBox Elvi to Online mode
- Set the EVBox Elvi to autostart mode
- Set the EVBox Elvi to “start via RFID chip”
- Performed a factory reset on the EVBox Elvi
- Re-added all RFID chips
- Reconnected Wi-Fi
- Upgraded OCPP to v0.10.4

So in troubleshooting I verified the following:

- EVBox Elvi starts charging in both offline and online mode
- EVBox Elvi responds to ping
- Once the OCPP integration is installed in HA, the EVBox Elvi is detected
- EVBox Elvi added with limited measurands and non-default ping settings
  a. Websocket ping interval: 45 seconds
  b. Websocket ping timeout: 10 seconds
  c. Measurands: “Energy.Active.Import.Register” is the only checkbox checked

After that the EVBox Elvi was available in the app between approximately 16:00 and 19:00 today, after which it again became completely “unavailable.” The logs didn't indicate any reason for the outage/unavailability. It’s a complete mystery to me.

What I did notice today were the sensor values — some fields (i.e. idtag, transactionid, stop reason) were regularly reset to “unknown” or left blank. That did not happen the previous time the charger was working.

Tomorrow is another day to continue troubleshooting.

[image]
[image]
[image]
[image]